framps/raspiBackup #719: Bug in raspiBackupInstallUI - if you want custom time using systemd, it create incorrect format, fix is attached in text

marulinko, 2023-12-16:
There is bug in function - function systemd_update_execute
local v=$(awk -v minute=$CONFIG_SYSTEMD_MINUTE -v hour=$CONFIG_SYSTEMD_HOUR -v day=$systemd_day ' { print "OnCalendar="day "*-*-*", hour":"minute":42" }' <<< "$l")

The fix is:
local v=$(awk -v minute=$CONFIG_SYSTEMD_MINUTE -v hour=$CONFIG_SYSTEMD_HOUR -v day=$systemd_day ' { print "OnCalendar="day,"*-*-*", hour":"minute":42" }' <<< "$l")

OR

local v=$(awk -v minute=$CONFIG_SYSTEMD_MINUTE -v hour=$CONFIG_SYSTEMD_HOUR -v day=$systemd_day ' { print "OnCalendar="day" *-*-*", hour":"minute":42" }' <<< "$l")

How to test:
awk -v minute=00 -v hour=03 -v day=Sun ' { print "OnCalendar="day,"*-*-*", hour":"minute":42" }' <<< $(echo a)
OR
awk -v minute=00 -v hour=03 -v day=Sun ' { print "OnCalendar="day" *-*-*", hour":"minute":42" }' <<< $(echo a)

Explanation.
day is variable, so you don't need add space after that. You can create space adding , after variable
or add space at the begin of text " *-*-*-"

framps, 2023-12-17:
Thank you very much for your hint. I struggled with this and finally added following line https://github.com/framps/raspiBackup/blob/14133a129928da24770752bd785e44bf1ae3a671/installation/raspiBackupInstallUI.sh#L4826 which adds an additional space. So it's not a bug.

But I will change the code in next release according your proposal. Your code is much cleaner to read. :+1:

framps, 2023-12-17:
I just noticed your proposal doesn't work for daily because there no space should be present.

```
#	OnCalendar=Sun *-*-* 05:00:42 # on sunday
#	OnCalendar=*-*-* 05:00:42     # daily
```

closed 2023-12-17 by framps

marulinko, 2023-12-18:
Thank you for checking this. It was late night, and I checked only that line. And I didn't notice that if there is no Day defined,then no space is required. Great code anyway, I learned many things from it.