framps/raspiBackup #722: Support F2Fs

framps, 2023-12-29:


framps, 2023-12-29:
See https://github.com/framps/raspiBackup/tree/f2fs

framps, 2023-12-29:
@kosi2801 Please verify

kosi2801, 2023-12-30:
Thank you very much for your efforts. I managed to finish one backup with this version now.

The backup process with rsync itself looks okay and produces a similar result on the NFS mounted directory as expected also with other filesystems. So the filedata itself is backed up properly I think.

What does not work seems to be to reproduce the exact options that were used to create the f2fs filesystem initially with mkfs.f2fs. eg. I have created the filesystem with the command 
`mkfs.f2fs -l raspi -O extra_attr,inode_checksum,sb_checksum,compression /dev/mmcblk0p3` 
and I think during restore operation raspiBackup just uses the default options.

I'm not sure how this works with special fs features on the other filesystem types, for f2fs this could result in a system that is unable to boot because eg. /boot/commandline.txt or /etc/fstab contain mount options that rely on certain f2fs options to be present.
But at least those are scenarios that do not result in unrecoverable data loss and can be mitigated by either externally removing/changing the mount options that rely on fs features or by manually creating the partition structure the proper way and copying back the data.

I tried to find out if there is a way to recover the options that were used for mkfs.f2fs but I was unsuccessful to come up with any command that returns those.
The only way I can think of currently would be to conlcude from the 'mount' output of the mounted fs which options rely on fs features (eg. 'atgc' mount option relies on 'extra_attr' mkfs.f2fs option) and add those to the mkfs.f2fs execution. But I doubt that that's easy to add. (Also, I'm not sure if just mounting to a tmp mountpoint shows already all required options or if it has to be mounted according to the /etc/fstab config).

But for my purposes, that's already all I need, I would likely create the partition structure manually in a recovery scenario anyway and copy back the partitions contents by hand. :-)

Disclaimer: I did *not* yet try to restore a backup, I currently don't have a spare Raspberry Pi around for testing.

framps, 2023-12-30:
> What does not work seems to be to reproduce the exact options that were used to create the f2fs filesystem initially with mkfs.f2fs. eg. I have created the filesystem with the command
mkfs.f2fs -l raspi -O extra_attr,inode_checksum,sb_checksum,compression /dev/mmcblk0p3
and I think during restore operation raspiBackup just uses the default options.

Right. During restore all filesystems are created with default options only. 

> for f2fs this could result in a system that is unable to boot because eg. /boot/commandline.txt or /etc/fstab contain mount options that rely on certain f2fs options to be present.

I saw there are a lot of options available for f2fs. Maybe you can use a restore exit to manipulate and update the options of the f2fs?

> I would likely create the partition structure manually in a recovery scenario anyway and copy back the partitions contents by hand. :-)

Maybe you can do this in an restore extension?

> Disclaimer: I did not yet try to restore a backup, I currently don't have a spare Raspberry Pi around for testing.

Can't you just restore to a different SD card or is your Raspberry in production 7/24? Anyway - do you see a chance to verify the fix? Take your time. I don't have any schedule for next release of raspiBackup but usually every 6-9 months a new release is published. Given latest release was published recently you have enough time  :wink:

I'll keep this issue open until you verified the fix with a restore.

kosi2801, 2024-01-02:
Ok, I could now try to restore a backup, unfortunately it failed with folloging log:

> root@dockerpi:~# ./raspiBackup.sh -d /dev/sda /backup/raspitouch/raspitouch-rsync-backup-20231229-215847/
> --- RBK0009I: dockerpi: raspiBackup.sh V0.6.9.1-dev-f2fs - 2023-12-28 (40cf5ed) Tue  2 Jan 23:19:34 CET 2024 gestartet.
> !!! RBK0192W: =========> HINWEIS <=========
> !!! RBK0192W: Dieses ist eine Entwicklerversion welcher nicht in Produktion benutzt werden sollte.
> !!! RBK0192W: =========> HINWEIS <=========
> !!! RBK0161W: Ziel /dev/sda mit 119.08 GiB ist größer als die Backupquelle mit 29.77 GiB. 89.30 GiB sind ungenutzt.
> !!! RBK0065W: Gerät /dev/sda wird repartitioniert und die gesamten Daten werden gelöscht.
> 
> !!! RBK0100W: Gerät /dev/sda wird mit dem Backup beschrieben.
> --- RBK0038I: Bist Du sicher? j/N
> j
> --- RBK0095I: Backup wird auf Partition /dev/sda1 zurückgespielt.
> --- RBK0099I: Zurückspielen des Backups auf Partition /dev/sda1 beendet.
> --- RBK0095I: Backup wird auf Partition /dev/sda2 zurückgespielt.
> --- RBK0099I: Zurückspielen des Backups auf Partition /dev/sda2 beendet.
> ??? RBK0001E: Unerwarteter Programmfehler trat auf. (40cf5ed), Zeile: 8763, Fehler: Invalid arg.
> ??? RBK0148E: @@@@@@@@@@@@@@@@@@@@ Stacktrace @@@@@@@@@@@@@@@@@@@@
>   File "./raspiBackup.sh", line 9683, in main
>     doit # no return
>   File "./raspiBackup.sh", line 6406, in doit
>                 doitRestore
>   File "./raspiBackup.sh", line 8104, in doitRestore
>                 restorePartitionBasedBackup
>   File "./raspiBackup.sh", line 7533, in restorePartitionBasedBackup
>                 restorePartitionBasedPartition "$partitionBackupFile"
>   File "./raspiBackup.sh", line 7754, in restorePartitionBasedPartition
>                                 check4RequiredCommands f2fs
>   File "./raspiBackup.sh", line 8763, in check4RequiredCommands
>                         assertionFailed $LINENO "Invalid arg"
>   File "./raspiBackup.sh", line 2517, in assertionFailed
>         logStack
> --- RBK0033I: Bitte warten bis aufgeräumt wurde.
> ??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
> --- RBK0010I: raspitouch: raspiBackup.sh V0.6.9.1-dev-f2fs - 2023-12-28 (40cf5ed) Tue  2 Jan 23:25:27 CET 2024 beendet mit Returncode 101.
> --- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.logr gesichert.
> 

The `f2fs-tools` package is installed properly.

> root@dockerpi:~# dpkg -l | grep -i f2fs
> ii  f2fs-tools                           1.15.0-1                         arm64        Tools for Flash-Friendly File System
> 

Before I upload the `/root/raspiBackup.logr` to Github, does this file contain any sensitive/private information I should mask before doing so?

kosi2801, 2024-01-07:
I managed to get backup and restore with basic/default options working, see PR #725 .

framps, 2024-01-19:
Code rebased into development branch

framps, 2024-10-02:
Merged into m_778_upd

framps, 2024-10-05:
@kosi2801 Next release 0.6.10 is system tested :wink: f2fs support is included now in the development branch. I don't have any means to test f2fs. Would be great if you can verify f2fs support will work in next release :wink:

kosi2801, 2024-10-06:
@framps thanks for the notification. I did a quick update to the development branch with 
`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- development`
and tried to start a backup using that downloaded version with
`./raspiBackup.sh`

Unfortunately it fails to perform the backup using the existing settings giving:
```
--- RBK0009I: dockerpi: raspiBackup.sh V0.6.10-dev - 2024-10-06 (0ec128e) Sun  6 Oct 21:58:40 CEST 2024 gestartet.
!!! RBK0192W: =========> HINWEIS <=========
!!! RBK0192W: Dieses ist eine Entwicklerversion welcher nicht in Produktion benutzt werden sollte.
!!! RBK0192W: =========> HINWEIS <=========
??? RBK0298E: Filesystem vfat auf boot und/oder f2fs auf root ist nicht unterstützt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0320I: Unvollständiges Backup wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: dockerpi: raspiBackup.sh V0.6.10-dev - 2024-10-06 (0ec128e) Sun  6 Oct 21:58:42 CEST 2024 beendet mit Returncode 102.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.
```

I also tried branch m_778_upd which yields also a "unsupported" error but has also another additional issue:
```
--- RBK0009I: dockerpi: raspiBackup.sh V0.6.10-dev - 2024-10-02 (be7a082) Sun  6 Oct 22:04:11 CEST 2024 gestartet.
!!! RBK0192W: =========> HINWEIS <=========
!!! RBK0192W: Dieses ist eine Entwicklerversion welcher nicht in Produktion benutzt werden sollte.
!!! RBK0192W: =========> HINWEIS <=========
??? RBK0298E: Filesystem vfat auf boot und/oder f2fs auf root ist nicht unterstützt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
./raspiBackup.sh: line 5342: umountSDPartitions: command not found
--- RBK0320I: Unvollständiges Backup wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: dockerpi: raspiBackup.sh V0.6.10-dev - 2024-10-02 (be7a082) Sun  6 Oct 22:04:13 CEST 2024 beendet mit Returncode 102.
--- RBK0321I: Logdatei wird gesichert.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.
```

When I do the same procedure with the **f2fs** branch it properly starts backing up without bailing out with a RBK0298E. 
Am I checking the wrong branches or maybe I'm hitting a merge issue?

kosi2801, 2024-10-07:
@framps Thank you very much, taking a backup now seems to work :-)

But restoring doesn't seem to work properly now. I'm not sure if there is something totally off with my available setup (the card to restore seems a bit smaller than the backup source).

My command and the output:
```
root@dockerpi:/backup# ./raspiBackup.sh -d /dev/sda /backup/raspitouch/raspitouch-rsync-backup-20241007-030042/
--- RBK0009I: dockerpi: raspiBackup.sh V0.6.10-dev - 2024-10-07 (2d34d2d) Mon  7                                                                                                                                                              Oct 23:19:36 CEST 2024 gestartet.
!!! RBK0192W: =========> HINWEIS <=========
!!! RBK0192W: Dieses ist eine Entwicklerversion welcher nicht in Produktion benutzt werden sollte.
!!! RBK0192W: =========> HINWEIS <=========
--- RBK0067I: Momentane Partitionen auf /dev/sda:
NAME  SIZE FSTYPE LABEL
sda  28.8G vfat   NEW
!!! RBK0325W: Gerät /dev/sda wird aktualisiert.
--- RBK0038I: Bist Du sicher? j/N
j
!!! RBK0305W: Partition(en) "1 2" werden auf /dev/sda zurüchgespielt.
--- RBK0038I: Bist Du sicher? j/N
j
!!! RBK0097W: Partitioniere /dev/sda.
--- RBK0038I: Bist Du sicher? j/N
j
!!! RBK0322W: Ziel /dev/sda mit 28.78 GiB ist kleiner als die Backupquelle mit 29.77 GiB. Die letzte Partition wird entsprechend verkleinert. HINWEIS: Der Restore kann fehlschlagen wenn sie zu klein wird.
--- RBK0300I: Letzte Partition wird von 23.66 GiB auf 22.67 GiB angepasst.
??? RBK0088E: Option -R wird nicht beim partitionbasierten Backup unterstützt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: raspitouch: raspiBackup.sh V0.6.10-dev - 2024-10-07 (2d34d2d) Mon  7 Oct 23:39:26 CEST 2024 beendet mit Returncode 108.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.logr gesichert.
```

Things I'm not sure about:
* the backup source has 3 partitions, mmcblk0p1 (fat32), mmcblk0p2 (ext4), mmcblk0p3 (f2fs) but the restore only lists partitions "1 2" to be restored to /dev/sda. Is this correct?
* is the order correct that the partitioning happens _after_ the restore of the partions "1 2"?
* I did nowhere give the option "-R" that it complains about, is this maybe an artefact of the shrinking decision?

Sorry if those questions maybe sound silly, I haven't had the need to perform a restore until now (yes I know.... -.-) and I don't know the output of a proper restore process yet.
Does the log output above already suffice to give further instructions or do you need further info from the files in the backup (eg. more detailled structure etc.)?

framps, 2024-10-08:
Next release will have some enhancements for the partition oriented mode: 

No longer are all partitions per default restored. You can specify which partitions should be restored now. Therefore the default for restore changed from `-T "*" `to `-T "1 2"` to be consistent with the default for backup. You have to use `-T "*"` for the backup and also for restore now if you want to backup and restore all partitions.

> * is the order correct that the partitioning happens _after_ the restore of the partions "1 2"?

This doesn't make sense and is not done. Why do you think so?

> * I did nowhere give the option "-R" that it complains about, is this maybe an artefact of the shrinking decision?

No. It was just a bug :cry: 

> Sorry if those questions maybe sound silly

They are not :wink: 

> Does the log output above already suffice to give further instructions or do you need further info from the files in the backup (eg. more detailled structure etc.)?

The debug log usually has all the info I need to find bugs.

Thank you very much for your patience. I just executed you scenario again - but with ext4 only - and everything went through. I double checked also the code and f2fs is present in the format path. So I expect your test will succeed or will fail because of some f2fs reasons. 

Crossing fingers the test will succeed now.

kosi2801, 2024-10-09:
@framps Thank you for your efforts. I did a retest of the restore procedure after updating the script again using the suggested `-T "*"` on an existing backup and now everything went through without errors :-)

The restored image on the SD card also boots up properly and from my perspective everything looks fine!
I did not yet have the time to test further possible combinations of options and ways of backup/restore as I lack the time for testing scenarios that I don't have in use, but I think the main functionality and "happy path" should be okay now :-)

Thank you very much!

framps, 2024-10-09:
Great 👍 Thank you very much @kosi2801

closed 2024-10-09 by framps