framps/raspiBackup #723: RBK0034E ... blkid not found

hegien, 2023-12-31:
Hi,
i created a backup of my raspberry "raspi02" without any errors, using the command
raspiBackup -m detailed
The backup device is an USB stick mounted on /mnt/raspiBackup.
rsync is used as backup type
For restore test i moved the Backup USB stick to my other raspberry "raspi01" and  mounted it there also at /mnt/raspiBackup.
A SD card is pluged in via USB adapter and is recognized by the system as sdb.
Both raspis are running with bookworm and are up to date.
When i try to restore the backup, with command
raspiBackup.sh -d /dev/sdb /mnt/raspiBackup/raspi02/raspi02-rsync-backup-20231231-132319/
then i get the error
??? RBK0034E: File /mnt/raspiBackup/raspi02/raspi02-rsync-backup-20231231-132319/raspi02-backup.blkid not found.
Indeed there is no file raspi02-backup.blkid. 
What is going wrong here.
I already searched for this error but couldn't find any hint.

framps, 2023-12-31:
Would you please attach the debug log of the restore run?

hegien, 2023-12-31:
root@raspi01:~# cat raspiBackup.logr
20231231-135034 DBG 2388:  --> isSupportedEnvironment
20231231-135034 DBG 2399:          *** cat /etc/os-release
20231231-135034 DBG 2399:              PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
20231231-135034 DBG 2399:              NAME="Debian GNU/Linux"
20231231-135034 DBG 2399:              VERSION_ID="12"
20231231-135034 DBG 2399:              VERSION="12 (bookworm)"
20231231-135034 DBG 2399:              VERSION_CODENAME=bookworm
20231231-135034 DBG 2399:              ID=debian
20231231-135034 DBG 2399:              HOME_URL="https://www.debian.org/"
20231231-135034 DBG 2399:              SUPPORT_URL="https://www.debian.org/support"
20231231-135034 DBG 2399:              BUG_REPORT_URL="https://bugs.debian.org/"
20231231-135034 DBG 2407:      --- Modelpath: Raspberry Pi 4 Model B Rev 1.5
20231231-135034 DBG 2412:      --- /etc/rpi-issue: Raspberry Pi reference 2023-12-11
20231231-135034 DBG 2412:      --- Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2
20231231-135034 DBG 2413:  <-- isSupportedEnvironment 0
20231231-135034 DBG 8728:  --> check4RequiredCommands
20231231-135034 DBG 8749:  <-- check4RequiredCommands
20231231-135034 DBG 9565:  --- RESTORE: 1 - fileParameter: /mnt/raspiBackup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20231231-132319
20231231-135034 DBG 9586:  --- Enabling trap handler
20231231-135034 DBG 1943:  --> trapWithArg cleanup SIGINT SIGTERM EXIT
20231231-135034 DBG 1948:  <-- trapWithArg
20231231-135034 DBG 4758:  --> lockMe
20231231-135034 DBG 4760:  <-- lockMe
20231231-135034 MSG 9590:  --- RBK0009I: raspi01: raspiBackup.sh V0.6.9.1 - 2023-11-30 (36c728f) started at Sun 31 Dec 13:50:34 CET 2023.
20231231-135034 DBG 3877:  --> setupEnvironment
20231231-135034 DBG 3937:      --- BACKUPTARGET_DIR:
20231231-135034 DBG 3938:      --- BACKUPTARGET_FILE:
20231231-135034 DBG 3940:  <-- setupEnvironment
20231231-135034 MSG 9622:  --- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
20231231-135034 DBG 9623:  --- Read config /usr/local/etc/raspiBackup.conf : 0.1.7
20231231-135034 DBG 9623:  --- VERSION_CONFIG="0.1.7"
20231231-135034 DBG 9623:  --- DEFAULT_BACKUPPATH="/mnt/raspibackup"
20231231-135034 DBG 9623:  --- DEFAULT_BACKUPTYPE="rsync"
20231231-135034 DBG 9623:  --- DEFAULT_ZIP_BACKUP="0"
20231231-135034 DBG 9623:  --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20231231-135034 DBG 9623:  --- DEFAULT_KEEPBACKUPS="3"
20231231-135034 DBG 9623:  --- DEFAULT_KEEPBACKUPS_DD=0
20231231-135034 DBG 9623:  --- DEFAULT_KEEPBACKUPS_DDZ=0
20231231-135034 DBG 9623:  --- DEFAULT_KEEPBACKUPS_TAR=0
20231231-135034 DBG 9623:  --- DEFAULT_KEEPBACKUPS_TGZ=0
20231231-135034 DBG 9623:  --- DEFAULT_KEEPBACKUPS_RSYNC=0
20231231-135034 DBG 9623:  --- DEFAULT_SMART_RECYCLE="0"
20231231-135034 DBG 9623:  --- DEFAULT_SMART_RECYCLE_DRYRUN="1"
20231231-135034 DBG 9623:  --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"
20231231-135034 DBG 9623:  --- DEFAULT_STOPSERVICES="systemctl stop cron && systemctl stop NetworkManager && systemctl stop polkit"
20231231-135034 DBG 9623:  --- DEFAULT_STARTSERVICES="systemctl start polkit && systemctl start NetworkManager && systemctl start cron"
20231231-135034 DBG 9623:  --- DEFAULT_BEFORE_STOPSERVICES=""
20231231-135034 DBG 9623:  --- DEFAULT_AFTER_STARTSERVICES=""
20231231-135034 DBG 9623:  --- DEFAULT_EMAIL=""
20231231-135034 DBG 9623:  --- DEFAULT_SENDER_EMAIL=""
20231231-135034 DBG 9623:  --- DEFAULT_EMAIL_PARMS=""
20231231-135034 DBG 9623:  --- DEFAULT_MAIL_PROGRAM="mail"
20231231-135034 DBG 9623:  --- DEFAULT_TELEGRAM_TOKEN=""
20231231-135034 DBG 9623:  --- DEFAULT_TELEGRAM_CHATID=""
20231231-135034 DBG 9623:  --- DEFAULT_TELEGRAM_NOTIFICATIONS="F"
20231231-135034 DBG 9623:  --- DEFAULT_PUSHOVER_TOKEN=""
20231231-135034 DBG 9623:  --- DEFAULT_PUSHOVER_USER=""
20231231-135034 DBG 9623:  --- DEFAULT_PUSHOVER_NOTIFICATIONS="F"
20231231-135034 DBG 9623:  --- DEFAULT_PUSHOVER_SOUND_SUCCESS=""
20231231-135034 DBG 9623:  --- DEFAULT_PUSHOVER_SOUND_FAILURE=""
20231231-135034 DBG 9623:  --- DEFAULT_PUSHOVER_PRIORITY_SUCCESS="0"
20231231-135034 DBG 9623:  --- DEFAULT_PUSHOVER_PRIORITY_FAILURE="1"
20231231-135034 DBG 9623:  --- DEFAULT_SLACK_WEBHOOK_URL=""
20231231-135034 DBG 9623:  --- DEFAULT_SLACK_NOTIFICATIONS=""
20231231-135034 DBG 9623:  --- DEFAULT_NOTIFY_START=0
20231231-135034 DBG 9623:  --- DEFAULT_MAIL_ON_ERROR_ONLY=0
20231231-135034 DBG 9623:  --- DEFAULT_MSG_LEVEL="0"
20231231-135034 DBG 9623:  --- DEFAULT_COLORING="CM"
20231231-135034 DBG 9623:  --- DEFAULT_COLOR_CODES=("#FF8000 33" "#FF0000 31")
[... 300 lines cut ...]
20231231-135036 DBG 4642:                  --- Masquerading hostname
20231231-135036 DBG 4647:                  --- Masquerading sensitive non local IPs
20231231-135036 DBG 2123:                  --> logFinish
20231231-135036 DBG 2134:                      --- 108 2 0
20231231-135036 DBG 2139:                      --- LOG_OUTPUT=3
20231231-135036 DBG 2143:                      --- LOG_OUTPUT: 3
20231231-135036 DBG 2170:                      --- DEST_LOGFILE: /root/raspiBackup.logr
20231231-135036 DBG 2171:                      --- DEST_MSGFILE: /root/raspiBackup.msgr
20231231-135036 DBG 2174:                      --- Moving Logfile: /tmp/raspiBackup.logr
20231231-135036 DBG 2177:                      --- Logfiles used: /root/raspiBackup.logr and /tmp/raspiBackup.msgr
[... cut ...]
20231231-135036 DBG 4707:                  --> callNotificationExtension 108
20231231-135036 DBG 2230:                      --> callExtensions notify
20231231-135036 DBG 2302:                      <-- callExtensions
20231231-135036 DBG 4716:                      --- NotificationExtension rc: 0
20231231-135036 DBG 4721:                  <-- callNotificationExtension 0
20231231-135036 DBG 4901:              <-- cleanup
20231231-135036 DBG 4764:              --> unLockMe
20231231-135037 DBG 4766:              <-- unLockMe
root@raspi01:~#

hegien, 2023-12-31:
I'm sorry. Its the first time I post an issue in Github.

framps, 2023-12-31:
Please don't copy and paster the contents of the log but attach the debug log file as a file with drag and drop :wink: 

Please also attach the debug log of the backup run (as file via drag and drop please)

framps, 2023-12-31:
> I'm sorry. Its the first time I post an issue in Github.

No worries :smile:

hegien, 2023-12-31:
[raspibackup.zip](https://github.com/framps/raspiBackup/files/13801321/raspibackup.zip)

framps, 2023-12-31:
I found the root cause for your issue: You created a normal backup. According the debug log you have following files in the backup directory:
```
20231231-135036 DBG 7928:              *** ls -1 /mnt/raspiBackup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20231231-132319*
20231231-135036 DBG 7928:                  backup
20231231-135036 DBG 7928:                  bin
20231231-135036 DBG 7928:                  boot
20231231-135036 DBG 7928:                  dev
20231231-135036 DBG 7928:                  etc
20231231-135036 DBG 7928:                  home
20231231-135036 DBG 7928:                  lib
20231231-135036 DBG 7928:                  lost+found
20231231-135036 DBG 7928:                  media
20231231-135036 DBG 7928:                  mnt
20231231-135036 DBG 7928:                  opt
20231231-135036 DBG 7928:                  portainer
20231231-135036 DBG 7928:                  proc
20231231-135036 DBG 7928:                  @HOSTNAME@-backup.img
20231231-135036 DBG 7928:                  @HOSTNAME@-backup.mbr
20231231-135036 DBG 7928:                  @HOSTNAME@-backup.sfdisk
20231231-135036 DBG 7928:                  raspiBackup.log
20231231-135036 DBG 7928:                  raspiBackup.msg
20231231-135036 DBG 7928:                  root
20231231-135036 DBG 7928:                  run
20231231-135036 DBG 7928:                  sbin
20231231-135036 DBG 7928:                  sda1
20231231-135036 DBG 7928:                  srv
20231231-135036 DBG 7928:                  sys
20231231-135036 DBG 7928:                  tmp
20231231-135036 DBG 7928:                  usr
20231231-135036 DBG 7928:                  var
```

Because you have sda1 in your backup raspiBackup assumes it's a partition oriented backup and this backup mode requires a blkid file.

I have no clue why you called a directory sda1 in your root directory. Just rename this such it doesn't start with `sda1`, e.g.` disk_sda1`, create another backup and then the restore will succeed :wink:

hegien, 2023-12-31:
Thanks a lot for your very fast response and help. I have no idea, why this file was there. Probably created by accident. I've removed the file and started new backup. Will let you know if the restore works. 👍

hegien, 2024-01-01:
Restore and booting the raspi with the restored SD card worked without any problem.
Again thanks a lot for your support and this great tool.
Wish you a happy new year.
😀

closed 2024-01-01 by hegien