framps/raspiBackup #724: Backup to path within rootpartition?

heikoh81, 2024-01-06:
Hi,

is there an option to allow backup to a path within the root-partition?
At the moment, trying to do this results in an error.

I have several raspis booting from external USB SSDs with enough space (256 GB), no SD inserted.
raspibackup does not allow to save the backup to a path on the root partition of the SSD.
I can't see why this should be a problem?

I need an option to allow this, as some of my raspberries are remote with slow uplink.
From the local backup path, I will save the backups to my main backup system overnight later.
Saving to a mounted remote path is not an option, as of the slow uplink would take from 4-6 hours.

Greetings,
Heiko

framps, 2024-01-06:
> I can't see why this should be a problem?

This check was added because raspiBackup users by error forget to mount an external partition and thus create the backup on their root partition which often is too small to keep the full backup and the system stops working.

But there is an option available to disable this check :wink:. Option `-c` or set `DEFAULT_SKIPLOCALCHECK` to 1 in the config file.

This option was undocumented until now but I just documented this option on my website because there are more and more raspiBackup users who use big SSDs or USB disks and transfer the backup asynchronously to a remote system and they know what they are doing.

Keep in mind the primary purpose of raspiBackup is to create a backup as easy as possible even for unexperienced Linux users.

I saw your comment on my website. I will give a short answer there and link to this issue.

heikoh81, 2024-01-06:
Hi framps,

thanks for your very fast reply.

> This check was added because raspiBackup users by error forget to mount an external partition and thus create the backup on their root partition which often is too small to keep the full backup and the system stops working.

I understand now.
I myself started with a Raspberry 1 with a 16 GB SD-Card (not MicroSD, but indeed SD-Card that did not fit completely into the SD-Slot on the first Raspberry :-)

> This option was undocumented until now but I just documented this option on my website because there are more and more raspiBackup users who use big SSDs or USB disks and transfer the backup asynchronously to a remote system and they know what they are doing.

Thanks, That's what I need.
I was searching everywhere for an option but couldn't find it.

> I saw your comment on my website. I will give a short answer there and link to this issue.

Yes, I've been using raspibackup for many many years now and just discovered on your website that the project is now also on Github. That's why I opened an issue here. Thanks for linking.

I will close the issue now.

Greetings,
Heiko

closed 2024-01-06 by heikoh81

framps, 2024-01-06:
@heikoh81 Great to hear you're a faithful raspiBackup user :+1: 

Just one comment: My tool ist called **r**aspi**B**ackup, lowercase r and uppercase R :wink:

heikoh81, 2024-01-06:
> Just one comment: My tool ist called **r**aspi**B**ackup, lowercase **r** and uppercase **B** 😉

👍 Of course, as I should know from command line :-)