framps/raspiBackup #731: RaspiBackup not triggered (sheduled once a week)

lls-hcr, 2024-01-22:
Hi,

I love RaspbiBackup and have been using it for years on several Raspberry Pi, I never had any issue with it. I recently installed RaspbiBackup on a Raspberry Pi 5 with Debian Bookworm lite 64-bit OS, I have used the installer and have scheduled a backup to run once a week. If I launch the backup manually, it works well and I receive an email (msmtp) that notify me that the backup was successful. However, the automated backup is not triggered and no mail sent.

I don't know where I could have made a mistake since the configuration is set by the script through the UI. Could it be a permission issue?

Thanks

[raspiBackup.log](https://github.com/framps/raspiBackup/files/14015037/raspiBackup.log)

framps, 2024-01-22:
In the log I see raspiBackup terminates because of
```
??? RBK0039E: Mail program msmtp not installed to send emails.
```
You wrote when you start raspiBackup manually you get an eMail :thinking: 

Please check where msmtp is installed on your system with

```
which msmtp
```
raspiBackup adds following directories toithe PATH:

```
DEFAULT_PATHES="/usr/local/sbin /usr/local/bin /usr/sbin /usr/bin /sbin /bin"
```
Make sure msmtp is available in one of these directories.

lls-hcr, 2024-01-22:
`which msmtp` return /usr/bin/msmtp

framps, 2024-01-22:
That's fine.

Is the debug log you provided the log of a scheduled run? Given 
```
20240121-091428 DBG 6390:      --- Startingdirectory: /home/lls
```
it looks like the debug log was created when you started raspiBackup manually. I need a debug log from a scheduled run.

lls-hcr, 2024-01-22:
To launch manually: `sudo raspiBackup.sh -g`, below is the output on the email I receive.

--- RBK0009I: mizuki-serv: raspiBackup.sh V0.6.9.1 - 2024-01-10 (ce0fa3f) started at Mon 22 Jan 22:30:46 CET 2024.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0151I: Using backuppath /media/usb/backup with partition type ext4.
--- RBK0044I: Creating backup of boot partition in /media/usb/backup//mizuki-serv/mizuki-serv-rsync-backup-20240122-223046/mizuki-serv-backup.img.
--- RBK0158I: Creating native rsync backup "/media/usb/backup//mizuki-serv/mizuki-serv-rsync-backup-20240122-223046".
--- RBK0085I: Backup of type rsync started. Please be patient.
--- RBK0078I: Backup time: 00:01:25.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0159I: 3 backups kept for rsync backup type. Please be patient.
--- RBK0017I: Backup finished successfully.
--- RBK0010I: mizuki-serv: raspiBackup.sh V0.6.9.1 - 2024-01-10 (ce0fa3f) stopped at Mon 22 Jan 22:32:35 CET 2024 with rc 0.

framps, 2024-01-22:
If raspiBackup is started with systemd in the background it's running in a different environment. 

if you don't get any eMail notification and the scheduled raspiBackup run fails you will find the debug log in /root. Please provide this debug log.

lls-hcr, 2024-01-30:
Hi, I don't really know what happened, maybe I just messed up some configuration. Finally, I reinstalled RaspiBackup from scratch and now it seems to work well. Many thanks for the support.

closed 2024-01-30 by lls-hcr