framps/raspiBackup #733: FeatureRequest: Add json file

Moosbueffel, 2024-02-11:
A json file parallel to the .log and .msg would be desirable to facilitate the analyse the backup result  by a machine. This could look like this, for example:

{
  "error": {
    "status": false,
    "text": null
  },
  "warning": [
    {
    "code": "RBK1004E",
    "text": "bc nicht gefunden. bc muss installiert werden mit 'sudo apt-get install bc'." 
    },
    {
      "code": "RBK0021E",
      "text": "Backupprogramm des Typs tar beendete sich mit RC 1"
    }
  ],
  "type": "smart",
  "target": {
       "path": "/hdd/raspiBackup/",
       "file": "mySHS-rsync-backup-20240210-000001",
      "freeDiskspace": {
        "Unit": "GB",
        "before": 123,
        "after": 100
     }
  },
  "time": {
     "start": "11.02.2024 10:12:24",
     "end":   "11.02.2024 10:32:34",
     "duration": "00:20:10"
  },
    "userInfo": [
    1,
    "HD schläft nach 10 min",
    451
  ],
  "version": {
     "current": "V0.6.9.1 - 2023-11-30",
     "updateAvilable": false
  }
}

Whether the userInfo can be realized - and is necessary - I do not know. Just an idea, as it is possible to integrate your own routines.

framps, 2024-02-11:
I see your point. There are already folks who parse the message file and a JSON file will simplify the message processing.

In your example you also included messages from the sample extensions which complicates the processing because they are just samples and not supposed to be updated.

I'll think about your request :wink:

framps, 2024-02-27:
I created branch `json` and in there you find [this helper script](https://github.com/framps/raspiBackup/blob/json/helper/raspiBackupMsg2JSON.sh) which converts raspiBackup messages into JSON :wink:

framps, 2024-02-28:
code just merged into master