framps/raspiBackup #734: Restore a btrfs file systems fails

RolandTrenz, 2024-02-12:
I have a btrfs file system and back it up with raspibackup (Bookworm). It works well. But during a restore, an ext4 file system is created instead of the btrfs file system. However, "rootfstype=btrfs" is still entered in the file /boot/firmware/cmdline.txt, so booting the system is no longer possible.

framps, 2024-02-13:
That's definitely a bug. I should mention a system with btrfs is not included in the regression suite because it's a non standard setup. 

Please provide the debug log.

RolandTrenz, 2024-02-13:
Hallo „framps“.
vielen Dank für ihre Rückmeldung. Anbei die Logdateien für die Sicherung und die Rücksicherung.
Ja, sie haben Recht, das Btrfs ist kein Standard Setup. Es ist auch jetzt kein Problem für mich, ich kann mir ja entsprechend helfen. Wäre natürlich schön, wenn es automatisch funktioniert.
Ansonsten ist das raspiBackup ist wirklich super Programm zur Datensicherung!
Viele Grüße
Roland Trenz

￼￼￼￼

framps, 2024-02-13:
> Anbei die Logdateien für die Sicherung und die Rücksicherung.

Sorry. Ich sehe keine Logs. Du kannst sie mit dragndrop hier einfuegen :wink:

RolandTrenz, 2024-02-13:
? 

Dann versuche ich mal die gezippte Version

￼

framps, 2024-02-13:
Keine Ahnung wie Du versuchst hier die Logs anzuhaengen. Anbei wie es aussehen muss:

[sample.txt](https://github.com/framps/raspiBackup/files/14269277/sample.txt)

RolandTrenz, 2024-02-13:
Die Mails waren signiert. Dann mal ohne.

framps, 2024-02-13:
Ah ... Du nutzt eMail. Ich weiss nicht ob es per eMail geht eine Datei hier anzuhaengen. Ich melde mich immer direkt per Webbrowser im github an und da kann man per drag and drop die Dateien anhaengen.

RolandTrenz, 2024-02-13:
[logs.zip](https://github.com/framps/raspiBackup/files/14269476/logs.zip)

Dann sollte es jetzt gehen ...

framps, 2024-02-13:
Das sieht jetzt gut aus. Aber heute sehe ich da nicht mehr rein ...

framps, 2024-02-14:
raspiBackup unterstuetzt btrfs nur im partitionsorientierten Modus. Du machst ein Backup im normalen Modus wo immer ein RaspbianOS mit der Standardpartitionierung, d.h. mit ext4, genutzt wird.

Dummerweise wird im normalen Modus von raspiBackup nicht geprueft ob die Rootpartition ext4 formatiert ist und der Backup abgewiesen :-(

Stelle auf den partitionsorientierten Modus um und dann sollte alles mit btrfs klappen :-)

RolandTrenz, 2024-02-14:
OK, Danke.