framps/raspiBackup #737: Backup aborts to changed files

DirkLaurenz, 2024-02-16:
Backup (with tar) arborts when files change (especially log files like syslog, journal, etc) - could it be possible to ignore such changes on /var?

`--- RBK0009I: iob-master: raspiBackup.sh V0.6.9 - 2023-11-21 (6638571) Fr 16. Feb 07:21:19 CET 2024 gestartet.
--- RBK0151I: Backuppfad /autofs/backup/raspibackup mit Partitionstyp nfs wird benutzt.
--- RBK0044I: Backup der Bootpartition wird in /autofs/backup/raspibackup/iob-master/iob-master-tgz-backup-20240216-072117/iob-master-backup.img erstellt.
--- RBK0158I: tgz Backup "/autofs/backup/raspibackup/iob-master/iob-master-tgz-backup-20240216-072117/iob-master-tgz-backup-20240216-072117.tgz" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ tgz gestartet. Bitte Geduld.
tar: /var/log/syslog: file changed as we read it
tar: /var/lib/samba/winbindd_privileged/pipe: socket ignored
??? RBK0021E: Backupprogramm des Typs tgz beendete sich mit RC 1.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK1001I: Speicherauslastung vor dem Backup - Belegt: 121 MB Frei: 1660 MB
--- RBK1001I: Speicherauslastung nach dem Backup: Belegt: 126 MB Frei: 70 MB
--- RBK1000I: CPU Temperatur vor und nach dem Backup: 47.7'C - 52.5'C
--- RBK1001I: Partitionsauslastung vor dem Backup: Belegt: 7.22 TiB Frei: 3.21 TiB
--- RBK1002I: Partitionsauslastung nach dem Backup: Belegt: 7.23 TiB Frei: 3.20 TiB
--- RBK1003I: Partitionsauslastung Änderung freier Platz: -4.15 GiB (0,00 %%)
--- RBK0043I: Unvollständiges Backup in /autofs/backup/raspibackup/iob-master/iob-master-tgz-backup-20240216-072117 wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: iob-master: raspiBackup.sh V0.6.9 - 2023-11-21 (6638571) Fr 16. Feb 07:41:20 CET 2024 beendet mit Returncode 109.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.`

framps, 2024-02-16:
Please check option -u ;-)

closed 2024-02-22 by framps

DirkLaurenz, 2024-03-10:
I would like to configure excludes - case the filenames change.... (so exclude directories would be cool)

framps, 2024-03-10:
Please read [this](https://www.linux-tips-and-tricks.de/de/aufruf-und-optionen#parm_u) :wink: