framps/raspiBackup #738: raspiBackupRestore2Image.sh bricht mit Fehler ab. Fortschrittsanzeige fehlt bei Restore.

Jarves020, 2024-02-16:
Hallo,

das neue Script bricht beim Restore eines rsync Backups in die Imagedatei mit einem Fehler ab.

[image]

Vorhergehende Version des Scripts funktionierte einwandfrei.

Wenn ich folgende Zeilen aus der vorherigen Scriptversion vor dem Start des Restores in das Script einfüge, läufts durch:

>'# prime partitions
>
>sfdisk -uSL -f $LOOP < "$SFDISK_FILE"
>
>echo "===> Reloading new partition table"
>partprobe $LOOP
>udevadm settle
>sleep 3

Kann es sein das hier im aktuellen Script noch etwas fehlt?
Außerdem wäre die Fortschrittsanzeige beim Restore mit dem Parameter -g sehr Informativ...

framps, 2024-02-17:
Vielen Dan  fuer den Hinweis :+1: . 

Das habe ich eben gefixed. Mit der neuen Version kann man auch einen Restore auf eine weitere SD Karte oder USB Stick vornehmen lassen. Also eine cold standby SD karte erstellen. Default ist weiterhin ein dd File zu erstellen.

> Außerdem wäre die Fortschrittsanzeige beim Restore mit dem Parameter -g sehr Informativ...

Das Script ist ein unsupportetes Beispiel wie man raspiBackup noch nutzen kann. Du kannst es beliebig nach Deinen Beduerfnissen anpassen :wink:

framps, 2024-02-19:
@Jarves020 Könntest Du den Fix noch verifizieren? Ich bin gerade länger unterwegs ...

framps, 2024-02-22:
Sorry - man sollte keinen Code changen ohne ihn testen zu koennen. Ich war wie gesagt unterwegs und dachte das waere eine leichte Aenderung um auch den Restore direkt auf ein Device einzubauen. Wie Du gemerkt hast waren aber immer wieder kleine Fehler drin :cry: 

Jetzt habe ich den Fix sowohl fuer die Erstellung eines dds - die bisherige Funktionalitaet - erfolgreich getestet wie auch die neue Funktionalitaet - den Restore direkt auf ein Device.

Waere nett wenn Du den letzten Stand jetzt noch mal kurz antesten und verifizieren koenntest :wink:

Jarves020, 2024-02-24:
Imageerstellung und Pishrink läuft durch! :)
Restore vom .dd Image konnte ich allerdings noch nicht testen....

framps, 2024-02-25:
Perfekt. Danke. Wenn Du noch Lust und Zeit hast teste noch den Restore. Ist aber kein Muss

closed 2024-02-28 by framps