framps/raspiBackup #740: Add Spanish and Italian language

diegofrances, 2024-02-26:
Addition of Spanish and Italian language to the Software

framps, 2024-02-26:
Hello Diego. I'm glad you volunteer to add Spanish and Italian to raspiBackup :+1: 

As far as I can see there are already a lot of raspiBackup users in Italy and Italian language support will be highly appreciated. :wink: Same for Spanish :+1: 

I don't know whether you already know about [this page](https://www.linux-tips-and-tricks.de/de/raspibackupcategoried/603-local-language-support-for-languages-other-than-de-and-en-l10n) I created to describe what has to be done to add another language to raspiBackup. Please read this page and ask if there is something unclear for you.

I have some questions:

1) I noticed you just registered on github. The best way you provide the new language would be if you provide the code via github. That way you will be listed as a contributor to raspiBackup. But you have to learn to work with github. But you don't have to setup a whole local git environment. You can use the Webfrontend only. I'll help you whenever you have any questions about github. 
Or if you don't want to work with github you also can provide the code as an attachement in this ussue and I will handle the github stuff and integrate your changes. But then you're not listed as a controbutor in github.

2) Unfortunately the language stuff is not separated from the code. So my question is whether you have basic bash knowledge. This will make everything smoother. But if you don't have that knowledge don't worry. The just add the translated messages and I will handle the coding aspect.

Anyhow - I'm pleased to work with you to get two additional languages into raspiBackup and will help you as much as possible to get this done :smile:

framps, 2024-02-26:
I just created a [dedicated branch](https://github.com/framps/raspiBackup/tree/m_740_IT_ES) for the new languages and added the initial code in order to enable IT and ES :wink:

framps, 2024-03-04:
@diegofrances Anything I can help you?

diegofrances, 2024-03-05:
All good, just busy with work and no time for any project.

framps, 2024-03-05:
Ok. No hurry. Take your time ;-)

framps, 2024-06-01:
Looks like you're not interested on this feature any more or are swamped with other stuff - unfortunately :cry: 

Would be glad to have you to support Italian and Spanish. Again: If you have any problems with github of the code just let me know and I will help you. 

Otherwise this issue will be closed in 2 weeks now.

framps, 2024-06-13:
Hope you're not unemployed now :anguished:

diegofrances, 2024-06-14:
Not unemployed but medical sick leave, I have an herniated disc and I struggle to be on a chair for a long period so not working at the moment as I work for a Telecommunication Company, around 8-10 hrs on a chair and is not doable.

closed 2024-06-29 by github-actions[bot]

framps, 2024-10-08:
@diegofrances I just cleanup my branches and there exists a branch I created for you. As far as I see you either have no time or are no longer interested to add Spanish and Italian to raspiBackup.

I frankly appreciate if you still will add these languages. 

Pls let me know whether I can delete the dedicated branch I created for you and you will not work on this issue :cry:

diegofrances, 2024-10-10:
Hello,

Please keep it opened as I keep working on the translation. It might take some time as I just got married and I am expecting a baby but I will do it anyway.

Regards

framps, 2024-10-10:
> I just got married and I am expecting a baby

Congratulations :wedding: :baby: