framps/raspiBackup #746: RaspiBackup Hangs on 'du -s'

seaniedan, 2024-03-31:
1. Explain in detail what's going wrong

My backup hung (for over a day!) - I eventually realised (looking in /tmp/raspiBackup.logr ) that it was because it was trying to run 'df-h'

`20240331-215411 DBG 5742:                          *** df -h
`
After finding my active mounts: 
mount|grep mnt
and unmounting them with, e.g.
sudo umount -f /mnt/broken_server

..my restore suddenly started working! 

Please can you try  adding '-x' to the du command to stop it traversing mount points? 

https://stackoverflow.com/questions/37287012/linux-du-command-without-traversing-mounted-file-systems

I tried to look for the code to add a pull request but I couldn't find the command!

Many thanks for this excellent software, it just saved my bacon again!

:-D

framps, 2024-04-01:
You're now the third one who reports an issue with a command on a mounted device. I'm unfortunately not able to reproduce the issue. So I need help of the raspiBackup users to locate the root cause. 

As far as I can see it may be related to 

1. OS Bookworm running on the Raspberry
2. Mounted filesystem is mounted with SMB/CIFS 

Can you confirm you also run Bookworm and use a SMB/CIFS mounted device? If not please let me know what you use.

> Please can you try adding '-x' to the du command to stop it traversing mount points?

I use `df` and not `du` to list the free space in the debug log for debugging purposes. `df` has the option `-l` to process local filesystems only and will also make sure the mounted filesystems are not processed.

I actually want to know the free space on the backup partition for debugging purposes. But I see the point that any networking issues on the remote filesystems may get raspiBackup to hang. I just added option `-l` and you can grab the fix with `sudo raspiBackup -U -S`.

coinish, 2024-04-01:
Ah, I meant df, of course! 

I feel your -l addition will fix the issue, but I understand it's hard to replicate. The unresponsive server was mounted as a samba share.

framps, 2024-04-01:
>  The unresponsive server was mounted as a samba share.

OK. And which OS Release do your Raspberry have?

coinish, 2024-04-01:
Bullseye on client and server

framps, 2024-04-01:
Thank you. Looks like it's a different issue. The other guy was able to use SMB on Buster but got a hang on Bookworm. Or the issue is already permanent on Buster also. I have an eye on this issue.

closed 2024-04-03 by framps