framps/raspiBackup #748: Support to include other mounted partitions in the backup

framps, 2024-04-07:
As of now only the boot and root partition are saved. The partition oriented backup mode alloes to save more than these two partitions but they have to reside on the same device as the root partition.

It' quite common to have additional data on other partitions which also should be saved.

Right now tar and rsync uses option -X which causes them to stay on the same device. Either -X should be removed and all external mounted partitions excluded if they are not included in an option or the included partition will be saved in a following step.

framps, 2024-04-20:
## Backup
1. An additional option `DEFAULT_ADDITIONAL_PARTITIONS allows to define a list of mountpoints of partitions which will be saved. The backup type selected is used to save these partitions (tar or rsync). Backuptype dd doesn't support additional partitions.
2. The additional partition backups are stored in the backup in /raspiBackup/<mountpoint>

## Restore
1. The restore uses the same mountpoints as used in the backup. This is tested before the restore starts. If there is no partition mounted on any mountpoint the restore fails.

## Installer
1. An new configuration menu option C10 `Additional partitions` will be added which allows to configure the new configuration option

ckantor8, 2024-04-21:
Just to double-check, this means that I would be able to include my external hard drive mounted to /media in the full backup? Because that would be amazing, it's the only thing I'm missing from raspiBackup

framps, 2024-04-21:
> I would be able to include my external hard drive mounted to /media in the full backup

Yes :smile:

ckantor8, 2024-04-21:
That's incredible, can't wait!

ckantor8, 2024-05-14:
Any update on this??

framps, 2024-05-15:
You can check on the issues page which issues I'm working on right now 😉

ckantor8, 2024-06-19:
How's this coming along? Can't wait to finally backup all my external media along with my system

framps, 2024-06-19:
Keep in mind I work on raspiBackup in my spare time and not 8 hours every day. You don't speed up the implementation of this feature if you ask for any progress.

framps, 2024-08-21:
@codkan As I can see you are interested in this new feature :smiley: . 

This feature is now mostly implemented and tested and I consider the current code in alpha state. Would be great if you execute tests on your side and give feedback. Please let me know if you are willing to execute tests and then I will create a dedicated alpha branch for you.

framps, 2024-10-02:
Test effort is very high and at this point in time too high. Therefore moved into backog

closed 2024-10-02 by framps

framps, 2024-10-02:
Current code archived in d_748_externalParitions branch