framps/raspiBackup #751: No UUID updated in /boot/cmdline.txt & /etc/fstab

andyzle, 2024-04-23:
Using a fully new 32 GB USB 3.2 stick to restore from NAS to Raspberry 3 (/dev/sdb2) I got following error message:

> --- RBK0184I: Rootpartitionscheck gestartet.
> !!! RBK0208W: Es konnte keine UUID in /boot/cmdline.txt für root= erneuert werden. Das Backup könnte nicht starten.
> !!! RBK0208W: Es konnte keine UUID in /etc/fstab für / erneuert werden. Das Backup könnte nicht starten.
> !!! RBK0208W: Es konnte keine UUID in /etc/fstab für /boot erneuert werden. Das Backup könnte nicht starten.

Unfortunately with sudo blkid following output:

>  /dev/sda1: LABEL="BOOT" UUID="74C7-F23F" TYPE="vfat" PARTUUID="479560c9-01"
> /dev/sda2: UUID="756282ab-fa72-42b4-89c6-68520dd02704" TYPE="ext4" PARTUUID="479560c9-02"
> /dev/sdb1: LABEL="BOOT" UUID="74C7-F23F" TYPE="vfat"
> /dev/sdb2: UUID="9504312d-fbe4-4944-9db6-8b546e3adda3" TYPE="ext4"

=> sdb is the restore USB storage - I have no info about PARTUUID to manually adapt /boot/cmdline.txt and /etc/fstab.
Any ideas?

Environment:
Raspberry with raspbian 8 (jessie) 
raspiBackup.sh V0.6.6
no changes of version (OS, raspiBackup) since last successful restore.

framps, 2024-04-23:
You use an old release of raspiBackup. Current release is 0.6.9.1 :wink: Maybe an update makes sense :smile: 

Anyhow: Would you please provide the debug log of the restore?

andyzle, 2024-04-24:
Debug log: [20240423_raspiBackup.log](https://github.com/framps/raspiBackup/files/15097418/20240423_raspiBackup.log)

Regarding raspiBackup release: had no restore problems in the past with the current setup - but sure, if the new USB device is the root cause I would try the new release. I've avoided so far because I think I have to reconfigure some raspiBackup.conf parameters.

framps, 2024-04-24:
That's the contents of /etc/fstab in the backup:
```
20240423-180210 DBG 6959:              --- proc            /proc           proc    defaults          0       0
20240423-180210 DBG 6959:              --- /dev/sda1  /boot           vfat    defaults          0       2
20240423-180210 DBG 6959:              --- /dev/sda2  /               ext4    defaults,noatime  0       1
```
As you can see there is no UUID used but the old deprecated /dev/sd<n> way used in Jessie.

So there is no way to update the UUIDs because there don't exist any :wink: 

Just ignore the warning. 

> I've avoided so far because I think I have to reconfigure some raspiBackup.conf parameters.

Any raspiBackup upgrade doesn't modify any configuration options. New options will be added but that's it. 
If for some reasons you have to look up the old config options: You have a backup with the old raspiBackup release and there you will have all options before you upgraded :wink:

closed 2024-04-25 by andyzle

framps, 2024-04-27:
Fix merged into master and published

commit 536e4706 (2024-04-27): Fixed #751. No error message no longer written if /dev is used