framps/raspiBackup #754: Not working on Ubuntu 24.04 LTS - Raspberry Pi 4 and 5

DoWenzl94, 2024-05-05:
Hi there,

Thank you for your great application. However, I noticed that the tool isn't working on Ubuntu 24.04. I tried a fresh installation, but it gets stuck after this point:

matthias@Raspi5:~$ sudo raspiBackup
--- RBK0009I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Sun May  5 04:53:56 PM CEST 2024 gestartet.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
--- RBK0151I: Backuppfad /mnt/PiBackup mit Partitionstyp cifs wird benutzt.

On my NAS, the folder is created, but then nothing happens. This is the log:

20240505-165357 DBG 5743:              <-- logSystemDiskState
20240505-165357 DBG 3431:              --> executeBeforeStopServices
20240505-165357 DBG 3443:              <-- executeBeforeStopServices
20240505-165357 DBG 3409:              --> stopServices
20240505-165357 DBG 3416:                  --- systemctl stop apache2 && systemctl stop cron && systemctl stop mysql && systemctl stop polkit && systemctl stop sn>
20240505-165357 DBG 2601:                  --> executeShellCommand systemctl stop apache2 && systemctl stop cron && systemctl stop mysql && systemctl stop polkit >
20240505-165741 DBG 4770:                      --> cleanup

tried on RPi4 and 5

framps, 2024-05-05:
Would you please provide the debug log?

DoWenzl94, 2024-05-05:
Here it is, so there is no specific error: It gets stuck after it stops or tries to stop the services. Then, after some minutes, I cancel it with Ctrl+C.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/15213814/raspiBackup.log)

framps, 2024-05-05:
Thank you very much for the debug log. There I find

```
20240505-190116 DBG 3409:              --> stopServices 
20240505-190116 DBG 3416:                  --- systemctl stop apache2 && systemctl stop cron && systemctl stop mysql && systemctl stop polkit && systemctl stop snmpd && systemctl stop zabbix-agent && systemctl stop zabbix-server
20240505-190116 DBG 2601:                  --> executeShellCommand systemctl stop apache2 && systemctl stop cron && systemctl stop mysql && systemctl stop polkit && systemctl stop snmpd && systemctl stop zabbix-agent && systemctl stop zabbix-server
20240505-190409 DBG 4770:                      --> cleanup 
??? RBK0163E: Scriptausführung mit CTRL C abgebrochen.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
```
I can only guess. But according the debug log services are stopped and either you have to wait a bit longer (you waited 3 minutes only) or there is some service hanging and refusing to stop.

I suggest to wait some more time first. If it still hangs execute the stop commands step by step from left to right to identify the service which refuses to stop. Then you have to find out why the service doesn't stop with the stop command.

DoWenzl94, 2024-05-05:
Okay, I have now deactivated the stopping of services for testing purposes. However, I am encountering another problem, which I've experienced before. The backup process runs smoothly without any issues, but it gets completely stuck during the cleanup process. I've attached the log. Additionally, I cannot stop it with Ctrl+C, waited more than 20min.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/15214084/raspiBackup.log)

framps, 2024-05-05:
```
20240505-203453 DBG 4514:          --> cleanupBackupDirectory 
20240505-203453 DBG 4516:                  *** ls -la /mnt/PiBackup/Raspi5/Raspi5-tar-backup--20240505-203413
```

```
20240505-203453 DBG 5741:                          //192.168.1.70/PiBackup on /mnt/PiBackup type cifs (rw,relatime,vers=3.0,cache=strict,username=RaspiBackup,uid=1000,noforceuid,gid=1000,noforcegid,addr=192.168.1.70,file_mode=0770,dir_mode=0770,iocharset=utf8,soft,nounix,serverino,mapposix,rsize=4194304,wsize=4194304,bsize=1048576,retrans=1,echo_interval=60,actimeo=1,closetimeo=1,x-systemd.automount,x-systemd.idle-timeout=30)
```

I unfortunately cannot see from the debug log what system provides the backup partition :-( Please tell me more details about the system which provides the backup space.

You use systemd automount. I frankly never used and tested automounted backup partitions. Please try a static mount defintiion and use the dynamic mount option of raspiBackup if you don't want to have the backup partition mounted all the time.

DoWenzl94, 2024-05-05:
I've been experimenting a bit, and it behaves really, really weird:

I mount the drive with the following command:

//192.168.1.70/PiBackup mnt/PiBackup cifs username=xxx,password=xxx,rw,uid=1000,gid=1000,nounix,iocharset=utf8,file_mode=0770,dir_mode=0770,vers=3.0,noauto,x-systemd.automount,x-systemd.idle-timeout=30  0  0

Sometimes it works sporadically and then again not. Without making any changes, I've tried to make a backup several times now. Sometimes it works, sometimes it doesn't. The debug log always hangs at this line (attachment with _nw) when it doesn't work, but when it does work, it goes through (attachment with _w).

I absolutely can't see any regularity. For example, three times in a row, without making any changes, a backup was created. The first two times it worked without any problems, but the third time it gets stuck at that point and blocks:

matthias@Raspi5:~$ sudo raspiBackup
--- RBK0009I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Sun May  5 10:44:29 PM CEST 2024 gestartet.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
--- RBK0151I: Backuppfad /mnt/PiBackup mit Partitionstyp cifs wird benutzt.
--- RBK0044I: Backup der Bootpartition wird in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224428/Raspi5-backup.img erstellt.
--- RBK0158I: tar Backup "/mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224428/Raspi5-tar-backup-20240505-224428.tar" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
tar: /var/agentx/master: socket ignored
--- RBK0078I: Backupzeit: 00:00:39.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0159I: 5 Backups werden für den Backuptyp tar aufbewahrt. Bitte Geduld.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0010I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Sun May  5 10:45:15 PM CEST 2024 beendet mit Returncode 0.
--- RBK0026I: Debug Logdatei wurde in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224428/raspiBackup.log gesichert.
matthias@Raspi5:~$ sudo raspiBackup
--- RBK0009I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Sun May  5 10:46:01 PM CEST 2024 gestartet.
--- RBK0151I: Backuppfad /mnt/PiBackup mit Partitionstyp cifs wird benutzt.
--- RBK0044I: Backup der Bootpartition wird in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224600/Raspi5-backup.img erstellt.
--- RBK0158I: tar Backup "/mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224600/Raspi5-tar-backup-20240505-224600.tar" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
tar: /var/agentx/master: socket ignored
--- RBK0078I: Backupzeit: 00:00:35.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0159I: 5 Backups werden für den Backuptyp tar aufbewahrt. Bitte Geduld.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0010I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Sun May  5 10:46:42 PM CEST 2024 beendet mit Returncode 0.
--- RBK0026I: Debug Logdatei wurde in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224600/raspiBackup.log gesichert.
matthias@Raspi5:~$ sudo raspiBackup
--- RBK0009I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Sun May  5 10:46:49 PM CEST 2024 gestartet.
--- RBK0151I: Backuppfad /mnt/PiBackup mit Partitionstyp cifs wird benutzt.
--- RBK0044I: Backup der Bootpartition wird in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224648/Raspi5-backup.img erstellt.
--- RBK0158I: tar Backup "/mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240505-224648/Raspi5-tar-backup-20240505-224648.tar" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
tar: /var/agentx/master: socket ignored
--- RBK0078I: Backupzeit: 00:00:36.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.  -> at this Point only a hard Reset by power-supply helps

[raspiBackup_w.log](https://github.com/framps/raspiBackup/files/15214544/raspiBackup_w.log)

[raspiBackup-nw.log](https://github.com/framps/raspiBackup/files/15214545/raspiBackup-nw.log)

framps, 2024-05-06:
> I unfortunately cannot see from the debug log what system provides the backup partition :-( Please tell me more details about the system which provides the backup space.

Please answer my question :wink: Given you have German messages I assume you're from Germany. In Germany the AVM Fritzbox is commonly used. I already got twice issue because the AVM FB SMB implementation starts to hand when used ...

DoWenzl94, 2024-05-06:
Hello, no, I'm from Italy, but my native language is German - but that's another story. I'm using a Synology 923+ with the latest DSM. Everything worked smoothly with 22.04.

Now I've modified the fstab file to disable `noauto` and automatically mount the partition during boot. Unfortunately, it didn't succeed; it exhibits the same strange behavior. The first backup works, but when I try it again, it gets stuck during cleanup.

matthias@Raspi5:~$ sudo raspiBackup
--- RBK0009I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Mon May  6 10:20:14 AM CEST 2024 gestartet.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
--- RBK0151I: Backuppfad /mnt/PiBackup mit Partitionstyp cifs wird benutzt.
--- RBK0044I: Backup der Bootpartition wird in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240506-102013/Raspi5-backup.img erstellt.
--- RBK0158I: tar Backup "/mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240506-102013/Raspi5-tar-backup-20240506-102013.tar" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
tar: /var/agentx/master: socket ignored
--- RBK0078I: Backupzeit: 00:00:39.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0159I: 5 Backups werden für den Backuptyp tar aufbewahrt. Bitte Geduld.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0010I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Mon May  6 10:21:01 AM CEST 2024 beendet mit Returncode 0.
--- RBK0026I: Debug Logdatei wurde in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240506-102013/raspiBackup.log gesichert.
matthias@Raspi5:~$ sudo raspiBackup
--- RBK0009I: Raspi5: raspiBackup.sh V0.6.9.1 - 2024-04-27 (536e470) Mon May  6 10:21:05 AM CEST 2024 gestartet.
--- RBK0151I: Backuppfad /mnt/PiBackup mit Partitionstyp cifs wird benutzt.
--- RBK0044I: Backup der Bootpartition wird in /mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240506-102104/Raspi5-backup.img erstellt.
--- RBK0158I: tar Backup "/mnt/PiBackup/Raspi5/Raspi5-tar-backup-20240506-102104/Raspi5-tar-backup-20240506-102104.tar" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
tar: /var/agentx/master: socket ignored
--- RBK0078I: Backupzeit: 00:00:36.
--- RBK0033I: Bitte warten bis aufgeräumt wurde. -> and now it stucks.

DoWenzl94, 2024-05-06:
Hi, it must be an SMB/CIFS problem. I just tried mounting with NFS without any issues...

DoWenzl94, 2024-05-07:
Thank you very much, and sorry for false alarm.