framps/raspiBackup #755: Error trying to save configuration for scheduled backup

SJMarty, 2024-05-12:
Hello,

I have raspiBackup running on five Pis.  They each run a scheduled backup at 08:00 on a different day of the week (Monday - Friday).  I just spun up a sixth Pi and installed/configured raspiBackup.  After successfully running a manual backup (`sudo raspiBackup.sh -g`), I launched the install/uninstall/configuration interface (`sudo raspiBackupInstallUI`).  When I try to schedule this one to run on Saturday at 08:00, I get the following error messages when saving the configuration...

```
/usr/local/bin/raspiBackupInstallUI: Line 3313: printf: 08: invalid octal number
/usr/local/bin/raspiBackupInstallUI: Line 3336: ((: 08: value too great for base (error token is "08")
```

The errors flash up quickly so I actually had to capture a video to see the messages.

I checked for a log file and there is an empty raspiBackupInstallUI.log.

framps, 2024-05-12:
I was able to reproduces your issue. Would be helpful if you let me know which code level you use. Please show the output of `raspiBackupInstallUI -h`.

framps, 2024-05-12:
I just fixed the issue. Please grab the latest installer code with
```
sudo raspiBackupInstallUI -U
```
Then you should get following version information:
```
framp@majestix:~/home/framp$ raspiBackupInstallUI --version
raspiBackupInstallUI.sh 0.4.8.1, 2024-05-12/15:43:10 - 225f671
```
Please verify the fix and then close the issue if everything is OK now.

SJMarty, 2024-05-12:
I updated to the latest installer code and I don't see the error message now but unfortunately, it's still not scheduling the backup job.

I was running...

`raspiBackupInstallUI.sh 0.4.8.1, 2024-04-20/16:46:32 - 5d9729b`

Now I'm running...

`raspiBackupInstallUI.sh 0.4.8.1, 2024-05-12/15:43:10 - 225f671`

SJMarty, 2024-05-13:
I also noticed that on the working Pis, when I change the schedule, I'm prompted with the following...

"Do you want to save the updated cron settings for raspiBackup now?"

On the one that's not working, I get this prompt instead...

"Do you want to save the updated systemd settings for raspiBackup now?"

They are all running 64-bit Bullseye.

framps, 2024-05-13:
> On the one that's not working, I get this prompt instead...

Starting with raspiBackupInstallUI 0.4.8 systemd is used instead of crond to schedule the backup. Looks like your other Raspberries use raspiBackup already for a long time :wink: 

Please show the output of following commands:

```
sudo systemctl status raspiBackup.timer
sudo systemctl status raspiBackup.service
cat /etc/systemd/system/raspiBackup.timer
cat /etc/systemd/system/raspiBackup.service
```

SJMarty, 2024-05-13:
> Starting with raspiBackupInstallUI 0.4.8 systemd is used instead of crond to schedule the backup. Looks like your other Raspberries use raspiBackup already for a long time 😉

Definitely...for years!

> sudo systemctl status raspiBackup.timer

```
● raspiBackup.timer - Timer for raspiBackup.service to start backup
     Loaded: loaded (/etc/systemd/system/raspiBackup.timer; disabled; vendor preset: enabled)
     Active: inactive (dead)
    Trigger: n/a
   Triggers: ● raspiBackup.service

May 12 08:55:46 pi-aware systemd[1]: Stopped Timer for raspiBackup.service to start backup.
May 12 08:57:04 pi-aware systemd[1]: Started Timer for raspiBackup.service to start backup.
May 12 08:59:02 pi-aware systemd[1]: raspiBackup.timer: Succeeded.
May 12 08:59:02 pi-aware systemd[1]: Stopped Timer for raspiBackup.service to start backup.
May 12 08:59:54 pi-aware systemd[1]: Started Timer for raspiBackup.service to start backup.
May 12 18:19:00 pi-aware systemd[1]: raspiBackup.timer: Succeeded.
May 12 18:19:00 pi-aware systemd[1]: Stopped Timer for raspiBackup.service to start backup.
May 12 18:19:32 pi-aware systemd[1]: Started Timer for raspiBackup.service to start backup.
May 12 20:12:36 pi-aware systemd[1]: raspiBackup.timer: Succeeded.
May 12 20:12:36 pi-aware systemd[1]: Stopped Timer for raspiBackup.service to start backup.

```
> sudo systemctl status raspiBackup.service

```
● raspiBackup.service - Creation of a Raspberry backup with raspiBackup
     Loaded: loaded (/etc/systemd/system/raspiBackup.service; disabled; vendor preset: enabled)
     Active: inactive (dead)

```
> cat /etc/systemd/system/raspiBackup.timer
```

[Unit]
Description=Timer for raspiBackup.service to start backup

[Timer]
OnCalendar=*-*-* 00:00:42
# Create a backup once a week on Sunday morning at 5 am (default)
Unit=raspiBackup.service

[Install]
WantedBy=multi-user.target
```

> cat /etc/systemd/system/raspiBackup.service
```

[Unit]
Description=Creation of a Raspberry backup with raspiBackup

[Service]
Type=simple
ExecStart=/usr/local/bin/raspiBackup.sh
# For Use with Wrapper Script: ExecStart=/usr/local/bin/raspiBackupWrapper.sh
[Install]
WantedBy=multi-user.target
```

framps, 2024-05-14:
The raspiBackup timer was not activated for some unknown reasons. Please execute following commands to activate the timer:
```
sudo systemctl enable raspiBackup.timer
sudo systemctl start raspiBackup.timer
sudo systemctl status raspiBackup.timer
```
Last command should return 
```
pi@raspberrypi:~ $ systemctl status raspiBackup.timer
● raspiBackup.timer - Timer for raspiBackup.service to start backup
     Loaded: loaded (/etc/systemd/system/raspiBackup.timer; enabled; preset: en>
     Active: active (waiting) since Tue 2024-05-14 10:24:29 CEST; 5s ago
    Trigger: Sun 2024-05-19 08:00:42 CEST; 4 days left
   Triggers: ● raspiBackup.service

May 14 10:24:29 raspberrypi systemd[1]: Started raspiBackup.timer - Timer for raspiBackup.service to start backup.
```

> They are all running 64-bit Bullseye.

I suggest to use latest OS Bookworm if it's a new system that has been set up :wink: .

framps, 2024-05-14:
I just tested latest installer on a bullseye system and configured a schedule of 09:00 am to make sure it's not an issue on Bullseye. Everything works fine.

SJMarty, 2024-05-14:
> sudo systemctl status raspiBackup.timer

Now returns this...
```

● raspiBackup.timer - Timer for raspiBackup.service to start backup
     Loaded: loaded (/etc/systemd/system/raspiBackup.timer; enabled; vendor preset: enabled)
     Active: active (waiting) since Tue 2024-05-14 16:25:40 EDT; 1min 29s ago
    Trigger: Wed 2024-05-15 00:00:42 EDT; 7h left
   Triggers: ● raspiBackup.service

May 14 16:25:40 pi-aware systemd[1]: Started Timer for raspiBackup.service to start backup.
```
> I just tested latest installer on a bullseye system and configured a schedule of 09:00 am to make sure it's not an issue on Bullseye. Everything works fine.

It's still not creating a scheduled job when I run through the `sudo raspiBackupInstallUI` steps.

What are the recommended steps to completely uninstall and reinstall cleanly?

Thanks for investigating this.

framps, 2024-05-14:
> Trigger: Wed 2024-05-15 00:00:42 EDT; 7h left

Now the backup will be triggered in 7h and you should get a new backup. Everything should be OK now. You also should be able to update the time with the installer.

> What are the recommended steps to completely uninstall and reinstall cleanly?

I don't think it's neede because the next backup is schedule in 7 hours. But if you still want to uninstall raspiBackup and install it again (you will loose the existing raspiBackup configuration) that's the way to go:
```
sudo raspiBackupInstallUI -u
```
will uninstall raspiBackup from your system. Then use the documented curl call to install latest raspiBackup again and will include the fix for your 8 AM issue

framps, 2024-05-14:
> Thanks for investigating this.

You're welcome. You found a valid issue :wink:

SJMarty, 2024-05-18:
This is working on Pi #6 now.  :)

On my other five, I ran `sudo systemctl status raspiBackup.timer` and got this for output...

`Unit raspiBackup.timer could not be found.`

What should I do to get these set up correctly using systemd instead of crond?

framps, 2024-05-18:
Great it works now. 

It's not bad to use crond instead of systemd even systemd is the standard. There are folks who prefer to use crond.

But you can switch to use systemd instead of crond very easy. The installer has an option `-t` with parameters `crond` and `systemd`. Just invoke `sudo raspiBackupInstallUI -t systemd` and raspiBackup will use systemd. If you don't like systemd use parameter `crond` and you will use crond again. But please note the scheduling information is not migrated when you switch. You have to configure the schedule again.

SJMarty, 2024-05-18:
All six Pis using systemd for scheduling now.  I'm all set!

Thank you!

closed 2024-05-18 by SJMarty