framps/raspiBackup #759: failed backup

adriansergheev, 2024-05-22:
hi, and thanks for your work!
running `sudo raspiBackup` returns  on `Debian GNU/Linux 12 (bookworm)`:
```
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.9.1 - 2024-05-09 (5d6b026) started at Wed May 22 17:24:41 CEST 2024.
??? RBK0039E: Mail program mail not installed to send emails.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /mnt/ssd/archive/rpi-backup/raspberrypi/raspberrypi-rsync-backup-20240522-172440. This may take some time. Please be patient.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.9.1 - 2024-05-09 (5d6b026) stopped at Wed May 22 17:24:41 CEST 2024 with rc 124.
--- RBK0026I: Debug logfile saved in /home/sergheev/raspiBackup.log.
```
here's the log:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/15405587/raspiBackup.log)

framps, 2024-05-22:
According the error message you didn't install and configure an eMail Client. Therefore raspiBackup cannot send any eMail. Either remove the configured eMailAddress or install some eMailClient like exim4, nullmailer, msmtp, postfix et al.

adriansergheev, 2024-05-22:
sorry for the confusion, but i don't think it matters
i get this
```
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.9.1 - 2024-05-09 (5d6b026) started at Wed May 22 18:52:22 CEST 2024.
??? RBK0164E: Unable to create hardlinks on /mnt/ssd/archive/rpi-backup. RC 102.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /mnt/ssd/archive/rpi-backup/raspberrypi/raspberrypi-rsync-backup-20240522-185221. This may take some time. Please be patient.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.9.1 - 2024-05-09 (5d6b026) stopped at Wed May 22 18:52:22 CEST 2024 with rc 102.
--- RBK0026I: Debug logfile saved in /home/sergheev/raspiBackup.log.
```

[raspiBackup.log](https://github.com/framps/raspiBackup/files/15406453/raspiBackup.log)

with email disabled.

framps, 2024-05-22:
Ok. That's a different story. Please read [here](https://www.linux-tips-and-tricks.de/en/raspibackupmessagese#m0164).
According the log you use an exfat partition. This partition type does not support hardlinks which are required for an rsync backup
```
/dev/sda1 on /mnt/ssd type exfat (rw,noatime,fmask=0000,dmask=0000,allow_utime=0022,iocharset=utf8,errors=remount-ro)
```
Please check [this page](https://www.linux-tips-and-tricks.de/en/raspibackupcategorye/580-which-filesystem-can-be-used-on-the-backup-partition).

adriansergheev, 2024-05-22:
thank you, that worked!