framps/raspiBackup #764: Fehlercode: RBI0021E: Es kann nicht auf www.linux-tips-and-tricks.de zugegriffen werden. wget RC: 4

rogersennhauser, 2024-06-11:
Ausgangslage:
Ich habe einen RaspberryPI 3 als Testumgebung und einen RaspberryPI 5 als Hauptumgebung. 
Nachdem ich Deine Dokumentation auf der Homepage gelesen und die Videos auf Deinem YouTube Kanal studiert hab, 
wagte ich mich an die Installation an meinem RaspberryPI 3. Das ging alles wie beschrieben und auch das Einrichten des Testbackups war erschreckend einfach. Die Test-Backups laufen echt großartig!

Raspberry 3: (Testumgebung)
System mit Hersteller Image neu aufgesetzt und ein USB-Stick (ext4) Mappings in der Vorbereitung. 
Es sind alle Updates gemacht und das System ist auf dem neusten Stand. (apt update/apt upgrade)

Raspberry 5: (Hauptumgebung)
System mit Hersteller Image vor ca. 2 Monaten neu in Betrieb genommen. Hier läuft Docker-Compose + Portainer in der neusten Version. In den Containern sind ein PiHole und ein paar kleine Projekte. Es sind alle Updates gemacht und das System ist auf dem neusten Stand. (apt update/apt upgrade)

Natürlich habe ich das PiHole testhalber ausgeschalten. Obwohl ich die erfolgreiche «raspiBackup» Installation auf dem Raspberry 3 mit aktivem PiHole gemacht hab und der Raspberry 5 eine Fixe IP-Adresse hat und da der DNS fix auf 1.1.1.1 zeigt.

Nun wollte ich das echt geniale Tool «raspiBackup» auf meinem Raspberry 5 installieren. 
Und scheitere bereits beim zweiten Menüpunkt in der Installation. 
Ich ernte den Fehlercode: «RBI0021E: Es kann nicht auf www.linux-tips-and-tricks.de zugegriffen werden. wget RC: 4»

Da zum einen der Raspberry 3 und der Raspberry 5 am gleichen Switch hängen und ich ansonsten auf dem Raspberry 5 alles problemlos aus den Internet installieren kann. Schliesse ich aus das es sich um ein Internetproblem handelt.

Auch habe ich mich mal nach dem Fehlercode: RBI0021E / wget RC: 4 umgesehen. Und die 3 von mir gefundenen Support Diskussionen scheinen unterschiedliche Wege zu verfolgen. Am besten gefiel mir der Ansatz das sich das Problem von selbst behebt. ;(Ist bei mir leider nicht so) 

Vielleicht noch als Warnung (Info) Ich bin Linux Neuling! Arbeite seit 30 Jahren in der Netzwerksupport unter Windows. Und ich bin, falls Du das anbietest auch einer Fernwartung auf meinen Windows Rechner nicht abgeneigt. (Natürlich gegen Bezahlung)
Die RaspberryPI sind via PuTTY und SSH erreichbar. 

Ich hoffe Du kannst mir weiter helfen denn ich finde das «raspiBackup» ein tolles Programm ist.

Liebe Grüsse aus der Schweiz
Roger

PS: Habe Dir mal mit etwas «Kaffee Geld» geworfen. Ich hoffe Du verbrauchst nicht alles beim Support an meinem Problem.

framps, 2024-06-13:
Moin Roger,

vielen Dank fuer Dein Trinkgeld :+1: 

Mir ist jetzt nicht ganz klar wo bei Dir raspiBackup laeuft. Auf dem System auf dem Docker laeuft oder im Dockercontainer?

Der einfachste Test ob das Netzwerk tut ist

```
ping www.linux-tips-and-tricks.de
```

Wenn der erfolgreich ist fuehre mal folgendes aus:

Der Test der von raspiBackup vorgenommen wird ist

```
wget -q --spider -t 1 -T 3 www.linux-tips-and-tricks.de
```
Lass mal das `-q` weg und zeige was fuer eine Ausgabe Du erhaelst. 

Wenn allerdings beim Ping schon ein Fehler kommt ist das ein allgemeines Linuxsetup- oder -configproblem. U.U. auch ein PiHole Configproblem im Zusammenhang mit Docker. Zu denen gebe ich keinen Support denn dann waere ich 7/24 beschaeftigt. Ich hoffe Du verstehst das :wink: . In dem Falle empfehle ich Dir Dein Problem im [deutschen Raspberryforum](https://forum-raspberrypi.de/forum/) anzusprechen. Dort wirst Du sicherlich Hilfe bekommen :wink: 

Cu framp

rogersennhauser, 2024-06-13:
Hallo framp,

Herzliche Dank für Dein Feedback und die Fragen. Ich möchte raspiBackup direkt auf dem System auf dem Docke liegt nutzen. So das ich dann Backups (dd) auf meinen USB Stick am Raspberry Pi machen kann.

Gerne habe ich Dir hier meine Ergebnisse:
**ping www.linux-tips-and-tricks.de**
[image]

**wget -q --spider -t 1 -T 3 www.linux-tips-and-tricks.de (mit -p)**
[image]

**wget --spider -t 1 -T 3 www.linux-tips-and-tricks.de (ohne -p)**
[image]

Hoffe das Du damit etwas anfangen kannst. Der Ping scheint zu klappen und das Ergebnis mit dem Test ohne -P lässt Fragen offen.

Sonnigen Gruss Roger

framps, 2024-06-13:
> Ich möchte raspiBackup direkt auf dem System auf dem Docke liegt nutzen.

Also nicht im Docker Container. Das reduziert die Komplexitaet.

> So das ich dann Backups (dd) auf meinen USB Stick am Raspberry Pi machen kann.

Ein dd Backup ist sehr ineffizient. Du solltest lieber rsync bei einem lokal angeschlossenen USB Stick nutzen.

> Hoffe das Du damit etwas anfangen kannst. Der Ping scheint zu klappen und das Ergebnis mit dem Test ohne -P lässt Fragen offen.

Jein - der super dupa Netzwerkexperte bin ich nicht. Es ist schon merkwuerdig dass die Namensaufloesung beim ping funktioniert und beim wget nicht :thinking: Das sollte eigentlich entweder in beiden Faellen nicht funktionieren oder in beiden Fallen eben nicht :thinking:

Ich kann Dir jetzt helfen raspiBackup auch ohne Netzwerkverbindung zu installieren - aber dann hast Du natuerlich auch keine Benachrichtigungsfunktion per eMail oder den anderen Notificationmoeglichkeiten.

rogersennhauser, 2024-06-14:
Guten Morgen framp,

Das mit dem raspiBackup ohne Netzwerkverbindung liest sich gut. Ich würde die Benachrichtigungsfunktion per eMail oder die anderen Notificationmoeglichkeiten eh nicht nutzen. Wie gehen wir vor?

framps, 2024-06-14:
Auf [dieser Seite](https://www.linux-tips-and-tricks.de/de/raspibackupcategorie/538-raspibackup-manuelle-installation/) sind jetzt die notwendigen Schritte beschrieben :wink: 

PS: Wenn Du keine Benachrichtigung konfigurierst solltest Du aber irgendwie in Intervallen kontrollieren ob die Backups erstellt werden. raspiBackup laeuft zwar sehr zuverlaessig aber ich kann nicht 100% ausschliessen dass doch mal irgendwas schieflaeuft und Du in dem Moment wo Du ein Backup benoetigst feststellst dass keine Backups existieren :scream:

rogersennhauser, 2024-06-15:
Guten Morgen framp,

Danke für die Anleitung der Manuellen Installation. Ich habe das mal so Schritt für Schritt abgearbeitet - nur will das nicht so richtig. Ich zeige Dir mal was ich gemacht hab und was ich noch so mitbekommen hab. Vielleicht bringt das etwas Licht ins dunkle...

Zur raspiBackup - Manuelle Installation und Konfiguration:

Punkt 1/3 - Hier scheitere ich bereits beim ersten Befehl:
[image]

Punkt 2/3 - Habe ich ohne Fehlermeldung Schritt für Schritt abgearbeitet:
[image]

Punkt 3/3 - Hier will raspiBackup nicht wirklich starten:
[image]

PS: Ich hab dann nochmal mal so naiv wie ich bin mit dem Befehl: "curl -o install -L https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh; sudo bash ./install" versucht die Installation anzustossen. Was schlussendlich wieder zur Fehlermeldung: "RBI0021E: Es kann nicht auf www.linux-tips-and-tricks.de zugegriffen werden. wget RC: 4" führt. Bekomme aber beim mussglückten Installationsversuch kurz diese Meldung. (Bild unten) Vielleicht gibt Dir das einen hinweis was hier falsch läuft.
[image]

Danke für Deinen Einsatz und Gruss
Roger

framps, 2024-06-15:
Punkt 1 setzt eine Netzverbindung voraus :wink: 

Du hast den zweiten `ln` Befehl nicht ganz korrekt eingegeben. Deshalb wird der Installer nicht gefunden.

rogersennhauser, 2024-06-15:
Sali Framp,

Sorry - aber ich stehe etwas auf dem Schlauch! 😉

Du meinst mit "den zweiten ln Befehl" in der Anleitung: "raspiBackup - Manuelle Installation und Konfiguration" unter Punkt: 2/3 den Befehl: "sudo ln -s /usr/local/bin/raspiBackupInstallUI.sh /usr/local/bin/raspiBackupInstallU"?
_(Ich arbeite mit copy paste)_
[image]

framps, 2024-06-15:
Oha ... Sorry. Ich habe auf meiner Seite einen Typo und ein grosses I am Ende vergessen zu schreiben 😢 

Ist jetzt korrigiert.

rogersennhauser, 2024-06-15:
UPS! 😉

Ich hab den Befehl "sudo ln -s /usr/local/bin/raspiBackupInstallUI.sh /usr/local/bin/raspiBackupInstallUI" jetzt nachträglich nochmals ausgeführt. _(das hat geklappt - keine Fehlermeldung)_

Das sieht dann im Verzeichnis so aus. Aber der Befehl: "sudo raspiBackupInstallUI" will immer noch nicht...
[image]

framps, 2024-06-15:
Hm ... komisch. Zeige doch mal die Ausgabe von `ls -la /usr/local/bin`.

rogersennhauser, 2024-06-15:
[image]

framps, 2024-06-15:
Oha. Sorry. Da war noch ein kleiner Fehler in der Beschreibung :cry: . Fuehre die folgenden Befehle aus um den Fehler zu korrigieren:
```
sudo chmod 755 /usr/local/bin/raspiBackup*.sh
sudo chmod 600 /usr/local/etc/raspiBackup*.conf
```

Dann sollte es funktionieren. Falls wider Erwarten immer noch nicht zeige noch mal `ls -la /usr/local/bin` und `ls -la /usr/locacl/etc`.

Es waere uebrigens besser Du laedst keinen Screenshot hier hoch sondern kopierst die Befehle und die Ausgaben mit Copy&Paste hier rein und schreibst davor und danach in einer separaten Zeile drei Backkticks ``` . Das hat den Vorteil dass Leute die im github suchen diesen Issue finden :wink:

framps, 2024-06-15:
Kannst Du mal das Debuglog raspiBackupInstallUI.log welches sich im aktuellen Verzeichnis befindet hier hochladen? Einfach per drag and drop als Datei.

rogersennhauser, 2024-06-15:
Hier die gewünschte Datei: 😢
[raspiBackupInstallUI.log](https://github.com/user-attachments/files/15847635/raspiBackupInstallUI.log)

rogersennhauser, 2024-06-15:
Hab gerade gesehen das die Datei auf meinem Windows Kübel 0KB gross ist. Ist das normal? 🤔

framps, 2024-06-15:
Nein. Ist es nicht. Ich habe am 20. April eine kleine Aenderung bzgl des Logfiles im Installer gemacht und dummerweise durch einen Typo dabei die Logdatei gecleared :cry:  

Kurzum: Das habe ich eben gefixed. Du musst Dir also den Installer noch einmal runterladen und rueberkopieren. Sorry for the inconvenience. Du bist seit dem fehlerhaften Fix der Erste der mir das Log zeigen soll und hast somit den Fehler gefunden.

rogersennhauser, 2024-06-17:
Hallo Framp,

Wünsche einen gute Wochenstart! 😢 
Ich war heute Mittag fleissig und habe Dir nun die besagte Log Datei:
[raspiBackupInstallUI.log](https://github.com/user-attachments/files/15865721/raspiBackupInstallUI.log)
Gruss Roger

rogersennhauser, 2024-06-17:
Also unser eins versteht nur Bahnhof! 🤔 Aber hier die gewünschte Datei:
[raspiBackup.zip](https://github.com/user-attachments/files/15874615/raspiBackup.zip)
_(ich musste Dir das Ding Zipen)_

framps, 2024-06-17:
> Also unser eins versteht nur Bahnhof!

Dafuer verstehe ich worum es geht :laughing: 

Die Configdatei ist falsch. Das liegt daran dass ich in der Beschreibung noch einen winzigen Typo drin hatte :cry: Ist auf der Webseite schon korrigiert.

Mit `curl -sSL https://www.linux-tips-and-tricks.de/raspiBackup_de.conf > raspiBackup.conf` laedst Du die korrekte Datei runter und musst sie wieder rueberkopieren und die richtigen Rechte setzen.

Ist alles etwas langwierig geworden aber das liegt daran dass Du das Guineapig bist da bislang noch niemand raspiBackup auf einem System ohne Netzwerkzugriff installieren wollte.

rogersennhauser, 2024-06-17:
Hey Framp, 

Ich habe das mal so neu durchlaufen lassen und die Fehlermeldung ist weg! Danke!
Werde mal ein Backup für heute Nacht konfigurieren und Dir morgen berichten ob es so läuft. 😉
Nochmals herzlichen Dank für Deinen Einsatz!

Gruss Roger

framps, 2024-06-17:
> Ich habe das mal so neu durchlaufen lassen und die Fehlermeldung ist weg!

War gut dass wir nach der Ursache der Fehlermeldung gesehen haben. Ansonsten waere wohl auch raspiBackup mit irgendwelchen Fehlern abgebrochen.

Wenn Du den Installer nimmst um raspiBackup zu installieren passt alles. Wenn irgendwas manuell gemacht wird gibt es eben alle moeglichen Fehlerquellen ...

rogersennhauser, 2024-06-19:
Hallo Framp,

Hier meine versprochene Rückmeldung. Die Backup App ist prinzipiell am laufen. Das mit den Backups muss ich noch testen. 
Zwei Fragen haben sich aber schon ergeben:

**Frage 1:** Wenn ich ein Backup Manuell mit dem Befehl: "sudo raspiBackup" starte beginnt das Backup mit dem Vermerk: _Backuperstellung vom Typ dd gestartet. Bitte Geduld._ Da kommt dann keine Meldung mehr - hab auch schon mehrere Stunden die SSH Sitzung mit PuTTY offen gelassen. Ist das normal?

**Frage 2:** Nach dem dd Backup habe ich auf dem 256GB USB Stick (NTFS) einen Ordner: RaspberryPI und in diesem Ordner einen Unterordner mit einer img-Datei (z.B: RaspberryPI-dd-backup-20240619-040044.img) Die Grösse der img-Datei fällt in etwa mit 10GB aus. Die zu sichernde Mini-SSD hat 128GB. _(ohne Backup Komprimierung)_ Muss bei einem dd Backup die Datei nicht gleichgross wie die Quell SSD sein?

Lieben Gruss Roger

framps, 2024-06-19:
> Ist das normal?

raspiBackup schreibt immer am Ende eine Meldung dass der Backup erfolgreich oder nicht erfolgreich erstellt wurde.

> Muss bei einem dd Backup die Datei nicht gleichgross wie die Quell SSD sein?

Ja. Wenn das nicht so ist ist der Backup (noch) nicht durchgelaufen bzw abgebrochen.

Generell ist es eine schlechte Idee den Backupspace mit NTFS formatiert zu haben. Linux kann zwar NTFS aber die Performance ist nicht optimal. D.h. ich empfehle ext4 zu nhemen Ausserdem empfehle ich rsync zu nehmen statt dd. Das sind immer Deltasoicherungen, belegen weniger Platz und sind schneller erstellt.

rogersennhauser, 2024-06-19:
Hallo Framp,

Vielleicht sollte ich mal etwas Licht ins Dunkle bringen. Auf Meinem Raspberry PI OS läuft Docker und Portainer. 
Die darauf laufenden Docker Container Daten (Volumes) werden mit Duplicati Backup (im Container) täglich auf ein NAS im Netzwerk gesichert.

Jetzt hätte ich gerne noch eine Software, die mir 1x Wöchentlich und bei bedarf Manuell ein Image des kompletten Systems macht. So das wenn ich am System Installationen mache und etwas schief geht - zurück zum Ausgangspunkt kann. Und hier kommt RaspiBackup mit dd zum Zug. 😉

Bis anhin fahre ich das zu sichernde System herunter und ziehe mit «Win23Diskmanager» von der 128GB Mini SSD eine IMG Datei. Diese aktuelle Datei spiele ich dann auf eine zweite 128GB Mini SSD und boote das System mit dieser.
_(Platz für solche IMG Dateien habe ich mit 12TB genug und Zeit spielt auch keine Rolle kann bis 6 Stunden dauern)_

Da das aber umständlich ist und jeweils 60 Minuten Mein komplettes Netzwerk lahmlegt. (PiHole DNS + DHCP laufen im Docker) Suche ich nach einer Möglichkeit das im Betreib zu machen. Mir gefällt am «RaspiBackup» das es den Docker Service beim Backup aus und wider einstellt und das es in Deutsch ist.

Sonnige Grüsse Roger

PS: Ich habe noch bei keinem Testbackup von raspiBackup am Ende eine Meldung erhalten. Und die SSH Verbindung bei Manuellem Backup Start über Nacht (9 Stunden) offen gelassen. 😢

framps, 2024-06-19:
> Ich habe noch bei keinem Testbackup von raspiBackup am Ende eine Meldung erhalten.

Du sicherst 128GB per DD und das dauert natuerlich. Wenn Du keine Erfolgsmeldung von raspiBackup erhaelst stimmt irgendwas nicht bei Dir.

> Und die SSH Verbindung bei Manuellem Backup Start über Nacht (9 Stunden) offen gelassen. 😢

Da gibt es ein tool `screen` mit dem Du offline Tasks ausfuehren lassen und wieder reconnecten kannst.

Ich selbst nutze Docker nicht aber ich weiss dass divere Leute raspiBackup mit Docker erfolgreich nutzen um ihr gesamtes System zu sichern. Wo und warum es bei Dir klemmt kann ich Dir leider nicht sagen. Das laesst sich nicht aus den Debuglogs ersehen.

closed 2024-07-04 by github-actions[bot]