framps/raspiBackup #765: Restore bricht mit Fehler " ??? RBK0137E: Rootbackupdatei für den Typ tgz nicht gefunden" ab

RolandTrenz, 2024-06-17:
Ich sichere bisher mein BTRFS Filesystem im Partionsoriontierten Backup. Das Restore bricht jedoch mit der Fehlermeldung " ??? RBK0137E: Rootbackupdatei für den Typ tgz nicht gefunden" ab. Anbei die Logdateien der Sicherung und Restore. 
[raspibackup-logs.zip](https://github.com/user-attachments/files/15866179/raspibackup-logs.zip)

framps, 2024-06-17:
Der Backup brach schon ab wg einer falschen Option :wink: 

```
??? RBK0089E: Unbekannte Option -T*.
--- RBK0091I: 'raspiBackup.sh -h' liefert eine detailierte Beschreibung aller Scriptaufrufoptionen.
```

RolandTrenz, 2024-06-17:
Sorry, war das falsche Logfile. Hier das richtige.
[raspiBackup.log](https://github.com/user-attachments/files/15874724/raspiBackup.log)

framps, 2024-06-17:
Das sieht gut aus. Jetzt brauche ich das restore log :wink:

framps, 2024-06-17:
Die Erkennung von einem partitionsorientierten Backup funktionierte bei einem tar oder tgz Backup leider nicht. Das habe ich eben gefixed. 

Teste bitte den Fix - einfach den Restore ausfuehren. Den Hotfix kannst Du Du mit  `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_765` runterladen und anschliessend ausfuehren. Aber Achtung! Wichtig ist der Punkt am Anfang :wink:

RolandTrenz, 2024-06-17:
Einen Schritt weiter. Diesmal bricht er mit der Fehlermeldung ab, das das Zielmedium zu klein ist. Bisher hat er das doch immer automatisch angepasst ? Der Parameter --resizeRootFS brachte auch nichts.
[raspiBackup.logr.zip](https://github.com/user-attachments/files/15875522/raspiBackup.logr.zip)

framps, 2024-06-17:
Fix rebased into master

closed 2024-06-17 by framps