framps/raspiBackup #768: backup.img RC 1 wenn Backup-Storage leer

bcutter, 2024-06-28:
Ich sichere per tar nach "/mnt/localbackup", einen kleinen USB-Stick. Von dort wird nächtlich per rsync weg gesichert. raspiBackup endet bei und nutzt also per `raspiBackup.conf` den `DEFAULT_BACKUPPATH="/mnt/localbackup/hostname_raspiBackup"`.

Da der USB-Stick zu klein bzw. die tar-Backups zu groß wurden, habe ich den Backup-Storage mit einem pre-Skript `raspiBackup_ClearLocalBackupStorage_pre.sh` einfach geräumt:
`sudo rm -R /mnt/localbackup/hostname_raspiBackup/hostname/*`

Seitdem scheitert raspiBackup an der Erzeugung des dd-Backups für die Boot-Partition:

```

06-28-2024 21:25:29 --- RBK0009I: hostname: raspiBackup.sh V0.6.9 - 2023-11-20 (8cb71a8) Fr 28. Jun 21:25:29 CEST 2024 gestartet.
06-28-2024 21:25:30 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
06-28-2024 21:25:34 --- RBK0151I: Backuppfad /mnt/localbackup/hostname_raspiBackup mit Partitionstyp ext4 wird benutzt.
06-28-2024 21:25:35 --- RBK0008I: Services werden gestoppt: 'service cron stop; service webmin stop'.
06-28-2024 21:25:45 --- RBK0267I: Erweiterung raspiBackup_ClearLocalBackupStorage_pre.sh wird aufgerufen.
06-28-2024 21:25:48 --- RBK0081I: Backup vom Typ tar wird in /mnt/localbackup/hostname_raspiBackup/hostname/hostname-tar-backup-20240628-212526 erstellt.
06-28-2024 21:25:49 --- RBK0044I: Backup der Bootpartition wird in /mnt/localbackup/hostname_raspiBackup/hostname/hostname-tar-backup-20240628-212526/hostname-backup.img erstellt.
dd: failed to open '/mnt/localbackup/hostname_raspiBackup/hostname/hostname-tar-backup-20240628-212526/hostname-backup.img': No such file or directory
06-28-2024 21:25:49 ??? RBK0178E: Erzeugung von /mnt/localbackup/hostname_raspiBackup/hostname/hostname-tar-backup-20240628-212526/hostname-backup.img Datei endet fehlerhaft mit RC 1. Normalerweise ist die SD Karte fehlerhaft. Option -B kann helfen.
06-28-2024 21:25:49 --- RBK0033I: Bitte warten bis aufgeräumt wurde.
06-28-2024 21:25:49 --- RBK0267I: Erweiterung raspiBackup_PushingBox_NotifyBackup_post.sh wird aufgerufen.
06-28-2024 21:25:50 --- RBK0007I: Services werden gestartet: 'service webmin start; service cron start'.
06-28-2024 21:26:18 ??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
06-28-2024 21:26:18 --- RBK0010I: hostname: raspiBackup.sh V0.6.9 - 2023-11-20 (8cb71a8) Fr 28. Jun 21:26:18 CEST 2024 beendet mit Returncode 114.
06-28-2024 21:26:19 --- RBK0167I: Eine eMail wird versendet.
```

Das geschieht auch beim 2. und 3. und .... Versuch raspiBackup auszuführen, also wenn der Ziel-Ordner `/mnt/localbackup/hostname_raspiBackup` bereits leer ist.

**- Liegt es an der Logik von raspiBackup mit den pre-Skripten?
- Oder kann raspiBackup nicht mit einem leeren Backup-Storage umgehen?**

Irgendwie bin ich hier scheinbar auf einen Fehler gestoßen.

Zusatz-Info: SD-Karte (`/dev/mmcblk0p1`) ist OK, habe es gem. https://github.com/framps/raspiBackup/issues/296#issuecomment-752687674 getestet und dd funktioniert ohne Probleme.

bcutter, 2024-06-28:
Nachtrag: Es liegt am Timing. Ich vermute, dass raspiBackup in etwas so arbeitet:

1) Starten
2) Backup-Ordner am Backup-Storage anlegen
3) dd von boot etc.

Jetzt komm ich mit dem pre-Skript:

1) Starten
2) Backup-Ordner am Backup-Storage anlegen
3) Pre-Skript: Alles am Backup-Storage löschen (inkl. Ordner von Schritt 2)
4) dd von boot etc. --> Crash, da Ordner nicht mehr verfügbar

Genau das sagt auch

`dd: failed to open '/mnt/localbackup/hostname_raspiBackup/hostname/hostname-tar-backup-20240628-212526/hostname-backup.img': No such file or directory`

Ich dachte wirklich, die pre-Skripte laufen GANZ AM ANFANG und die post-Skripte GANZ AM ENDE.

Wie sollte ich das pre-Skript anpassen um nicht mit raspiBackup in Konflikt zu geraten? Ggf. über ein "nur alles löschen was älter ist als 1 Minute" - wenn ich nur den richtigen rm-Filter dafür wüsste...

framps, 2024-06-28:
> +++ Ich dachte wirklich, die pre-Skripte laufen GANZ AM ANFANG und die post-Skripte GANZ AM ENDE. Dem ist scheinbar nicht so. +++

So ist es :wink: Ohne Debuglog kann ich aber nicht weiterhelfen.

bcutter, 2024-06-28:
Wird schon so sein.

RaspiBackup legt den Backup-Ordner an, führt die Pre-Skripte aus (wobei der angelegte Ordner gelöscht wird) und beim ersten Schreibversuch (dem dd-Backup der Boot-Partition) scheitert das ganze.

Sinnvoller: vor den Pre-Skripten gar nichts anlegen.

Hab es für mich wie im letzten Post dokumentiert mit dem Workaround gelöst. Werde das ein zwei Backup-Intervalle (1-2 Wochen) beobachten.

Gern nach eigenem Ermessen schließen oder Optimieren. Danke für raspiBackup 👍🏼

framps, 2024-06-28:
> wobei der angelegte Ordner gelöscht wird

Ist natuerlich etwas ungeschickt den angelegten neuen Backupordner raspiBackup unter dem Hintern wegzuziehen :laughing: 

> Ich sichere per tar nach "/mnt/localbackup", einen kleinen USB-Stick. Von dort wird nächtlich per rsync weg gesichert.

Ich wuerde an der Stelle den USB Stick loeschen.

bcutter, 2024-06-28:
> Ist natuerlich etwas ungeschickt den angelegten neuen Backupordner raspiBackup unter dem Hintern wegzuziehen :laughing: 

Tja woher soll man ahnen dass noch VOR einem PRE-Skript was passiert 😀

> Ich wuerde an der Stelle den USB Stick loeschen.

Auf dem Stick sind noch andere Daten. Und als Zwischenlager macht er aus vielerlei Hinsicht Sinn (u. a. niedrigere Backuo-Dauer = kürzere Downtime) und wird daher erstmal beibehalten.

bcutter, 2024-07-12:
> Hab es für mich wie im letzten Post dokumentiert mit dem Workaround gelöst. Werde das ein zwei Backup-Intervalle (1-2 Wochen) beobachten.

Beobachtet, läuft so.

closed 2024-07-28 by github-actions[bot]