framps/raspiBackup #769: Fehler bei Backup

klajue, 2024-07-07:
**Note**
raspiBackup is supported on RaspberryOS and Ubuntu as operating system and Raspberry HW only. There are environments out there which successfully run raspiBackup but any support requests or issues on unsupported environments will be rejected. 

*If you have questions about raspiBackup please create a `discussion` entry instead of an `issue` entry in github*

If you have an issue with raspiBackup execute following actions first
1. Read the FAQ pages of raspiBackup 
   * [English](https://www.linux-tips-and-tricks.de/en/faq)
   * [German](https://www.linux-tips-and-tricks.de/de/faq)
1. Check whether the error messages of raspiBackup were listed on the messages page and all suggested actions don't fixes the issue
   * [English](https://www.linux-tips-and-tricks.de/en/raspibackupmessagese)
   * [German](https://www.linux-tips-and-tricks.de/de/raspibackupmeldungen)
1. Use your preferred search engine and look up the error message number or the error message text you receive

If you still don't get any idea how to solve your issue with raspiBackup
1. Explain in detail what's going wrong
1. Attach raspiBackup debug log. Just **DRAG AND DROP** raspiBackup.log into the issue and don't copy and paste the contents of raspiBackup.log into the issue. **If you don't attach a debug log the issue will be rejected**

**Note**
Issues are closed automatically after 2 weeks if there is no activity any more

Hallo Framps,
folgende Meldung habe ich bei dem Backup erhalten:
--- RBK0159I: 10 Backups werden für den Backuptyp rsync aufbewahrt. Bitte Geduld.
--- RBK0205I: Älteste Backup /backup/raspberrypi5 in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0001E: Unerwarteter Programmfehler trat auf. (dcfab2f), Zeile: 6131, Fehler: push to /backup/raspberrypi5 failed.
File "/usr/local/bin/raspiBackup.sh", line 1, in main
??? RBK0148E: @@@@@@@@@@@@@@@@@@@@ Stacktrace @@@@@@@@@@@@@@@@@@@@
#!/bin/bash
File "/usr/local/bin/raspiBackup.sh", line 4791, in cleanup
			applyBackupStrategy
File "/usr/local/bin/raspiBackup.sh", line 6131, in applyBackupStrategy
				assertionFailed $LINENO "push to $BACKUPPATH failed"
File "/usr/local/bin/raspiBackup.sh", line 2508, in assertionFailed
logStack
pi@raspberrypi5:~ $ 
in /root ist keine Log-Datei eingetragen. Bitte um Info bzw. Hilfe.
Gruß
klajue

framps, 2024-07-07:
Das ist schon irre. raspiBackup gibt es seit mehr als 10 Jahren und Du bist der Erste der diesen Fehler bekommt :smiley: 

> in /root ist keine Log-Datei eingetragen

Hast Du die FAQs gelesen? Dort gibt es [FAQ39](https://www.linux-tips-and-tricks.de/de/faq#a39) :wink: 

Lade die Logdatei per DragnDrop (nicht Copy&Paste) hier hoch damit ich sie mir ansehen kann.

klajue, 2024-07-08:
Hallo Framps,,
danke für Deine Antwort. Es gibt leider keine Log-Datei, weder in home/user noch in /root oder /tmp. Ich habe das erste Backup am 02.06.24 erstellt und am 03.06.24 den Restore auf eine externe SSD geschrieben und getestet alles OK. am 09.06.24 ist automatisch ein Backup erstellt worden. Bei den Backups vom 02.06.24 und 09.06.24 sind die Log-Dateien in den Backup-Verzeichnissen eingetragen worden und für den Restore vom 03.06.24 in home/user. Für den Backup vom 07.07.24, der fehlerhaft beendet wurde, gibt es ein Backup-Verzeichnis mit scheinbar allen allen Dateien (Anzahl und Größe liegen im üblichen Bereich). Ich habe nur die Kopie von der Terminal-Anzeige des Backups, von der ich Dir den letzten Teil in meinem ersten Post schon geschrieben habe. Wie kann es  jetzt weitergehen? Bitte um Info bzw. Hilfe
Guß
klajue

framps, 2024-07-08:
Moin,

eben habe ich Deinen Fehlerfall mal provoziert und ich finde wie erwartet in /tmp die Datei raspiBackup.log. Bitte sieh noch mal genau nach. Der Grund warum ich die Debugdatei haben moechte ist dass Du mir und Dir dadurch eine Menge Arbeit ersparst denn ansonsten muss ich alle moeglichen Befehle zusammensuchen um den Status Deines Systems sowie Deine raspiBackup Konfiguration zu erfragen.

Der Fehler ist dass ein pushd Befehl fehlschlaegt. Das ist sehr unueblich.  

Lt der man page kann ein Fehler nur auftreten wenn in das Zielverzeichnis nicht gewechselt werden kann.

```
pushd [-n] [+n] [-n]
       pushd [-n] [dir]
              Adds  a  directory to the top of the directory stack, or rotates
              the stack, making the new top of the stack the  current  working
              directory.   With  no arguments, pushd exchanges the top two di‐
              rectories and returns 0, unless the directory  stack  is  empty.
              Arguments, if supplied, have the following meanings:
              -n     Suppresses  the  normal change of directory when rotating
                     or adding directories to the  stack,  so  that  only  the
                     stack is manipulated.
              +n     Rotates  the  stack  so  that the nth directory (counting
                     from the left of the list shown by  dirs,  starting  with
                     zero) is at the top.
              -n     Rotates  the  stack  so  that the nth directory (counting
                     from the right of the list shown by dirs,  starting  with
                     zero) is at the top.
              dir    Adds dir to the directory stack at the top, making it the
                     new current working directory as if it had been  supplied
                     as the argument to the cd builtin.

              If the pushd command is successful, a dirs is performed as well.
              If the first form is used, pushd returns 0 unless the cd to  dir
              fails.   With the second form, pushd returns 0 unless the direc‐
              tory stack is empty, a non-existent directory stack  element  is
              specified,  or the directory change to the specified new current
              directory fails.
```
Der konkrete Befehl ist `pushd "$BACKUPPATH"`wobei die Variable $BACKUPPATH Dein konfiguriertes Backupverzeichnis `/backup/raspberrypi5` ist. Der wird ausgefuehrt am Ende nachdem ein Backup erstellt wurde um die jeweilige Backuprecyclestrategie anzuwenden - also u.U. alte Backups zu loeschen. 

Im Logfile wuerden wir die genaue Fehlermeldung sehen. Als Ursache kann ich mir nur folgendes vorstellen:
1) Es besteht keine Berechtigung mehr auf das Backupverzeichnis ` /backup/raspberrypi5` zuzugreifen. Das kann eigentlich nicht sein denn es wurden ja wie Du schreibst Backups angelegt. 
2) Das Backupverzeichnis ` /backup/raspberrypi5` ist ploetzlich nicht mehr gemounted. Das kann eigentlich auch nicht passieren. Ausser Du oder irgendein Service auf Deinem System hat das Backupverzeichnis umounted - warum auch immer. 

Kurzum: Ohne das Log bin ich nur am raetseln :cry: 

Kannst Du noch mal einen Backup manuell - also nicht von systemd - erstellen lassen und nachdem der Backup wieder so abgebrochen ist mal die Ausgabe von `mount | grep "/backup"` zeigen sowie den Inhalt von Deiner /etc/fstab?

klajue, 2024-07-09:
Moin Moin,

Danke für Deine Antwort. Ich habe heute ein Backup erfolgreich durchgeführt siehe Terminal_Kopie_Backup_09_07_24.txtund auch die fstab bereitgestellt.
Es gibt keine Log-Datei in /tmp siehe tmp_dir.jpg. Aber ich habe einen Verdacht wie der Fehler möglicherweise entstanden ist. Ich habe zwei RPi5 und einen RPi4B die ich wahlweise einzel an einem Monitor über einen HDMI Switch und einer Kabel losen Tastaur/Maus übereinen 1 auf 4 Switch an die RPis angeschlossen habe. Gestern wollte ich mit dem RPi4B arbeiten. Der RPi4B bootete nicht. ich habe den RPi4B aus dem Gehäuse ausgebaut und mußte feststellen, daß die USB-Verbindung zum USB-Empfänger der Tastatur/Maus über den 1 auf 4 Switch und seine Spannungsversorgung 5Volt auf  den USB-Eingang des RPi4B legte. Ich hatte erwartet das diese 5Volt nur auf der Input Seite des Switch anliegt(für höheren Leistungsbedarf bei HDD's). Nach dem Abschalten der Spannungsversorgung des Switch bootete der RPi4B normal. Diese Konfiguration war am 07.07.24 bei dem fehlerhaften Backup auch aktive. Es könnte also sein, daß das Problem über eine instabile Spannungsversorgung entstanden ist. Was meinst Du dazu?
Gruß
klajue
[Fehler beim Backup.txt](https://github.com/user-attachments/files/16140800/Fehler.beim.Backup.txt)
[Terminal_Kopie_Backup_09_07_24.txt](https://github.com/user-attachments/files/16140801/Terminal_Kopie_Backup_09_07_24.txt)
[image]

framps, 2024-07-09:
> Es könnte also sein, daß das Problem über eine instabile Spannungsversorgung entstanden ist. Was meinst Du dazu?

Das kann sehr gut der Grund fuer Dein Problem sein. Wenn HDDs ploetzlich mehr Strom benoetigen und dieser nicht geliefert wird ist ein Abbruch der USB Verbindung das normale Verhalten der HDDs. 

Ich hatte ja schon unter (2) vermutet dass ein mount zur Backuppartition zu dem Zeitpunkt nicht mehr existiert. Wenn die USB Verbindung wegbricht ist das ja genau der Fall.

Ich gehe davon aus dass das die Ursache ist. Beobachte die Backuperstellung noch ein paar mal und schliesse dann den Issue :wink:

closed 2024-07-25 by github-actions[bot]