framps/raspiBackup #775: Invalid start date

monojk, 2024-09-15:
I have chosed Sunday 05:00 via the installation script using the GUI and noticed that the backup did not happen at night. I think there is an error in raspiBackup.timer
```

pi@rp4b:~ $ sudo systemctl status raspiBackup.timer
â— raspiBackup.timer - Timer for raspiBackup.service to start backup
   Loaded: bad-setting (Reason: Unit raspiBackup.timer has a bad unit file setting.)
   Active: inactive (dead)
  Trigger: n/a

Sep 15 09:49:55 rp4b systemd[1]: /etc/systemd/system/raspiBackup.timer:4: Failed to parse calendar specification, ignoring: Sun*-*-* 05:00:42
Sep 15 09:49:55 rp4b systemd[1]: raspiBackup.timer: Timer unit lacks value setting. Refusing.
Sep 15 09:49:55 rp4b systemd[1]: /etc/systemd/system/raspiBackup.timer:4: Failed to parse calendar specification, ignoring: Sun*-*-* 05:00:42
Sep 15 09:49:55 rp4b systemd[1]: raspiBackup.timer: Timer unit lacks value setting. Refusing.
Sep 15 09:50:00 rp4b systemd[1]: /etc/systemd/system/raspiBackup.timer:4: Failed to parse calendar specification, ignoring: Sun*-*-* 05:00:42
Sep 15 09:50:00 rp4b systemd[1]: raspiBackup.timer: Timer unit lacks value setting. Refusing.
Sep 15 09:50:01 rp4b systemd[1]: /etc/systemd/system/raspiBackup.timer:4: Failed to parse calendar specification, ignoring: Sun*-*-* 05:00:42
Sep 15 09:50:01 rp4b systemd[1]: raspiBackup.timer: Timer unit lacks value setting. Refusing.
Sep 15 09:50:05 rp4b systemd[1]: /etc/systemd/system/raspiBackup.timer:4: Failed to parse calendar specification, ignoring: Sun*-*-* 05:00:42
Sep 15 09:50:05 rp4b systemd[1]: raspiBackup.timer: Timer unit lacks value setting. Refusing.
pi@rp4b:~ $ cd /etc/systemd/system
pi@rp4b:/etc/systemd/system $ ls -l *timer
-rw-r--r-- 1 root root 232 Sep 14 18:22 raspiBackup.timer
pi@rp4b:/etc/systemd/system $ cat raspiBackup.timer
[Unit]
Description=Timer for raspiBackup.service to start backup

[Timer]
OnCalendar=Sun*-*-* 05:00:42
# Create a backup once a week on Sunday morning at 5 am (default)
Unit=raspiBackup.service

[Install]
WantedBy=multi-user.target
```
According to the https://www.freedesktop.org/software/systemd/man/latest/systemd.time.html there should be a blank after Sun.

framps, 2024-09-15:
That's definitely wrong in the timer unit.  Do you use the latest installer? 
```
raspiBackupInstallUI -h
raspiBackupInstallUI.sh 0.4.8.1, 2024-08-01/13:47:18 - 72bcb10
```
If not please update the installer with
```
sudo raspiBackupInstallUI -U
```
and configure the backup time with the installer. If you still get the issue please provide the installer debug log file raspiBackupInstallUI.log .

monojk, 2024-09-16:
Yes, it was the latest installer 0.4.8.1. I installed it the first time on my third raspi. One the other two raspis it was also a fresh install and O got the same problem.

monojk, 2024-09-16:
I encounter another problem during the backup mode configuration of the third raspi.
` (*) Backup more than two partitions`: I am stuck as I cannot go beyond the selection giving me an error screen.
[image]
[image]

framps, 2024-09-16:
> Yes, it was the latest installer 0.4.8.1. I installed it the first time on my third raspi. One the other two raspis it was also a fresh install and O got the same problem.

That's weired. I cannot reproduce your issue.

Please edit /etc/systemd/system/raspiBackup.timer and add the missing space.  Then update the time again with latest installer. If you get the issue again please drag'n drop the installer log in this issue.

framps, 2024-09-16:
> I encounter another problem during the backup mode configuration of the third raspi.

That's another issue and usually a new issue should be created for this :wink: . 

Anyhow: You don't have the standard Raspbian paritioning where partition 1 is the boot device and parition 2 the root device. Please provide the output of

```
lsblk -fs
mount
cat /etc/os-release
```

monojk, 2024-09-16:
As for the issue with the regular backup, I fail to reproduce the issue when running `sudo raspiBackupInstallUI.sh` again. Maybe the issue was due to the default?

framps, 2024-09-16:
> Maybe the issue was due to the default?

I don't think so.

That's the default:

```
read -r -d '' SYSTEMD_TIMER <<-'EOF'
[Unit]
Description=Timer for raspiBackup.service to start backup

[Timer]
OnCalendar=Sun *-*-* 05:00:42
# Create a backup once a week on Sunday morning at 5 am (default)
Unit=raspiBackup.service

[Install]
WantedBy=multi-user.target
EOF
```

> As for the issue with the regular backup, I fail to reproduce the issue

Same for me. Did you manually modify the timer unit in any way? The installer is really picky and may fail to parse the file if it was not modified by himeself.

monojk, 2024-09-16:
I only edited the  /etc/systemd/system/raspiBackup.timer after I saw that there was no backup taken and inspected the parsing error messages from the journal

framps, 2024-09-16:
Ok. I'm just guessing because your issue is unreproducable. 

As long as you don't modify the time or scheduling interval you will not get the issue again. But if you modify one of these settings again in the future check the timer unit file and if you see the issue again please attach the installer debug log which should help me to find the root cause of the issue.

closed 2024-10-03 by github-actions[bot]