framps/raspiBackup #776: problem during the backup mode configuration of the third raspi.

monojk, 2024-09-16:
I encounter another problem during the backup mode configuration of the third raspi.
` (*) Backup more than two partitions`: I am stuck as I cannot go beyond the selection giving me an error screen.
[image]
[image]

_Originally posted by @monojk in https://github.com/framps/raspiBackup/issues/775#issuecomment-2352464588_

monojk, 2024-09-16:
As requested the debug outputs:
```
pi@rp3b0:~ $ lsblk -fs
NAME      FSTYPE LABEL    UUID                                 FSAVAIL FSUSE% MOUNTPOINT
mmcblk0p1 vfat   RECOVERY F1DD-AA4B                                           
â””â”€mmcblk0                                                                     
mmcblk0p2                                                                     
â””â”€mmcblk0                                                                     
mmcblk0p5 ext4   SETTINGS 498213f1-0202-42b8-9fb1-b72f36777dee                
â””â”€mmcblk0                                                                     
mmcblk0p6 vfat   boot     6259-CBFF                               204M    19% /boot
â””â”€mmcblk0                                                                     
mmcblk0p7 ext4   root     b235ddeb-47f8-498c-91a0-9020c5a8baf0   11.1G    52% /
â””â”€mmcblk0                                                                     
mmcblk0p8 ext4   data     ca3b8f2b-5e5a-417d-8d2f-3808a066294a  451.7M     0% /media/pi/data1
â””â”€mmcblk0                                                                     
```
```
pi@rp3b0:~ $ mount
/dev/mmcblk0p7 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=439400k,nr_inodes=109850,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p6 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94536k,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /home/pi/.gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
/dev/mmcblk0p8 on /media/pi/data1 type ext4 (rw,nosuid,nodev,relatime,uhelper=udisks2)
rp4b:/raspiBackupDir on /raspiBackupDir type nfs4 (rw,relatime,vers=4.2,rsize=131072,wsize=131072,namlen=255,soft,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.178.10,local_lock=none,addr=192.168.178.11)
```
```
pi@rp3b0:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

monojk, 2024-09-16:
Thanks, it's running now, not yet complete but the backup disk is filling. Yes, I used NOOBS to create the image several years ago...
I'll update the comment once the backup has completed

monojk, 2024-09-18:
Initial backup completed after 37 hours...

closed 2024-09-18 by monojk

framps, 2024-09-18:
From time to time users of raspiBackup reported a slow backup with rsync over nfs. I unfortunately didn't get any feedback whether they just stopped to use raspiBackup because of this or found the root cause and fixed it. Actually it's not a bug of raspiBackup. It's some environmental and configuration issue with rsync, nfs and network latency.

An alternative is to use a local connected USB disk. This works if you need to backup only one RPi. If you have multiple RPis it's expensive to to connect a dedicated USB disk to all RPis just for a backup. An nfs server is the right way to go with multiple RPis.