framps/raspiBackup #778: Add an option to disable linking of rsync backups to previous backups to force a full backup

framps, 2024-09-22:
Unfortunately binLMU packages may not updated with rsync because they get the same timestamp and size during package build and therefore rsync doesn't detect a file change and doesn't update the file in the backup.

In particular when an OS update was made the first backup of this updated system should be a full update to catch the issue with the binLMU packages. 

As of now there is not way to get raspiBackup to create a new full rsync backup which will catch this issue.

gchriz, 2024-09-23:
Hint: "bin**L**MU packages" should read "bin**N**MU ..."

framps, 2024-09-23:
Thx. Typo fixed.

framps, 2024-09-23:
I got a proposal to handle this issue in a more elegant way:

Use a different naming scheme of the backup directories which includes the os version. 
Use
`raspberrypi@debian12-rsync-backup-20240416-094106` 
instead of 
`raspberrypi-rsync-backup-20240416-094106` 

Then every time a new OS version is saved a new backup directory is used and thus a new full backup is created.

It's a low hanging fruit to add another option. A change of the backup directory name requires much more effort.

I think I will add a new option first and change the backup directory name in a later release.

framps, 2024-10-03:
This issue is obsolete and will be handled by #780

closed 2024-10-03 by framps