framps/raspiBackup #781: Restore of an rsync backup fails when using '-v' (Fix is provided)

rpi-simonz, 2024-09-25:
When using option '-v' on a restore rsync fails with 
```
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sda1 zurückgespielt.
--- RBK0055I: Zweite Partition (Rootpartition) wird auf /dev/sda2 zurückgespielt.
rsync: --delete-v: unknown option
rsync error: syntax or usage error (code 1) at main.c(1795) [client=3.2.7]
??? RBK0035E: Backupprogramm rsync endete beim Restore mit RC 1.
```

Reason:

In function restore() there is a space character missing (last line here):
```
                                $BACKUPTYPE_RSYNC)
                                        local excludePattern="--exclude=/$HOSTNAME-backup.*"
                                        logItem "Excluding excludePattern"
                                        local progressFlag=""
                                        (( $PROGRESS && $INTERACTIVE )) && progressFlag="--info=progress2"
                                        local cmd="rsync $progressFlag --numeric-ids ${RSYNC_BACKUP_OPTIONS}${verbose} ${RSYNC_BACKUP_ADDITIONAL_OPTIONS} $excludePattern \"$ROOT_RESTOREFILE/\" $MNT_POINT"
```
```
${RSYNC_BACKUP_OPTIONS}${verbose} 
```
needs to be 
```
${RSYNC_BACKUP_OPTIONS} ${verbose} 
```

framps, 2024-09-25:
Good catch :+1:  Do you want to create a PR on master which I will immediately accept or do you want me to fix this?

rpi-simonz, 2024-09-25:
Just do it yourself. It's easier and faster.

framps, 2024-09-25:
Just fixed and published. Use `raspBackup -U -S` to update raspiBackup.

commit a902cec4 (2024-09-25): Fixed #781