framps/raspiBackup #791: Include the OS release info in backup directory names

framps, 2024-10-02:


framps, 2024-11-09:
PR merged right now. Thank you very much @rpi-simonz for the new feature and PR.

closed 2024-11-09 by framps

framps, 2024-11-09:
PR branch (m_778) now merged into new branch d_778 which will be used to system test the new feature.

bcutter, 2025-03-08:
So any clever idea on how to semi-automatically delete all oder versions with the old naming convention, now that a second set of "DEFAULT_KEEP_BACKUPS" backups is stored (a completely new series, which is why the first raspiBackup run with 0.7.x took 6 hours with all services down...)?

My backup storage is limited and I neither want to delete all backups of the old naming nor manually delete every single backup after a new one has been taken. This really is some kind of breaking change (where I still don't get the actual benefit/advantage), at least the e-mail log complains about this (5 times).

framps, 2025-03-08:
I see two ways to mitigate your issue:

1) Restore the old release 0.6.9.1. There are a bunch of folks out there who use very old releases of raspiBackup and don't update raspiBackup. 

2) Write a small script which just renames all old backup directories so they get the new name with the release included. Then raspiBackup will include them in the recycle process.

bcutter, 2025-03-08:
> 2\. Write a small script which just renames all old backup directories so they get the new name with the release included. Then raspiBackup will include them in the recycle process.

Wow, that's indeed a great tip, thank you very much! I wasn't aware this would work at all. No side effects?

So I'm just gonna make a `hostname@OSinfo-rsync-backup-20240906-053001` out of `hostname-rsync-backup-20240906-053001` for all folders with the old naming. That's easy.

bcutter, 2025-03-08:
**Small addition (precise questions) regarding side effects:**
Obviously the first backup created with the v0.7.* series was a "full backup" which is the reason for the huge backup creation duration. Using rsync, now renaming all previous backup(folder)s, won't I have two full backups? Is rsync taking care with all the links? Or shall I simply delete the first "full backup" created by v0.7.0.1 now with all folders renamed to the new scheme?

My `sudo du -sh /mnt/backup/hostname/*` is already running for a long time, not sure it will ever finish. Wanted to have a look at the actual backup size taken by all backups.

framps, 2025-03-08:
> No side effects?

No. raspiBackup uses just the new backup directory names in the recycle process. Old backup directory names are ignored. You even can create any dummy test backup directories with the new naming with no contents and they will be included in the recycle process. 

> That's easy.

Would be great if you share your script 😉

framps, 2025-03-08:
> Using rsync, now renaming all previous backup(folder)s, won't I have two full backups?

Yes. You will have the 0.6.9.1 full backup and the new 0.7 full backup.

> Is rsync taking care with all the links?

It''s not rsync - it's the ext4 filesystem which offers hardlinks who handle everything. 

> Or shall I simply delete the first "full backup" created by v0.7.0.1 now with all folders renamed to the new scheme?

Well, if you delete the first 0.7 full backup, then rename all old backups, and execute raspiBackup 0.7 again you will get another incremental backup which will finish much faster and saves you a lot of disk space.

bcutter, 2025-03-08:
I renamed all of them manually (using `mv`), no script or magic involved. Thanks for answering the other questions. I
- renamed all previous backup folders to match the new naming convention,
- renamed the new full backup to something like `delete-me` (aiming at taking care ext4 doesn't do anything with the previous 0.6.9.1 backup series) and
- started raspiBackup once again which should finish shortly.

Thanks 👍