framps/raspiBackup #800: raspiBackup v0.6.9.1 creates invalid systemd timer "OnCalender" entries

Ch3wi3, 2024-10-29:
I ran into this problem on all of my four Raspberry Pi v4 B running on Ubuntu 22.04 LTS Server. 
After installing and configuring raspiBackup as decribed [here](https://linux-tips-and-tricks.de/de/installation/) I found that the systemd-Timer Entries where not working but spitting out error messages because of a syntax error in the systemd timer unit file. 

Example timer unit file created by raspibackup: 

``` 
[Unit]
Description=Timer for raspiBackup.service to start backup

[Timer]
OnCalendar=Sat*-*-* 05:00:42
# Create a backup once a week on Sunday morning at 5 am (default)
Unit=raspiBackup.service

[Install]
WantedBy=multi-user.target
```

Notice the missing space in the `OnCalendar` string between: `Sat` and `*-*-* 05:00:42`. Without this space char systemd complains on all my Raspberri Pis that the syntax of this unit file is wrong. 

If I manually add the space char so that the `OnCalendar` line looks like this:
`OnCalendar=Sat *-*-* 05:00:42` then everything is fine and the timer works as intended.

framps, 2024-10-29:
It's the installer which updates the time in the timer unit file.  Please make sure you use the latest installer and execute `sudo raspiBackupInstall -U` to install latest version.

Ch3wi3, 2024-10-30:
I did `sudo raspiBackupInstallUI.sh -U` because `raspiBackupInstall` was not available on my system somehow. And yes the issue is gone now. Thanks!

closed 2024-10-30 by Ch3wi3