framps/raspiBackup #801: /etc/systemd/system/raspiBackup.timer not being created correctly.  Error in OnCalendar line.

ratsputin, 2024-11-03:
Fresh installation of raspiBackup with custom date/time selected.  Here is the `/etc/systemd/system/raspiBackup.timer` file created:

```[Unit]
Description=Timer for raspiBackup.service to start backup

[Timer]
OnCalendar=Tue*-*-* 02:00:42
# Create a backup once a week on Sunday morning at 5 am (default)
Unit=raspiBackup.service

[Install]
WantedBy=multi-user.target
```
There should be a space after the `OnCalendar=Tue`.  With this error, raspiBackup will not run on schedule.

The comment following the `OnCalendar` line is incorrect as well, not reflecting the custom date/time.  This is, of course, just a visual thing and inconsequential.

ratsputin, 2024-11-03:
As in #800, I did have to run `raspiBackupInstallUI.sh -U` to do the upgrade.

Once complete, I ran it again without the `-U` switch, changed the date to Monday and the time to 01:00, and this was the result:

```
[Unit]
Description=Timer for raspiBackup.service to start backup

[Timer]
OnCalendar=Mon *-*-* 01:00:42
# Create a backup once a week on Sunday morning at 5 am (default)
Unit=raspiBackup.service

[Install]
WantedBy=multi-user.target
```
So, it did work correctly.  Apparently, installing it using the command
```
curl -o install -L https://raw.githubusercontent.com/framps/raspiBackup/master/installation/install.sh; sudo bash ./install
```
doesn't pick up the latest version of the UI?  That is the command I used for the "fresh install".
This is on a Pi 3B+ with Bookworm if that matters.

framps, 2024-11-04:
> doesn't pick up the latest version of the UI?

I already tried to reproduce the issue when #800 was created but failed. I tried it again and again fail. I called `raspiBackupInstallUI --version` for the downloaded and upgraded version and both git commits are identical.

The issue seems to be related to the steps you execute when you configure raspiBackup with the installer. I think you don't have to update the installer for fix the issue. It's sufficient to stop the installer and start it again. 

I installed raspiBackup and updated the time and day and everything was OK. I tried different config sequences and all the time the config file was OK. Would be great if you remember exactly the sequence of config steps you executed.

framps, 2024-11-04:
> I will remove the comment in the default template.

Removed right now :-)

streamdp, 2025-03-10:
Last week I updated **raspiBackup** and reinstall timer with **raspiBackupInstallUI**, but still have this problem. So I just added space manually in the **raspiBackup.timer** and re-enable and restart timer.  Maybe we need to check if the timer enabled after setting it up with **raspiBackupInstallUI**? I mean parsing the output of `systemctl list-timers` and checking for the presence of **raspiBackup.timer**?

```
raspiBackupInstallUI --version
raspiBackupInstallUI.sh 0.4.8.2, 2025-02-26/14:11:11 - 69b8fac
```

framps, 2025-03-10:
> I mean parsing the output of `systemctl list-timers` and checking for the presence of **raspiBackup.timer**?

Well, when this check fails an exception will be thrown. But this doesn't fix the issue. This is a nasty issue because I cannot reproduce it and I also didn't find the root cause when I checked the code 😒 

Given it's the second time this issue pops up I'm going to rewrite the code dealing with the timer configuration.

streamdp, 2025-03-10:
> But this doesn't fix the issue.

Sure, but we'll know about the problem on the fisrt stages, but not when we need a new backup that we don't have =)

streamdp, 2025-03-10:
> I mean parsing the output of `systemctl list-timers` and checking for the presence of **raspiBackup.timer**?

Or we could check the configuration before installing timer or something else. I mean, in that case we can be sure that everything is fine.  
```
systemd-analyze verify /etc/systemd/system/raspiBackup.*
```

streamdp, 2025-03-10:
https://github.com/framps/raspiBackup/blob/a6f6260401197e3efa4a86adaa71d48f535271c9/installation/raspiBackupInstallUI.sh#L2444
I tried running this on my local machine and got the same result:
```
awk -v minute=1 -v hour=1 -v day=Monday ' { print "OnCalendar="day "*-*-*", hour":"minute":42" }'
OnCalendar=Monday*-*-* 1:1:42
```
I think it's because we need additional comma after `day`.
```
awk -v minute=1 -v hour=1 -v day=Monday ' { print "OnCalendar="day, "*-*-*", hour":"minute":42" }'
OnCalendar=Monday *-*-* 1:1:42
```

framps, 2025-03-10:
I think I found the root cause of the issue and created a fix. Thank you very much for your hint about  `systemd-analyze verify ` which I also call whenever some modifications in the systemd config files are made.

I still have to execute final tests but I'm confident it's OK now.

streamdp, 2025-03-10:
Thanks for your job.

commit 4666b7f8 (2025-03-10): Fixed #801
closed 2025-03-10 by framps