framps/raspiBackup #802: Add "device" parameter for Pushover Notifications

TheMegamind, 2024-11-06:
The [Pushover API](https://pushover.net/api) includes an optional "device" parameter, which allows the user to name the specific device(s) to send the notification to instead of all devices under the user key.

I have several devices in my pushover user account (phones, tablets, etc.) and as currently implemented, raspiBackup sends any requested notifications to ALL of them. It would be preferable to have these instead sent only to selected devices as I do with other applications where I have integrated Pushover.  

Looking through the raspiBackup code, it appears this could be implemented with just 6 or 7 simple lines of code. If I felt capable of submitting a proper pull request, I would do so, but this is probably best left to someone who understand the code in full.  

Thanks for considering this addition.

framps, 2024-11-07:
Makes sense to me to add the device parameter :+1: . It's just another parameter which has to be passed to the API and a new optional pushover configuration parameter `PUSHOVER_DEVICE`. 

If you like you can update raspiBackup right now and add [here](https://github.com/framps/raspiBackup/blob/dc2cd01966e3413d919f3308dc76afdfc87d8548/raspiBackup.sh#L4251) another line
```
--form-string "device=MYDEVICE"\
```
as a workaround until it's implemented in raspiBackup. Will try to get it into next upcoming release :wink:

TheMegamind, 2024-11-07:
Thanks! 

I've also added the optional `ttl` parameter, so that (in my case) the pushover message self-destructs after 24 hours (when the next daily backup is run). It's just housekeeping, but I find that parameter useful as well. 

`--form-string "ttl=86400"\`

framps, 2024-11-07:
Nice option. I'm frankly no pushover user and not familiar with all the options. 

I'll add another option `PUSHOVER_ADDITIONAL_OPTIONS` where any other API options can be defined. In your case you have to define 
```
PUSHOVER_ADDITIONAL_OPTIONS='--form-string "ttl=86400"'
```

framps, 2024-11-08:
Implemented and will be available with 0.6.10

closed 2024-11-08 by framps