framps/raspiBackup #805: Backup diskspace takes 10 times the size of the original diskspace

Hans-Sh, 2024-11-22:
I'm a happy user of raspiBackup on several RPI's but do see strange behavior on my new RPI5.  Yes I did read all the FAQ's and closed bug reports available but still cant find why the disk size used op the backup is more than 10 time the size of the origin according to Windows Folder.

Configuration:
Hardware: Rpi5-4GB, SD 32GB, M2 NMVe 512 GB, Booting from NVMe
Software: Raspbian Bookworm
Docker software mainly: mariaDB. Grafana

Procedure used:
Stop docker with: sudo docker-compose down
start: sudo raspiBackup (configured as rsync backup, stopping containered docker e.o., backup to NFS mounted directory on Synology DSM7 server): email& status backup succesfull in 9 minutes.

Origin:
df -h:
Filesystem                                  Size  Used Avail Use% Mounted on
udev                                        1.9G     0  1.9G   0% /dev
tmpfs                                       405M  6.2M  399M   2% /run
/dev/nvme0n1p2                     468G  7.3G  437G   2% /
tmpfs                                         2.0G  368K  2.0G   1% /dev/shm
tmpfs                                         5.0M   48K  5.0M   1% /run/lock
/dev/nvme0n1p1                     511M   76M  436M  15% /boot/firmware
192.168.2.141:/volume2/HL Backup NL/HSpi17  7.3T  4.2T  3.2T  57% /mnt/backup
tmpfs                                       405M  192K  405M   1% /run/user/1000

Result:  W10 Folder Diskspace used >70GB,    Screendump: 
[image]

            NAS Folder Diskspace used  approx.  8,6 GB    , Screendump: 
[image]

Backup log file is attached.  [raspiBackup.log](https://github.com/user-attachments/files/17871432/raspiBackup.log)

Thanks for your help in advance,
Hans

framps, 2024-11-22:
Please read [FAQ 17](https://www.linux-tips-and-tricks.de/en/faq#a17)

Hans-Sh, 2024-11-22:
Thanks for the link with explanation!