framps/raspiBackup #807: Subject and sender address in email missing (V0.7.0-beta)

UomoInvisibile, 2024-11-27:
I know I'm on an unsupported environment (ArchLinuxARM). But I'm not sure whether this is a general issue. 

Unfortunately, with V0.7.0-beta the configured sender address is not used and the subject is missing. With V0.6.9 everything works as expected.

Any ideas? Thanks in advance, Andreas
[raspiBackup.log](https://github.com/user-attachments/files/17938519/raspiBackup.log)

framps, 2024-11-27:
> I know I'm on an unsupported environment (ArchLinuxARM). But I'm not sure whether this is a general issue.

NP. Thank you very much you report your finding.

I checked the debug log and found

```
20241127-184739 DBG 4845:              --- mail  -@@@@l(28) -s "O.o @HOSTNAME@: Backup erfolgreich beendet." b@@@@l(16)
mail: -r is an invalid alias name
mail: -s is an invalid alias name
mail: 'O.o @HOSTNAME@: Backup erfolgreich beendet.' contains invalid byte ' '
```
This code was not modified at all. I just compared 0.6.9.1 with 0.7.0 beta to double check.

Sounds to me `-@@@@l(28` starts with `-r` because mail complaints about `-r`. I have no idea why mail assumes it's an alias name.

Please double check you use the same mail client and the same `EMAIL_PARMS=-@@@@l(28)`

If you provide a backup run log from 0.6.9 I may be able get some idea what's the root cause.

UomoInvisibile, 2024-11-27:
I only updated raspibackup with

`
sudo /usr/local/bin/raspiBackup.sh -U --unsupportedEnvironment
`

No other changes were made. After downgrading to V0.6.9, the subject and the sender address in the sent email are correct.

[raspiBackup.log](https://github.com/user-attachments/files/17939136/raspiBackup.log)

framps, 2024-11-28:
You use 0.6.9 and not 0.6.9.1. There was indeed an update from 0.6.9 to 0.6.9.1 in the eMail handling (See [here](https://github.com/framps/raspiBackup/commit/6638571fb157eeba8c1bd6de99d87e69081c6332#diff-1f94627977853a94ac6ec67a52f59e64a74be29d61890a56a98d61260c90bbf5R4424)) which seems to be the root cause of your issue.

I use exim4 on my systems. What do you use? I assume DEFAULT_EMAIL_PARMS is set to something like `-r <emailAddress>` ?

UomoInvisibile, 2024-11-28:
You are absolutely right about everything. I use postfix on my system and the parameter

`DEFAULT_EMAIL_PARMS="-r <emailAddress>"`

to make it easier to distinguish the emails raspiBackup sends me.

I'm very sorry, but I'm not nearly as good at Linux as I'd like to be and as I should be. Could you please help me with where exactly I can place `set -x` and `set +x`?

UomoInvisibile, 2024-11-29:
Long story short: everything works great with the `mail` program from the `gnu-mailutils` package. At first glance, even the color scheme. Unfortunately, the program is called `gnu-mail` under ArchLinux - I hope I can somehow teach raspiBackup to use that instead of the `mail` that is still installed. Thanks for your support, the beer money will definitely come. Should I close this topic now?

framps, 2024-11-29:
Great you found a workaround with `gnu-mailutils`. 

> I hope I can somehow teach raspiBackup to use that instead of the `mail` that is still installed.

If you don't need the old `mail` command why don't you just change the link of `/usr/bin/mail` to `gnu-mail`?

> Should I close this topic now

Up to you :wink. If you're fine now please close this issue.

UomoInvisibile, 2024-11-29:
> If you don't need the old `mail` command why don't you just change the link of `/usr/bin/mail` to `gnu-mail`?

Thanks for the feedback, I had already thought that. Unfortunately, in Archlinux `mail` is not a link, but the program from `s-nail`. And I'm not sure whether it's needed somewhere somehow. I think I will just uninstall the package and then set the link, as you suggested. If something stops working after that, fortunately and thanks to you I can restore my system from a backup :relaxed:.

closed 2024-11-29 by UomoInvisibile

framps, 2024-11-30:
@UomoInvisibile Thank you very much for your donation :+1: