framps/raspiBackup #810: RBK0142E: Bootgerät kann nicht erkannt werden.

jl61, 2024-12-06:
Hallo,

raspiBackup hat das Backup abgebrochen, weil es die SD-Karte nicht erkennt (?). Die sieht für mich aber völlig normal aus und der Rechner läuft auch ohne sonstige Probleme.

[raspiBackup.log](https://github.com/user-attachments/files/18035811/raspiBackup.log)

Viele Grüße
Jürgen

framps, 2024-12-06:
/boot/firmware muss gemountet sein - ist es aber nicht. Aber sollte sein gemaess fstab:
```
20241205-231543 DBG 2142:          *** cat /etc/fstab
20241205-231543 DBG 2142:              LABEL=writable	            /                     ext4	defaults	0	0
20241205-231543 DBG 2142:              LABEL=system-boot           /boot/firmware        vfat	defaults	0	1
20241205-231543 DBG 2142:              LABEL=NextCloudDrive        /media/USB-Drive2TB   btrfs   defaults,nofail 0 0
```

jl61, 2024-12-06:
Merkwürdig. Vor ein paar Tagen habe ich noch ein Update für Ubuntu installiert, danach muss /boot/firmware/ dann verschwunden sein. Nach einem Reboot ist das Verzeichnis wieder da und raspiBackup läuft auch wieder. Gleich mal notieren, dass ich nach dem nächsten Update von Ubuntu nachschaue, ob das Verzeichnis /boot/firmware/ noch gemountet ist.

closed 2024-12-06 by framps