framps/raspiBackup #813: Report malformed HTML

wspivak, 2024-12-16:
Ver: raspiBackup.sh V0.7.0-beta - 2024-11-26 (2767035) 

If the email is supposed to be HTML, it is malformed.

```
<br/>
--- RBK0009I: Allstar: raspiBackup.sh V0.7.0-beta - 2024-11-26 (2767035) started at Mon 16 Dec 01:00:54 EST 2024.<br/> <span style=color:#FF8000>!!! RBK0165W: =========> NOTE  <========= </span><br/> <span style=color:#FF8000>!!! RBK0165W: This is a betaversion and should not be used in production. </span><br/> <span style=color:#FF8000>!!! RBK0165W: =========> NOTE <=========</span><br/>
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.<br/>
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.<br/>
--- RBK0151I: Using backuppath /mnt/raspi_share/BACKUP/ALLSTARbackup with partition type nfs4.<br/>
--- RBK0271I: Smart recycle strategy will be applied.<br/> <span style=color:#FF8000>!!! RBK0157W: No services to stop.</span><br/>
--- RBK0267I: Extension raspiBackup_mem_pre.sh called.<br/>
--- RBK0267I: Extension raspiBackup_temp_pre.sh called.<br/>
--- RBK0267I: Extension raspiBackup_disk_pre.sh called.<br/>
--- RBK0081I: Creating backup of type tgz in [/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/Allstar@debian12-tgz-backup-20241216-010054.<br/](mailto:/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/Allstar@debian12-tgz-backup-20241216-010054.%3cbr/)>
--- RBK0044I: Creating backup of boot partition in [/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/Allstar@debian12-tgz-backup-20241216-010054/Allstar-backup.img.<br/](mailto:/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/Allstar@debian12-tgz-backup-20241216-010054/Allstar-backup.img.%3cbr/)>
--- RBK0292I: Bootpartition check started.<br/>
--- RBK0045I: Creating backup of partition layout in [/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/tmp/Allstar@debian12-tgz-backup-20241216-010054/Allstar-backup.sfdisk.<br/](mailto:/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/tmp/Allstar@debian12-tgz-backup-20241216-010054/Allstar-backup.sfdisk.%3cbr/)>
--- RBK0046I: Creating backup of master boot record in [/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/Allstar@debian12-tgz-backup-20241216-010054/Allstar-backup.mbr.<br/](mailto:/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/Allstar@debian12-tgz-backup-20241216-010054/Allstar-backup.mbr.%3cbr/)>
--- RBK0158I: Creating native tgz backup "[/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/tmp/Allstar@debian12-tgz-backup-20241216-010054/Allstar@debian12-tgz-backup-20241216-010054.tgz](mailto:/mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/tmp/Allstar@debian12-tgz-backup-20241216-010054/Allstar@debian12-tgz-backup-20241216-010054.tgz)".<br/>
--- RBK0085I: Backup of type tgz started. Please be patient.<br/>
--- RBK0078I: Backup time: 00:07:45.<br/>
--- RBK0033I: Please wait until cleanup has finished.<br/>
--- RBK0299I: Backup directory /mnt/raspi_share/BACKUP/ALLSTARbackup/Allstar/Allstar@debian12-tgz-backup-20241216-010054 created.<br/>
--- RBK0218I: Applying smart backup strategy. Daily:7 Weekly:4 Monthly:3 Yearly:1.<br/>
--- RBK0219I: No backups will be smart recycled.<br/>
--- RBK0267I: Extension raspiBackup_mem_post.sh called.<br/>
--- RBK1001I: Memory usage pre backup - Used: 1492 MB Free: 158 MB<br/>
--- RBK1001I: Memory usage post backup - Used: 1486 MB Free: 221 MB<br/>
--- RBK0267I: Extension raspiBackup_temp_post.sh called.<br/>
--- RBK1000I: CPU temperature pre and post backup: 49.4'C - 57.1'C<br/>
--- RBK0267I: Extension raspiBackup_disk_post.sh called.<br/>
--- RBK1001I: Disk usage pre backup: Used: 316.66 GiB Free: 3.27 TiB<br/>
--- RBK1002I: Disk usage post backup: Used: 319.94 GiB Free: 3.26 TiB<br/>
--- RBK1003I: Disk usage change: -3.28 GiB (0.00 %%)<br/>
--- RBK0017I: Backup finished successfully.<br/>
--- RBK0010I: Allstar: raspiBackup.sh V0.7.0-beta - 2024-11-26 (2767035) stopped at Mon 16 Dec 01:08:41 EST 2024 with rc 0.<br/>
--- RBK0167I: Sending email.<br/>
<br/>
<br/>
```

framps, 2024-12-16:
raspiBackup just passes `Content-Type: text/html; charset=utf-8` in the header. The html formatting is done by the MUA or MTA you use. Given this there is an issue with your MUA or MTA but no issue with raspiBackup.

wspivak, 2024-12-16:
I’m using Outlook…  Just a note that something may not be formatted properly…

 

Thanks for the response!

 

W

framps, 2024-12-16:
Just checked the native eMail sent by my MTA and it has same html formatting like you. Looks like that's a common format used in eMails.

closed 2024-12-16 by framps

wspivak, 2024-12-16:
Ok,Have a great holiday..

framps, 2024-12-16:
Thank you. You too :wink: