framps/raspiBackup #823: [0.7.0-beta] E-Mail notification sent after every execution

cmonty14, 2025-01-02:
Hi,
I deployed beta version_0.7.0-beta_ on my raspi.
Automatic backup creation and cleanup using smart backup strategy works.

However, I get an e-mail notification after every execution although I configured  `DEFAULT_MAIL_ON_ERROR_ONLY=1` and backup _raspiBackup.sh_ exists with returncode 0.
Here's the output of this e-mail:
```
01-02-2025 05:21:56 --- RBK0017I: Backup erfolgreich beendet.
01-02-2025 05:21:56 --- RBK0010I: loxberry: raspiBackup.sh V0.7.0-beta - 2024-12-09 (6cc2feb) Do 2. Jan 05:21:56 CET 2025 beendet mit Returncode 0.
01-02-2025 05:21:56 --- RBK0167I: Eine eMail wird versendet.
```

Please advise how to fix this.

THX

framps, 2025-01-02:
Thank you very much you tested the beta version. I'll check and fix the issue. But it will take some time. When it's fixed I'll update this issue and you will get notified.

framps, 2025-01-02:
I was just able to reproduce your issue.

I then checked the code and detected the beta version will send an email all the time. raspiBackup sends an email if a warning message was written even raspiBackup finishes successfully and `DEFAULT_MAIL_ON_ERROR_ONLY=1` is set. Every beta will write a warning not to use the beta in production :wink:

So it's not a bug and works as designed.

Thank you very much again for your test and report.

closed 2025-01-02 by framps