framps/raspiBackup #824: Feature Request: Docker-compose.yml for Rasberry OS

kaljade, 2025-01-02:
Hi there,

I have raspiBackup installed on my RPi using the recommended installation method. I noticed that there is Docker support but it appears to be bundled with a Raspbian image for other OSs. I was wondering if there was a way to install raspiBackup as a Docker container on the RPi as I use Watchtower to auto-update my containers so I'd love it if that were possible?

Feel free to close as Not Planned if this is not something you are interested in supporting.

Cheers,

Kal

framps, 2025-01-03:
Not sure I get your point. raspiBackup is a script which can backup Raspberries and not docker containers.

kaljade, 2025-01-03:
Sorry

framps, 2025-01-03:
I strongly recommend to update raspiBackup manually and then test the backup and restore to make sure everything works still fine. 

Actually there is no need to update raspiBackup if you don't need new features or there is any bug fixed you encountered.

roisec, 2025-01-05:
I think it might be beneficial to run raspiBackup in a Docker container. This would make it easier to monitor the process, limit CPU usage, and manage resources effectively. It could also help the backup run independently without software dependencies on the host system.

What do you think?

framps, 2025-01-05:
raspiBackup is able to backup the system raspiBackup is running on. The system has to be a Raspberry OS. The only scenario I see raspiBackup makes sense to run in a docker container if the container runs RaspbianOS.But I don't know whether this will work because I never tested this scenario and thus is not supported.

kaljade, 2025-01-05:
Okay, thanks for clarifying this for me, if it's not something you plan on supporting you can close this now.

Kind regards,

Kal

framps, 2025-01-05:
Maybe I don't get you. But as far as I see it's a very specific use case and it doesn't make sense to implement a feature for 1 user out of 5000. A new feature makes sense if there a a lot of users waiting for the feature and will use it.

kaljade, 2025-01-11:
That's fair enough, perhaps I'm overengineering a simple solution. I guess the end result would simply be an automated update solution that could be scheduled via cron whereby you could specify what release version you would like the update to be triggered on: major, minor, patch. The only reason I suggested Docker is that through Watchtower or Whatsupdocker these updates can be automated/scheduled.

Thanks for taking the time to answer my question.

Cheers,

Kal

framps, 2025-01-11:
> Thanks for taking the time to answer my question.

You're welcome

closed 2025-01-11 by framps