framps/raspiBackup #825: Evaluate usage of btrfs snapshots

framps, 2025-01-03:
With btrfs-convert it's easy to convert ext3/4 into btrfs so snapshots can be used.

DBa20-24, 2025-01-03:
Suggestion:
- shutdown services (as configured by user)
- for every partition to be saved, determine if it is a snapshottable one (btrfs is not the only one with snapshot support) and snapshot it if it is snapshottable
- backup all non-snapshottable partitions
- startup services again (as configured by user)
- backup the snapshots created earlier
- optional: smart-delete snapshots (not the worst idea to have them at hand in case you accidentially delete a non-system-critical file)

This could become the default behavior, with a command-line option to ignore snapshotting and proceed with classical backup instead.

DBa20-24, 2025-01-06:
Some code snippets, maybe useful. $PART contains the currently handled partition, $NAME contains the name of the snapshot (suggestion: the name of the current backup directory, so like hostname@osrelease-rsync-backup-timestamp or similar).

Identify filesystem type ("none" if not mounted or other exceptions):
findmnt -n -o FSTYPE /dev/$PART|| echo "none"

Create the snapshot:
btrfs subvolume snapshot "$(findmnt -n -o TARGET /dev/$PART)" "/$(basename "$NAME")"

Remove the snapshot after successfull backup:
btrfs subvolume delete "$(findmnt -n -o TARGET /dev/$PART)/$NAME"

DBa20-24, 2025-01-13:
I managed to make it work, with following method: you can mount a snapshot (rather than the volume itself), so I did the following:

- manually entered "`service stop mariadb && btrfs subvol snapshot / /.snapshot-raspiBackup && service start mariadb`" into DEFAULT_STOPSERVICES in the config file
- manually entered "`btrfs subvolume delete /.snapshot-raspiBackup`" into DEFAULT_STARTSERVICES in the config file
- modified around 5 lines of raspiBackup.sh to add options "`-o ro,subvol=/.snapshot-raspiBackup`" for btrfs filesystems.

With this, following happens: in the "pre"-Step, certain services (in my case: mariadb) are stopped, then a snapshot is taken (under the name /.snapshot-raspiBackup), and then the services are started.
Then, the normal regular process kicks in, however, it mounts the subvolume "/.snapshot-raspiBackup" onto /tmp/raspiBackup/sda2" (because of the mount-option `subvol=/.snapshot-raspiBackup`, the "ro" option is just for safety reasons).
Finally, upon finishing the backup, the snapshot is deleted again to save space when/instead of executing the "start services" step.

DBa20-24, 2025-01-13:
Once my current backup run is finished, I am going to submit a pull request.

DBa20-24, 2025-01-14:
#829 created

framps, 2025-01-14:
Thank you very much for the PR. That way you help to get btrfs snapshot support into raspiBackup and I don't have to do any evaluation :+1: 

Given the fact the number of users who will use btrfs snapshots to minimize downtime of any services I want to minimize the code change in raspiBackup for this feature. 

It's a great idea to use the pre backup extension to create a snapshot and start the stopped services again immediately. I have to think about how you can add the required additional options for btrfs when the source filesystem for the backup is mounted. Will keep you posted.

DBa20-24, 2025-01-14:
Maybe an extra command line flag, to be passed when calling, in the form of "sda2:<...>"?

framps, 2025-01-14:
Would you please explain in more detail why this options should be required to support btrfs snapshots?

DBa20-24, 2025-01-14:
Sure:
- For the backup, you are mounting the file system to be backed up on an extra mount point.
- The trick with the snapshots is that by supplying the "subvol=/.snapshotXXX" you are mounting the snapshot with the name ".snapshotXXX" rather than the actual filesystem. After that,  normal rsync can be executed :)
So if I can pass "sda2:subvol=/.snapshotXXX" on the command line and raspiBackup would take "subvol=/.snapshotXXX" when mounting sda2, no sophisticated logic is needed within RBA.
Or, even more crazy: make it a parameter exported to extensions, so that an extension can populate it.

framps, 2025-01-16:
@DBa20-24 I created a [fork of current Beta](https://github.com/framps/raspiBackup/tree/b_825) which will include a mount extension for you. I'm still not complete. Will let you know when I'm done.

framps, 2025-03-25:
Closed because there is no activity on #829 any more

closed 2025-03-25 by framps