framps/raspiBackup #826: Raspberry Pi Crashes During Midnight Backup

roisec, 2025-01-05:
Description
When running raspiBackup at midnight, my Raspberry Pi hangs, causing all Docker containers to shutdown.

Questions

1. How can I confirm the backup completed successfully and is usable?
2. What’s the best way to monitor the backup process (e.g., journalctl)?
3. How can I detect and handle stuck backup processes?
4. What steps can I take to prevent this issue in the future?

framps, 2025-01-05:
1. First of all you should create a backup and test the restored backup carefully to make sure a backup created with raspiBackup is OK. Keep in mind there is a chance a backup from a running system may be not OK for some reasons. Then you should configure raspiBackup to create backups over night. If raspiBackup terminates successfully the backup should be OK. But from time to time test the restore again. raspiBackup will remind you to execute a restore test every 6 months. This interval is configurable.
2. User option `-g` to get some progress information or option `-v` to get a detailed progress information. Keep in mind option `-v` will create a huge log.
3. raspiBackup writes the debug log in /tmp/raspiBackup.log. You can tail this log to detect where raspiBackup is stuck. Keep in mind the first rsync backup is a full backup and may take a long time. In particular if you use a network attached backup partition rsync may be very slow if you have a lot of data to backup and/or a slow networking connection. raspiBackup doesn't hang. If the backup takes log it's the Linux backup  program used, network connection and root partition size which slows down the backup process. To check whether you have a networking issue you can attach a disk to your system and create the backup locally instead over the network.
4. Again: It's not raspiBackup which hangs. It's the used Linux backup program like dd, tar or rsync which is slow. Carefully test the backup process manually and tune your network connection or use a local attached disk.

roisec, 2025-01-05:
I’ve conducted all the tests you mentioned, including using rsync with delta, and the backups generally work fine. This is the first time I’ve encountered this issue.

In my case, when the backup started, raspiBackup terminated some critical services, like Docker. However, once the backup didn’t finish, those services didn’t restart automatically.

Additionally, while the backup is running, it consumes a significant amount of CPU resources, which causes the OS to become very slow. This, in turn, makes it difficult to manage the system. In some cases, I’ve had to reboot the machine, which interrupts the backup process and terminates it without completion.

Regarding the first question: I don’t see any clear historical logs indicating whether the backup process ended successfully or with an error. I also couldn’t find a success log in the files you mentioned. I think it’s important to include this in future updates for better clarity.

What do you think?

framps, 2025-01-05:
> In my case, when the backup started, raspiBackup terminated some critical services, like Docker. However, once the backup didn’t finish, those services didn’t restart automatically.

It depends on the applications you run whether they have to be stopped before the backup. I frankly think it's useful for a docker container. When the backup takes a long time the applications/container are then stopped. At the end of the backup they are started again. There is a nice  [feature request to use btrfs snapshots](https://github.com/framps/raspiBackup/issues/825) to make the application/docker container stop time as short as possible which will solve your issue. But as of now this feature is not available.

> Additionally, while the backup is running, it consumes a significant amount of CPU resources,

If you search in the net you'll get a lot of hits about this issue with rsync. You can fine tune rsync and add additional rsync options in the raspiBackup config to limit CPU and memory usage.

> Regarding the first question: I don’t see any clear historical logs indicating whether the backup process ended successfully or with an error. I also couldn’t find a success log in the files you mentioned. I think it’s important to include this in future updates for better clarity.

I see your point. It would be useful to use the systemd log to log the raspiBackup messages. I created a [feature request](https://github.com/framps/raspiBackup/issues/827) for this.

roisec, 2025-01-06:
What about adding a success log entry in the raspiBackup log file? Currently, I don’t see any clear indication of whether the backup finished successfully or not. This would be very helpful for tracking and debugging.

roisec, 2025-01-06:
Hbbb him

framps, 2025-01-06:
> I don’t see any clear indication of whether the backup finished successfully or not. This would be very helpful for tracking and debugging.

Not sure I get you :thinking: . That's the output of my last backup run of my RPi4:

```
-- RBK0009I: asterix: raspiBackup.sh V0.7.0-beta - 2024-12-09 (6cc2feb) started at Sun Jan  5 03:00:51 AM CET 2025.
!!! RBK0165W: =========> NOTE  <========= 
!!! RBK0165W: This is a betaversion and should not be used in production. 
!!! RBK0165W: =========> NOTE <=========
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0151I: Using backuppath /backup with partition type nfs4.
!!! RBK0157W: No services to stop.
--- RBK0081I: Creating backup of type rsync in /backup/asterix/asterix@debian12-rsync-backup-20250105-030050.
--- RBK0044I: Creating backup of boot partition in /backup/asterix/asterix@debian12-rsync-backup-20250105-030050/asterix-backup.tmg.
--- RBK0085I: Backup of type rsync started. Please be patient.
--- RBK0078I: Backup time: 00:06:32.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0299I: Backup directory /backup/asterix/asterix.framp.de@debian12-rsync-backup-20250105-030050 created.
--- RBK0159I: 60 backups kept for rsync backup type. Please be patient.
--- RBK0017I: Backup finished successfully.
--- RBK0010I: asterix: raspiBackup.sh V0.7.0-beta - 2024-12-09 (6cc2feb) stopped at Sun Jan  5 03:07:39 AM CET 2025 with rc 0.
--- RBK0026I: Debug logfile saved in /backup/asterix/asterix@debian12-rsync-backup-20250105-030050/raspiBackup.log.
```

and there you find message RBK0017I and RBK0010I. RBK0078I will report the backup time. If you configured raspiBackup to send a email or configured pushover or slack you will get a success or failure email or notification.

roisec, 2025-01-07:
Ok so I found these successful logs on raspiBackup.log on the backup folder , why all the daemons not started after the backup ? How can I figure out the root cause? Do you have any idea?

framps, 2025-01-07:
Not sure we're on the same page here. 

Issue subject is `Raspberry Pi Crashes During Midnight Backup`. When you got a new backup directory with a backup log raspiBackup didn't crash. raspiBackup created a backup successfully.

Sounds to me raspiBackup stopped some services and at the end of the backup the startup of the stopped services failed.

That's no issue with raspiBackup. It's an issue with your stop and start configuration. If the start services commands fail - this has to be fixed by you.

Check and test what you configured in `DEFAULT_STARTSERVICES` and `DEFAULT_STOPSERVICES` correctly.

roisec, 2025-01-08:
Ok thanks 
Closing the issue .

closed 2025-01-08 by roisec