framps/raspiBackup #841: Wrapper and pishrink

upD8R, 2025-02-25:
Hi there,

I recently upgraded to 0.7.0 and it seems this broke wrapper & pishrink integration, here's an example error message:

`pishrink.sh: ERROR occurred in line 239: /media/nfs/backups/tick/tmp/tick@raspbian10-dd-backup-20250225-203050/tick@raspbian10-dd-backup-20250225-203050.img is not a file...
`
I'm using the latest version of pishrink v24.10.23 and if I run it on the actual img it works as expected but the error message points to ../tmp/... within the path.

Maybe the file is being moved?

Thanks!

upD8R, 2025-02-25:
Many thanks!

framps, 2025-02-25:
There is a fix available now 😉 . Please verify and close this issue if the verification succeeds.

Download the fix with following command:

`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_841`

Then invoke raspiBackup. Don't forget the leading `.`. Otherwise you will not use the fixed version 😉

upD8R, 2025-02-26:
Thanks, will try but not before tonight.

upD8R, 2025-02-27:
Sorry for the delay. The "tmp" part is gone in the hotfix but pishrink is stubborn ;-)

```
pi@fhem /u/l/bin> sudo ./raspiBackupWrapper.sh
--- /media/nfs/backups already mounted
--- RBK0009I: fhem: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (cdc3116) Do 27. Feb 07:08:31 CET 2025 gestartet
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0173W: Dieses ist ein temporärer Hotfix der auf die nächste Version upgraded werden muss sobald eine verfügbar ist.
!!! RBK0173W: =========> HINWEIS <=========
--- RBK0151I: Backuppfad /media/nfs/backups mit Partitionstyp nfs wird benutzt
--- RBK0081I: Backup vom Typ dd wird in /media/nfs/backups/fhem/fhem@raspbian10-dd-backup-20250227-070827 erstellt
--- RBK0085I: Backuperstellung vom Typ dd gestartet. Bitte Geduld
--- RBK0078I: Backupzeit: 00:22:49
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0299I: Backupverzeichnis /media/nfs/backups/fhem/fhem@raspbian10-dd-backup-20250227-070827 erstellt
--- RBK0218I: Wende smarte Backupstrategie an. Täglich:0 Wöchentlich:4 Monatlich:12 Jährlich:1
--- RBK0217I: 1 Backups werden smart recycled. 1 Backups werden aufgehoben. Bitte Geduld
--- RBK1001I: Speicherauslastung - Vor dem Backup - Belegt: 165 MB Frei: 167 MB - Nach dem Backup: Belegt: 174 MB Frei: 373 MB
--- RBK1000I: CPU Temperatur vor und nach dem Backup: 47.6'C - 53.0'C
--- RBK1002I: Partitionsauslastung vor dem Backup: Belegt: 1.54 TiB Frei: 1.82 TiB
--- RBK1003I: Partitionsauslastung nach dem Backup: Belegt: 1.54 TiB Frei: 1.82 TiB
--- RBK1004I: Änderung freier Platz: -3.87 MiB (0,00 %%)
!!! RBK0156W: Keine Systemd Services sind zu starten
--- RBK0017I: Backup erfolgreich beendet
--- RBK0010I: fhem: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (cdc3116) Do 27. Feb 07:32:24 CET 2025 beendet mit Returncode 0
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert
Backup succeeded :-)
pishrink.sh: Creating log file /usr/local/bin/pishrink.log
PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink

pishrink.sh: ERROR occurred in line 239: /media/nfs/backups/fhem/fhem@raspbian10-dd-backup-20250227-070827/fhem@raspbian10-dd-backup-20250227-070827.img is not a file...
```

framps, 2025-02-27:
Sounds the Backup directory was unmounted when das raspiBackup finished. Do you use dynamic mount?

upD8R, 2025-02-27:
No, I use autofs independent of raspiBackup, it seems the issue is only happening in case I start the backup through the Wrapper script. I can trigger a backup with the hotfix only, let me check.

framps, 2025-02-27:
I just setup an environment like yours:
1) Downloaded latest wrapper
2) Downloaded latest pishrink
3) Downloaded raspiBackup fix
4) Customized wrapper accordingly

Everything works fine ...

```
pi@raspberrypi-bookworm64-lite-beta:~ $ sudo ./raspiBackupWrapper.sh 
--- RBK0009I: raspberrypi-bookworm64-lite-beta: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (69b8fac) started at Thu Feb 27 10:37:37 AM CET 2025
!!! RBK0173W: =========> NOTE  <========= 
!!! RBK0173W: This is a temporary hotfix and has to be upgraded to next available version as soon as one is available. 
!!! RBK0173W: =========> NOTE <=========
--- RBK0151I: Using backuppath /backup with partition type ext4
!!! RBK0157W: No services to stop
--- RBK0081I: Creating backup of type dd in /backup/raspberrypi-bookworm64-lite-beta/raspberrypi-bookworm64-lite-beta@debian12-dd-backup-20250227-103737
--- RBK0085I: Backup of type dd started. Please be patient
--- RBK0078I: Backup time: 00:07:00
--- RBK0033I: Please wait until cleanup has finished
--- RBK0299I: Backup directory /backup/raspberrypi-bookworm64-lite-beta/raspberrypi-bookworm64-lite-beta@debian12-dd-backup-20250227-103737 created
--- RBK0159I: 20 backups kept for dd backup type. Please be patient
--- RBK0017I: Backup finished successfully
--- RBK0010I: raspberrypi-bookworm64-lite-beta: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (69b8fac) stopped at Thu Feb 27 10:44:39 AM CET 2025 with rc 0
--- RBK0026I: Debug logfile saved in /backup/raspberrypi-bookworm64-lite-beta/raspberrypi-bookworm64-lite-beta@debian12-dd-backup-20250227-103737/raspiBackup.log
Backup succeeded :-)
PiShrink v24.10.23 - https://github.com/Drewsif/PiShrink

pishrink.sh: Gathering data
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
pishrink.sh: An existing /etc/rc.local was not found, autoexpand may fail...
grep: /tmp/tmp.J7GRDY5sGn/etc/rc.local: No such file or directory
Creating new /etc/rc.local
pishrink.sh: Checking filesystem
/dev/loop0: 73033/458752 files (2.8% non-contiguous), 564262/1833984 blocks
resize2fs 1.47.0 (5-Feb-2023)
pishrink.sh: Shrinking filesystem
resize2fs 1.47.0 (5-Feb-2023)
Resizing the filesystem on /dev/loop0 to 783026 (4k) blocks.
Begin pass 2 (max = 102593)
Relocating blocks             XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 56)
Scanning inode table          XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 6510)
Updating inode references     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/loop0 is now 783026 (4k) blocks long.

pishrink.sh: Zeroing any free space left
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
pishrink.sh: Zeroed 914M
pishrink.sh: Shrinking partition
pishrink.sh: Truncating image
pishrink.sh: Shrunk /backup/raspberrypi-bookworm64-lite-beta/raspberrypi-bookworm64-lite-beta@debian12-dd-backup-20250227-103737/raspberrypi-bookworm64-lite-beta@debian12-dd-backup-20250227-103737.img from 7.5G to 3.5G
pishrink succeeded :-)
pi@raspberrypi-bookworm64-lite-beta:~ $ 
```

upD8R, 2025-02-27:
Hmm, interesting, I just ran a backup and I could see that the file was created at the destination with increasing file size under a tmp folder.

But after the backup is finished, it's not there. At all.
```
pi@fhem ~> sudo /usr/local/bin/raspiBackup.sh -g
--- RBK0009I: fhem: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (cdc3116) Do 27. Feb 10:49:49 CET 2025 gestartet
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0173W: Dieses ist ein temporärer Hotfix der auf die nächste Version upgraded werden muss sobald eine verfügbar ist.
!!! RBK0173W: =========> HINWEIS <=========
--- RBK0151I: Backuppfad /media/nfs/backups mit Partitionstyp nfs wird benutzt
--- RBK0081I: Backup vom Typ dd wird in /media/nfs/backups/fhem/fhem@raspbian10-dd-backup-20250227-104945 erstellt
--- RBK0085I: Backuperstellung vom Typ dd gestartet. Bitte Geduld
--- RBK0078I: Backupzeit: 00:22:54
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0299I: Backupverzeichnis /media/nfs/backups/fhem/fhem@raspbian10-dd-backup-20250227-104945 erstellt
--- RBK0218I: Wende smarte Backupstrategie an. Täglich:0 Wöchentlich:4 Monatlich:12 Jährlich:1
--- RBK0217I: 1 Backups werden smart recycled. 1 Backups werden aufgehoben. Bitte Geduld
--- RBK1001I: Speicherauslastung - Vor dem Backup - Belegt: 166 MB Frei: 290 MB - Nach dem Backup: Belegt: 173 MB Frei: 336 MB
--- RBK1000I: CPU Temperatur vor und nach dem Backup: 47.6'C - 53.0'C
--- RBK1002I: Partitionsauslastung vor dem Backup: Belegt: 1.54 TiB Frei: 1.82 TiB
--- RBK1003I: Partitionsauslastung nach dem Backup: Belegt: 1.54 TiB Frei: 1.82 TiB
--- RBK1004I: Änderung freier Platz: 0 Bytes (0,00 %%)
!!! RBK0156W: Keine Systemd Services sind zu starten
--- RBK0017I: Backup erfolgreich beendet
--- RBK0010I: fhem: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (cdc3116) Do 27. Feb 11:13:49 CET 2025 beendet mit Returncode 0
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert
```

upD8R, 2025-02-27:
I triggered a partition based tar backup and this works. The backup is there.
I wonder what teh difference may be ..
```
pi@fhem ~> sudo /usr/local/bin/raspiBackup.sh -g -t tar -P -T 1
--- RBK0009I: fhem: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (cdc3116) Do 27. Feb 11:32:02 CET 2025 gestartet
!!! RBK0173W: =========> HINWEIS <=========
!!! RBK0173W: Dieses ist ein temporärer Hotfix der auf die nächste Version upgraded werden muss sobald eine verfügbar ist.
!!! RBK0173W: =========> HINWEIS <=========
--- RBK0151I: Backuppfad /media/nfs/backups mit Partitionstyp nfs wird benutzt
--- RBK0081I: Backup vom Typ tar wird in /media/nfs/backups/fhem/fhem@raspbian10-tar-backup-20250227-113158 erstellt
--- RBK0347I: Partitionsorientierte Backup vom Typ tar started für die Partitionen 1
!!! RBK0324W: Es werden nicht alle OS Partition gesichert und das Backup wird nicht starten
--- RBK0092I: Partition mmcblk0p1 (48.39 KiB/55.73 KiB) wird gesichert
--- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld
--- RBK0112I: Partition mmcblk0p1 wurde gesichert
--- RBK0078I: Backupzeit: 00:00:10
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0299I: Backupverzeichnis /media/nfs/backups/fhem/fhem@raspbian10-tar-backup-20250227-113158 erstellt
--- RBK0218I: Wende smarte Backupstrategie an. Täglich:0 Wöchentlich:4 Monatlich:12 Jährlich:1
--- RBK0219I: Keine Backups werden smart recycled
--- RBK1001I: Speicherauslastung - Vor dem Backup - Belegt: 166 MB Frei: 631 MB - Nach dem Backup: Belegt: 170 MB Frei: 528 MB
--- RBK1000I: CPU Temperatur vor und nach dem Backup: 48.2'C - 48.7'C
--- RBK1002I: Partitionsauslastung vor dem Backup: Belegt: 1.54 TiB Frei: 1.82 TiB
--- RBK1003I: Partitionsauslastung nach dem Backup: Belegt: 1.54 TiB Frei: 1.82 TiB
--- RBK1004I: Änderung freier Platz: -48.50 MiB (0,00 %%)
!!! RBK0156W: Keine Systemd Services sind zu starten
--- RBK0017I: Backup erfolgreich beendet
--- RBK0010I: fhem: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (cdc3116) Do 27. Feb 11:33:19 CET 2025 beendet mit Returncode 0
--- RBK0026I: Debug Logdatei wurde in /media/nfs/backups/fhem/fhem@raspbian10-tar-backup-20250227-113158/raspiBackup.log gesichert
```

framps, 2025-02-27:
> But after the backup is finished, it's not there. At all.

Please read [slide 6](https://raspibackup.linux-tips-and-tricks.de/wp-content/uploads/simple-file-list/raspiBackup0_7_news_en.pdf)

> Even the log file path is different?

>-- RBK0017I: Backup erfolgreich beendet
>-- RBK0010I: fhem: raspiBackup.sh V0.7.0-hotfix_m841 - 2025-02-26 (cdc3116) Do 27. Feb 11:13:49 CET 2025 beendet mit Returncode 0
>--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert

That's interesting. When the backup succeeds the log should be moved into the backup directory. Could you please provide the debug log?

upD8R, 2025-02-27:
I found the issue, it's my smart recycling config, so (again) a user error. For whatever reason the config, which worked before 0.7.0 is now "broken" yet it may make sense, it didn't work:

My setting was
```
# Smart recycle parameters (daily, weekly, monthly and yearly)
DEFAULT_SMART_RECYCLE_OPTIONS="0 4 12 1"
```

and I change it to:
```
# Smart recycle parameters (daily, weekly, monthly and yearly)
DEFAULT_SMART_RECYCLE_OPTIONS="1 4 12 1"
```

Funnily, this old setting supposed to work before as you can see here:
[image]

framps, 2025-02-27:
> For whatever reason

Ok. Now I understand why the image disappeared: 

If you create multiple Backups at one day the first backup of the day will be kept in the recycle process all the time. When you created a second backup of the day with 0 in the config for days the recycle process deleted the current created backup. In raspiBackup.vars the filename of the currently but deleted filename was written and thus raspiBackupWrapper didn't file the created backup any more. If you use and value >0 in the config for the days two backups are kept by the recycle process now and thus the file will not be deleted.

I know that's a bit strange and I already tried to fix this. But given raspiBackup doesn't support more than 1 backup per day and the intelligent backup strategy code is quite complicate the time to fix this I can use for more important things to do for raspiBackup 😉

framps, 2025-03-02:
Published with v0.7.0.1

closed 2025-03-02 by framps