framps/raspiBackup #845: PUSHOVER_ADDITIONAL_OPTIONS not appending properly (v7.0)

TheMegamind, 2025-03-01:
In order to get the following option to append correctly to the curl command:

`DEFAULT_PUSHOVER_ADDITIONAL_OPTIONS="--form-string ttl=86400"`

I had to modify line 4760 by removing the quotes here: 

`[[ -n $PUSHOVER_ADDITIONAL_OPTIONS ]] && cmd+=( "$PUSHOVER_ADDITIONAL_OPTIONS" )`

so that the line instead reads

`[[ -n $PUSHOVER_ADDITIONAL_OPTIONS ]] && cmd+=( $PUSHOVER_ADDITIONAL_OPTIONS )`

Without this change, the following error occurs and the message is not sent: 

`
DBG 4760: --- Pushover curl call: --form-string message=âœ”ï¸ @HOSTNAME@: Backup finished successfully --form-string token=a@@@@f(30) --form-string user=u@@@@z(30) --form-string priority=0 --form-string html=1 --form-string message=RBK0010I: @HOSTNAME@: raspiBackup.sh V0.7.0 - 2025-02-21 (66089be) stopped at Sat Mar  1 07:52:40 AM PST 2025 with rc 0 --form-string title=âœ”ï¸ @HOSTNAME@: Backup finished successfully --form-string sound=vibrate --form-string device=phone --form-string ttl=86400
curl: option --form-string ttl=86400: is unknown
curl: try 'curl --help' or 'curl --manual' for more information
`

framps, 2025-03-01:
Thank you very much you report this issue. 

I just modified the code. Please verify the fix and download the fix with `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_845` and then close this issue.

There are already two fixes planned for 0.7.0.1 and this fix will also be included there when verified.

framps, 2025-03-02:
Published with v0.7.0.1

closed 2025-03-02 by framps