framps/raspiBackup #850: "value too great for base" when trying to backup a GPT partitioned drive

ylexus, 2025-03-13:
I am using a 4TB NVMe SSD in my Raspberry Pi. It looks like raspiBackup is not able to parse the "type" output from `sfdisk -d` and fails:

```
20250313-221035 DBG 4016:                  *** sfdisk -d /dev/nvme0n1
20250313-221035 DBG 4016:                      label: gpt
20250313-221035 DBG 4016:                      label-id: B070687D-B700-440E-9D85-FBB8A9F48E47
20250313-221035 DBG 4016:                      device: /dev/nvme0n1
20250313-221035 DBG 4016:                      unit: sectors
20250313-221035 DBG 4016:                      first-lba: 34
20250313-221035 DBG 4016:                      last-lba: 7814037134
20250313-221035 DBG 4016:                      sector-size: 512
20250313-221035 DBG 4016:                      
20250313-221035 DBG 4016:                      /dev/nvme0n1p1 : start=        8192, size=     1048576, type=EBD0A0A2-B9E5-4433-87C0-68B6B72699C7, uuid=A1D7A9AA-D2CD-4A61-A709-096793B25EEC, name="bootfs"
20250313-221035 DBG 4016:                      /dev/nvme0n1p2 : start=     1056768, size=   268435456, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=F768D69C-0A9D-4DB3-B08D-710F3FDA11A6, name="rootfs"
20250313-221035 DBG 4016:                      /dev/nvme0n1p3 : start=   269492224, size=  7544543232, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=CE30B17F-4FB4-414C-BA13-37E26E214131, name="@HOSTNAME@"
20250313-221035 DBG 4018:              --- DeviceSize: 7814037168

...

20250313-221035 DBG 7312:          --> collectPartitions 
20250313-221035 DBG 7329:              --- PARTITIONS_TO_BACKUP - 1: 1 2
20250313-221035 DBG 7342:              --- backupAllPartitions: 0
20250313-221035 DBG 7351:              --- partition: /dev/nvme0n1p1 - size: 1048576 - type: EBD0A0A2-B9E5-4433-87C0-68B6B72699C7
/usr/local/bin/raspiBackup: line 7352: ((: EBD0A0A2-B9E5-4433-87C0: value too great for base (error token is "87C0")
20250313-221035 DBG 7351:              --- partition: /dev/nvme0n1p2 - size: 268435456 - type: 0FC63DAF-8483-4772-8E79-3D69D8477DE4
/usr/local/bin/raspiBackup: line 7352: ((: 0FC63DAF: value too great for base (error token is "0FC63DAF")
20250313-221035 DBG 7351:              --- partition: /dev/nvme0n1p3 - size: 7544543232 - type: 0FC63DAF-8483-4772-8E79-3D69D8477DE4
/usr/local/bin/raspiBackup: line 7352: ((: 0FC63DAF: value too great for base (error token is "0FC63DAF")
20250313-221035 DBG 7386:              --- PARTITIONS_TO_BACKUP - 2: 1 2
```

I suppose GPT partitions are not supported, but it is also does not seem to be difficult to upgrade the script. Perhaps this line:
```bash
if (( $type != 5 && $type != 85 && $size > 0 )); then # skip empty and extended partitions
```
could be replaced with
```bash
if [[ "$type" != "5" && "$type" != "85" && "$size" -gt 0 ]]; then  # skip empty and extended partitions
```
to handle GTP partition types and then add more conditions for GPT partition types that need to be skipped?

Full log: [raspiBackup.log](https://github.com/user-attachments/files/19237663/raspiBackup.log)

framps, 2025-03-13:
It's funny 😄 I just faced the same issue today with my 4TB disk when I tried to move my regression tests to be executed on a RPi5. Root cause is - as you already identified correctly - the system uses a gpt partitioning and no dos partitioning.

And you're right - the issue is the id which is no longer a number but a string and causes raspiBackup to fail. I updated the code to accept the string but there are other issues which popped up. So it's not a low hanging fruit to support gpt partitionign with raspiBackup - unfortunately.

I created an [issue](https://github.com/framps/raspiBackup/issues/851) to get gpt support in raspiBackup. 

> I suppose GPT partitions are not supported

Yes. I missed to add a check for gpt. Actually RaspbianOS uses dos partitioning and that's why I wasn't aware of any issues with gpt partitioning.

framps, 2025-03-13:
gpt is not supported right now. There is already issue #374 about gpt support in raspiBackup

closed 2025-03-13 by framps