framps/raspiBackup #851: Support gpt

framps, 2025-03-13:


ylexus, 2025-03-13:
Funnily, you already had https://github.com/framps/raspiBackup/issues/374 :)

framps, 2025-03-13:
Good catch 😄 This was in 2021 ... and I frankly forgot this issue. I will delete the new one and grab the old one from the backlog now.

closed 2025-03-13 by framps

ylexus, 2025-03-13:
I also found that the GPT partition sizes are not logged correctly. KiB need to be MiB and MiB need to be GiB

```
--- RBK0092I: Saving partition nvme0n1p1 (76.16 KiB/510.98 KiB)
--- RBK0158I: Creating native tgz backup /home/pi/bin/GoogleDrive/Root/Backups/rpi/storage/tmp/storage@debian12-tgz-backup-20250313-224651/nvme0n1p1.tgz
--- RBK0085I: Backup of type tgz started. Please be patient
--- RBK0112I: Partition nvme0n1p1 was saved
--- RBK0092I: Saving partition nvme0n1p2 (80.83 MiB/118.51 MiB)
--- RBK0158I: Creating native tgz backup /home/pi/bin/GoogleDrive/Root/Backups/rpi/storage/tmp/storage@debian12-tgz-backup-20250313-224651/nvme0n1p2.tgz
--- RBK0085I: Backup of type tgz started. Please be patient

```

```
Filesystem      1K-blocks       Used  Available Use% Mounted on
/dev/nvme0n1p1     523244      77996     445248  15% /boot/firmware
/dev/nvme0n1p3 3711959104 2180968888 1342360252  62% /home/data/storage
```

framps, 2025-03-13:
Collecting and parsing of the partitioning information for the restore does not work correctly for gpt because gpt is not supported as of now.

raspiBackup started when the usual SD size was about 8GB. Later on USB boot was available which allowed to use HDs/SSDs of about 1-2 GB. Now you can buy HDs and SSDs with > 2GB which requires gpt. Right now RaspbianOS just uses dos partitioning and you're the first one asks for gpt support in raspiBackup - even it's not officially used. So this issue from 2021 now gets more attention to be implemented in next release 😉

ylexus, 2025-03-13:
Yes I had to manually partition the drive and rsync my data from the older 2TB drive manually as the raspiOS own "SD card copier" app does not work with drives larger than 2TB - it just fails. 
This particular raspberry pi is a storage server so it needs a large drive. I suppose it's not a typical application for a raspberry pi.

ylexus, 2025-03-14:
Nice to know, thanks.

ylexus, 2025-03-14:
It's a bit limited, it can only deal with a single root partition. In my case I have two: root and storage. Ironically, it's done like that exclusively for raspiBackup to be able to only backup the operating system and not terabytes of storage data.