framps/raspiBackup #852: Issue after upgrade from 0.6.9.1 to 0.7.0.1

heg79, 2025-03-14:
Hi, Backup worked fine in 0.6.9.1 and since I upgraded to 0.7.0.1 I always get following error message:

--- RBK0009I: raspberrypi-az-display: raspiBackup.sh V0.7.0.1 - 2025-03-02 (6bdac60) Fr Mär 14 16:47:57 CET 2025 gestartet
--- RBK0262I: Dynamischer mount von /home/pi/BackupDrive/share nicht ausgeführt da es schon gemounted ist
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt
??? RBK0328E: Bootgerät /dev/mmcblk0p1 ist nicht gemounted
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0320I: Unvollständiges Backup wird gelöscht. Das kann etwas dauern. Bitte Geduld
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen
--- RBK0010I: raspberrypi-az-display: raspiBackup.sh V0.7.0.1 - 2025-03-02 (6bdac60) Fr Mär 14 16:48:01 CET 2025 beendet mit Returncode 138
--- RBK0167I: Eine eMail wird versendet

fstab looks like that:
proc            /proc           proc    defaults          0       0
/dev/mmcblk0p6  /boot           vfat    defaults          0       2
/dev/mmcblk0p7  /               ext4    defaults,noatime  0       1

I did not change anything but also cannot find mmcblk0p1 in the raspibackup config file.

Thx

framps, 2025-03-14:
Well, raspiBackup is supposed to have /boot on /dev/mmcblk0p1 and / on dev/mmcblk0p2. You run the OS from partition 6 and 7. That's actually not supported. But it's interesting it worked with 0.6.9.1. Please attach the debug log. Maybe it's a low hanging fruit.

heg79, 2025-03-14:
[backup.log](https://github.com/user-attachments/files/19254052/backup.log)

framps, 2025-03-15:
It looks like you're a lucky guy: 😉 

Please download a patched version with
`curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/
raspiBackupDownloadFromGit.sh | bash -s -- m_852 raspiBackup.sh`

and test whether is works now for you.

But please note: Your environment is not tested by me and thus not supported. Please also test carefully whether the restore works for your environment with the new release.

heg79, 2025-03-15:
sorry, does not work:

--- RBK0009I: raspberrypi-az-display: raspiBackup.sh V0.7.0.1-hotfix-m_852 - 2025-03-15 (b54d8c0) Sa Mär 15 19:57:43 CET 2025 gestartet
!!! RBK0173W: =========> HINWEIS <========= 
!!! RBK0173W: Dieses ist ein temporärer Hotfix der auf die nächste Version upgraded werden muss sobald eine verfügbar ist. 
!!! RBK0173W: =========> HINWEIS <=========
--- RBK0262I: Dynamischer mount von /home/pi/BackupDrive/share nicht ausgeführt da es schon gemounted ist
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt
??? RBK0328E: Bootgerät /dev/mmcblk0p1 ist nicht gemounted
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0320I: Unvollständiges Backup wird gelöscht. Das kann etwas dauern. Bitte Geduld
!!! RBK0212W: Backup endete mit Warnungen. Siehe vorhergehende Warnmeldungen
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen
--- RBK0010I: raspberrypi-az-display: raspiBackup.sh V0.7.0.1-hotfix-m_852 - 2025-03-15 (b54d8c0) Sa Mär 15 19:57:50 CET 2025 beendet mit Returncode 138
--- RBK0167I: Eine eMail wird versendet
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert

framps, 2025-03-15:
The only source for me to check what's going wrong is the debug log.

heg79, 2025-03-15:
Sorry ...

[20250315.txt](https://github.com/user-attachments/files/19265627/20250315.txt)

framps, 2025-03-15:
Hm ... that's strange. The message reports the wrong device. I missed to update the message. So the check shpould be executed against /dev/mmcblk0p6.

I added some additional debug code. Try again ... and don't forget to attach the debug log 😉

heg79, 2025-03-15:
[202503152.txt](https://github.com/user-attachments/files/19265851/202503152.txt)

thx a lot!

framps, 2025-03-15:
I have no clue which locale you use but the output of findmnt looks strange. Now the check is executed in a different way. Try again.

heg79, 2025-03-15:
It is normal RaspberryOS, the only "special" is, that I had to restore it about 6m ago. Now it looks good and the backup is running!

framps, 2025-03-15:
Again: Test the restored backup carefully. Your environment is not supported and you're a lucky guy raspiBackup handles your unsupported environment and the issue is a low hanging fruit ...

If I don't find any issue with the supported environments (I frankly don't expect this) the fix will roll in in next release.

heg79, 2025-03-15:
> framps

THANK YOU!!