framps/raspiBackup #858: Regular backup fails to execute

Loggsie, 2025-03-20:
Debian 12 - RPi 5 (8GB)

Running the latest script and raspiBackup version.

Manual backups complete successfully but automated backup fails to run and no automated backup occurs.

Have uninstalled, reinstalled and reconfigured raspiBackup and the outcome is the same.

framps, 2025-03-20:
That's bad 😢  Does raspiBackup not start at all in the background? Please check whether there exists raspiBackup.log in `/root` and has a current creation date. If it exists please attach the debug log.

If it doesn't exist please provide following output:

```
systemctl status raspiBackup.service
systemctl status raspiBackup.timer
```

Loggsie, 2025-03-21:
Solved!

I checked the root log and located an RC1 I/O error.

I use an NTFS volume for interoperability with Windows. Repaired the disk on Windows and the backup is running successfully as scheduled.

I appreciate the prompt response 😀

Next time I will check the logs first 😜

closed 2025-03-21 by Loggsie