framps/raspiBackup #861: Backup fails if /boot is not in the first two partitions

Pete-C2, 2025-03-26:
With an old pi, the following partitions were created (not deliberately by me):
lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.4G  0 disk 
├─mmcblk0p1 179:1    0  1.6G  0 part 
├─mmcblk0p2 179:2    0    1K  0 part 
├─mmcblk0p5 179:5    0   32M  0 part /media/pi/SETTINGS
├─mmcblk0p6 179:6    0   69M  0 part /boot
└─mmcblk0p7 179:7    0 12.8G  0 part /

When I run the backup it fails:
--- RBK0009I: oven: raspiBackup.sh V0.7.0.1 - 2025-03-02 (6bdac60) started at Wed 26 Mar 10:39:25 GMT 2025
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available
??? RBK0328E: Boot device /dev/mmcblk0p1 not mounted
--- RBK0033I: Please wait until cleanup has finished
--- RBK0320I: Removing incomplete backup. This may take some time. Please be patient
??? RBK0005E: Backup failed. Check previous error messages for details
--- RBK0010I: oven: raspiBackup.sh V0.7.0.1 - 2025-03-02 (6bdac60) stopped at Wed 26 Mar 10:39:37 GMT 2025 with rc 138
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log

It looks like there is an assumption that /boot must be at partition 1

framps, 2025-03-26:
Please see https://github.com/framps/raspiBackup/issues/852

The fix is planned for next release 0.7.0.2. But I will cherry pick the fix this evening for you ;-)

But please note: NOOBs or PINN Images are not officially supported. But I learned in the referenced issue raspiBackup  can handle these images.

framps, 2025-03-26:
I just cherry-picked the fix. Please update your local version with `sudo raspiBackup -U -S` and test.

Pete-C2, 2025-03-27:
I've upgraded and started and it is backing up the three mounted partitions (5, 6 & 7). Interestingly it's used rsync instead of an image for the boot partition - hopefully that will be fine when I test a restore. It might take a while before I can verify it has fully worked as it is backing up over a Wi-Fi connection as this device has no wired ethernet connection!
There is a warning "RBK0324W: Not all OS partitions saved. Backup will not boot" - I guess that is because it doesn't know where the boot partition is.
Thank you.

framps, 2025-03-27:
> I guess that is because it doesn't know where the boot partition is.

Yes. As I already wrote: raspiBackup expects a RaspbianOS  which has /boot on partition 1 and / on partition 2. RBK0324W is written because the second partition is not included in the backup. As far as I can see this will not create any boot issue for you. 

> Interestingly it's used rsync instead of an image for the boot partition - hopefully that will be fine when I test a restore.

I don't expect any issue on this.

> It might take a while before I can verify it has fully worked

Take your time. But please let me know your test result. Even your partition layout is not officially supported it's interesting to know whether it works.

Pete-C2, 2025-03-30:
It's not worked (at least it doesn't boot when recovered).
I only backed up partitions 5, 6 & 7 - they were the only ones mounted as per my creation of this "bug"/ticket. Only those partitions are saved on my NFS share. Yet on restore I have:

peter@raspberrypi:~ $ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda           8:0    1   29G  0 disk 
->sda1        8:1    1  1.6G  0 part /media/peter/bootfs
->sda2        8:2    1    1K  0 part 
->sda5        8:5    1   32M  0 part 
->sda6        8:6    1   69M  0 part /media/peter/boot
->sda7        8:7    1 27.3G  0 part /media/peter/root
(-> is my addition as the format of lsblk didn't paste well)

sda1 is not empty either, so where did sda1 and sda2 come from? I wiped all partitions from the recovery SD card using a Windows PC before starting the restore. sda1 appears to be a newer version of sda6 - it has similar files but includes some for newer models (e.g. Pi 5) that were never on NOOBS. I suspect the answer is that an assumption is made to restore sda1 and sda2 from the current operating system even though it was not asked for.

raspiBackup would not allow (I think} mmcblk0p1 andmmcblk0p2 to be backed up from the original system (it was giving me an error). Possibly I should manually mount them and have backed them up?

The contents of sda6 appear to match the original mmcblk0p6 (using ls-l to compare folders and files they appear to have same file sizes and dates, albeit one notable difference is that on the original all the files are owned by root with execute permissions and on the restored card they are owned by peter and don't have execute permissions)

The contents of sda7 appear to match the original mmcblk0p7 (including owners and file permissions)

This was my restore command:
sudo raspiBackup.sh -d /dev/sda -P -T "5 6 7" -g /backup/oven/oven@raspbian9-rsync-backup-20250327-092121/

I had one error (probably bc was not installed with raspiBackup as the repository location had been moved, but the backup claims to have finished successfully. Extracts from the log:
RBK1004E: bc not found. Please install bc first with with 'sudo apt-get install bc'.
RBK0017I: Backup finished successfully

The backup was on NOOBS on a Raspberry Pi Zero W and the restore on Bookworm on a Raspberry Pi 3. I also plugged the recovered card into the Raspberry Pi 3 (I assumed that would work although my attempt to plug card running on a Pi 5 into a Pi 3 failed, I rather thought that it would work being as it was comparable or newer).

framps, 2025-03-30:
It's impossible for me to answer your questions without any debug log. If you attach the backup debug log and also the restore debug log I will be able to see what's going on on your system.

Pete-C2, 2025-03-30:
Here;s the backup log:
[raspiBackup.log](https://github.com/user-attachments/files/19526034/raspiBackup.log)

And the restore log (filename changed as .logr file was not acceptable to Github!):

[raspiBackup-restore.log](https://github.com/user-attachments/files/19526050/raspiBackup-restore.log)

I appreciate this is potentially unsupported, but would love to get this working - in the past few days I've backed up four systems (can't prove one works yet as I only have one Pi 5) but this is wonderful software for performing a vital taks.

framps, 2025-03-30:
> I appreciate this is potentially unsupported, but would love to get this working

It's unsupported but as long as it's easy for me to get it to work I'll get it to work ;-)

I'm not familiar with NOOBs but as far as I understand you also need the first two partitions in order to get a NOOBs system to boot.  Please create a -P backup with all partitions and try the restore.

Pete-C2, 2025-04-01:
Success now, with some caveats.
I was able to backup partitions 1,5,6,7. Partition 2 is extended and raspiBackup fails if that is included.
I restored and selected partitions 1,2,5,6,7. I wasn't sure if I should select partition 2 or not. It created the correct partition table including the extended partitions and the result booted.

The caveats:

- The backup messages said it had failed. It looks like this was caused by an attempt to mount sda2 even though it is an extended partition. So can probably be ignored (might need to consider a warning for anyone restoring a NOOBS system?). Log file attached just in case you want to look at it, but not requiring anything to be done.
- The IP address was not obeyed (although it was correct in dhcpcd.conf). Sort of handy as otherwise I would have ended up with duplicate IP's on the network. This is clearly NOT to do with raspiBackup - not sure of the cause but it's minor and might work if I put the SD card in identical hardware as the original system.

[raspiBackup-restore 2.log](https://github.com/user-attachments/files/19549653/raspiBackup-restore.2.log)

framps, 2025-04-01:
Congratulations 👍 Glad to hear you can create and restore a backup with raspiBackup even NOOBs is not supported.

> The backup messages said it had failed.

That's because you asked for partition 2 to be included. I thought the second partition is also required by NOOBs. You should try to backup partitions 1,5,6 and 7 only. I think the error then will be gone. if not please provide the backup log and I will check whether this can be fixed easily.

> might work if I put the SD card in identical hardware as the original system.

Yes, IPs are reserved for a given time frame for MACs and you will get the same IP again if you start the restored backup on the original HW.

Pete-C2, 2025-04-01:
So if I don't ask for partition 2 to be recovered then it all works without error (and still creates the extended partition 2 anyway to match the original SD card). Perfect thanks.

framps, 2025-04-01:
Perfect. So we can close this issue now?

Pete-C2, 2025-04-01:
Yes, I'll close it - just wanted to wait to make sure you didn't ask for anything more.

closed 2025-04-01 by Pete-C2