framps/raspiBackup #865: There are no services stopped before starting the backup

remirr, 2025-04-03:
Hi there, 

I followed your youtube tutorial, thank you!

I went through the configuration and stopped all the services using the flow - I just went through randomly because I wasn't sure which order I should do it in - i.e. I stopped docker first, then pihole etc... I then selected which order to restart them all in, it jumped to the end before I could select the last one CUPS Browser so I just ignored that. When I went to the end and hit finish, I get the following error "There are no services stopped before starting the backup."

I am not sure what to do?

Thank you!

[image]

framps, 2025-04-03:
When you're done with the configuration of the services you have to confirm the configuration should be updated.  Sounds to me you didn't confirm this. 

[image]

If you confirmed the update please attach the debug log for further analysis.

remirr, 2025-04-03:
Strange, the confirm screen never shows up, at the last one it just goes back to the main menu. 

How do I get the logs?

[image]

remirr, 2025-04-03:
Here it is - the only thing I removed were the email lines

`20250403-112043: DBG: 5267 -  raspiBackupInstallUI.sh 0.4.8.3, 2025-03-26/21:27:23 - ec5bd0f
20250403-112043: DBG: 5270 -  +================================================================================================================================================+
20250403-112043: DBG: 5271 -  | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20250403-112043: DBG: 5272 -  +================================================================================================================================================+
20250403-112043: DBG: 5274 -  whiptail version: whiptail (newt): 0.52.21
20250403-112043: DBG: 5275 -  SYSTEMD_DETECTED: 1
20250403-112043: DBG: 1627 -  Checking for /usr/local/bin
20250403-112043: DBG: 1627 -  Checking for /usr/local/etc
20250403-112043: DBG: 1627 -  Checking for /etc/cron.d
20250403-112043: DBG: 4994 -  -> uiInstall 
20250403-112043: DBG: 2300 -     -> parseConfig 
20250403-112043: DBG: 2314 -        CONFIG_BACKUPPATH=/backup
20250403-112043: DBG: 2314 -        CONFIG_BACKUPTYPE=rsync
20250403-112043: DBG: 2314 -        CONFIG_ZIP_BACKUP=0
20250403-112043: DBG: 2314 -        CONFIG_KEEPBACKUPS=3
20250403-112043: DBG: 2314 -        CONFIG_SMART_RECYCLE=0
20250403-112043: DBG: 2314 -        CONFIG_SMART_RECYCLE_DRYRUN=1
20250403-112043: DBG: 2314 -        CONFIG_SMART_RECYCLE_OPTIONS=7 4 12 1
20250403-112043: DBG: 2314 -        CONFIG_STOPSERVICES=
20250403-112043: DBG: 2314 -        CONFIG_STARTSERVICES=
20250403-112043: DBG: 2314 -        CONFIG_EMAIL=
20250403-112043: DBG: 2314 -        CONFIG_MAIL_PROGRAM=mail
20250403-112043: DBG: 2314 -        CONFIG_MSG_LEVEL=0
20250403-112043: DBG: 2314 -        CONFIG_LANGUAGE=EN
20250403-112043: DBG: 2314 -        CONFIG_PARTITIONBASED_BACKUP=0
20250403-112043: DBG: 2314 -        CONFIG_PARTITIONS_TO_BACKUP=1 2
20250403-112043: DBG: 2321 -     <- parseConfig 
20250403-112043: DBG: 1741 -     -> isInternetAvailable 
20250403-112044: DBG: 1745 -     <- isInternetAvailable 0
20250403-112044: DBG: 4797 -     -> downloadPropertiesFile_do 
20250403-112044: DBG: 4829 -        -> shouldRenewDownloadPropertiesFile 
20250403-112044: DBG: 4843 -           20250403 : 00000000
20250403-112044: DBG: 4852 -        <- shouldRenewDownloadPropertiesFile 0
20250403-112044: DBG: 3949 -        -> progressbar_do 
20250403-112044: DBG: 4808 -           -> downloadPropertiesFile_execute 
MSG: Downloading version information.
20250403-112044: DBG: 1992 -              -> downloadFile URL: https://www.linux-tips-and-tricks.de/raspiBackup/raspiBackup.properties, file: /tmp/raspiBackupInstallUI.properties
20250403-112044: DBG: 1998 -                 httpCode: 200 RC: 0
20250403-112044: DBG: 2011 -              <- downloadFile 0
20250403-112044: DBG: 4823 -           <- downloadPropertiesFile_execute 1
20250403-112046: DBG: 3995 -        <- progressbar_do 
20250403-112046: DBG: 4802 -     <- downloadPropertiesFile_do 
20250403-112046: DBG: 4764 -     -> parsePropertiesFile 
20250403-112046: DBG: 4772 -        Script properties: v: 0.7.0.2 i:  b: 
20250403-112046: DBG: 4780 -        Installer properties: v: 0.4.8.3 i:  b: 
20250403-112047: DBG: 4782 -     <- parsePropertiesFile 
20250403-112047: DBG: 4787 -     -> parseCurrentVersions 
20250403-112047: DBG: 4790 -        Current script version: 0.7.0.2
20250403-112047: DBG: 4791 -        Current installer version: 0.4.8.3
20250403-112047: DBG: 4792 -     <- parseCurrentVersions 
20250403-112047: DBG: 4047 -     -> isUpdatePossible 
20250403-112047: DBG: 4048 -        script: c:0.7.0.2 p:0.7.0.2
20250403-112047: DBG: 4705 -        -> isNewerVersion 0.7.0.2 0.7.0.2
20250403-112047: DBG: 4749 -        <- isNewerVersion isNewVersion 0.7.0.2 <-> 0.7.0.2 - RC: 1
20250403-112047: DBG: 4053 -        installer: c:0.4.8.3 p:0.4.8.3
20250403-112047: DBG: 4705 -        -> isNewerVersion 0.4.8.3 0.4.8.3
20250403-112047: DBG: 4749 -        <- isNewerVersion isNewVersion 0.4.8.3 <-> 0.4.8.3 - RC: 1
20250403-112047: DBG: 4058 -     <- isUpdatePossible 1
20250403-112224: DBG: 5062 -     M3
20250403-112224: DBG: 2894 -     -> config_menu 
20250403-112224: DBG: 2300 -        -> parseConfig 
20250403-112224: DBG: 2314 -           CONFIG_BACKUPPATH=/backup
20250403-112224: DBG: 2314 -           CONFIG_BACKUPTYPE=rsync
20250403-112224: DBG: 2314 -           CONFIG_ZIP_BACKUP=0
20250403-112224: DBG: 2314 -           CONFIG_KEEPBACKUPS=3
20250403-112224: DBG: 2314 -           CONFIG_SMART_RECYCLE=0
20250403-112224: DBG: 2314 -           CONFIG_SMART_RECYCLE_DRYRUN=1
20250403-112224: DBG: 2314 -           CONFIG_SMART_RECYCLE_OPTIONS=7 4 12 1
[... 704 lines cut ...]
20250403-144502: DBG: 4252 -                 <- systemd_timer_menu 0
20250403-144502: DBG: 4107 -              <- config_timer_menu 0
20250403-144507: DBG: 3038 -           <- config_menu 
20250403-144507: DBG: 4047 -           -> isUpdatePossible 
20250403-144507: DBG: 4048 -              script: c:0.7.0.2 p:0.7.0.2
20250403-144507: DBG: 4705 -              -> isNewerVersion 0.7.0.2 0.7.0.2
20250403-144507: DBG: 4749 -              <- isNewerVersion isNewVersion 0.7.0.2 <-> 0.7.0.2 - RC: 1
20250403-144507: DBG: 4053 -              installer: c:0.4.8.3 p:0.4.8.3
20250403-144507: DBG: 4705 -              -> isNewerVersion 0.4.8.3 0.4.8.3
[... cut ...]
20250403-144514: DBG: 4705 -                 -> isNewerVersion 0.4.8.3 0.4.8.3
20250403-144514: DBG: 4749 -                 <- isNewerVersion isNewVersion 0.4.8.3 <-> 0.4.8.3 - RC: 1
20250403-144514: DBG: 4058 -              <- isUpdatePossible 1
20250403-154045: DBG: 2850 -           -> cleanup 
20250403-154045: DBG: 2858 -              Signal: EXIT
20250403-154045: DBG: 2859 -              rc: 0
20250403-154045: DBG: 4974 -              Masquerading home directory name
20250403-154045: DBG: 4979 -              Masquerading hostname

framps, 2025-04-03:
I found 

```
20250403-114649: DBG: 3600 - Answer: "containerd" "cron" "cups-browsed" "cups" "docker" "nmbd" "smbd" "hd-idle" "lightdm" "packagekit" "pihole-FTL" "polkit" "rsync" "rtkit-daemon" "teamviewerd" "vncserver-x11d.service"
20250403-114649: DBG: 3621 - -> config_service_sequence_do containerd cron cups-browsed cups docker nmbd smbd hd-idle lightdm packagekit pihole-FTL polkit rsync rtkit-daemon teamviewerd vncserver-x11d.service
20250403-114653: DBG: 3651 - Answer: docker
20250403-114709: DBG: 3651 - Answer: containerd
20250403-114719: DBG: 3651 - Answer: pihole-FTL
20250403-114726: DBG: 3651 - Answer: teamviewerd
20250403-114734: DBG: 3651 - Answer: rsync
20250403-114741: DBG: 3651 - Answer: vncserver-x11d.service
20250403-114748: DBG: 3651 - Answer: rtkit-daemon
20250403-114755: DBG: 3651 - Answer: smbd
20250403-114800: DBG: 3651 - Answer: cron
20250403-114807: DBG: 3651 - Answer: polkit
20250403-114810: DBG: 3651 - Answer: packagekit
20250403-114814: DBG: 3651 - Answer: lightdm
20250403-114817: DBG: 3651 - Answer: nmbd
20250403-114819: DBG: 3651 - Answer: hd-idle
20250403-114821: DBG: 3651 - Answer: cups
20250403-114821: DBG: 3664 - <- config_service_sequence_do aborted
```
and as you can see the config service sequence was aborted. That's why you get the warning message.

> I then selected which order to restart them all in, it jumped to the end before I could select the last one CUPS Browser so I just ignored that.

As you can see in the log you answered cups. What do you mean exactly with `it jumped to the end before I could select the last one CUPS` ? As you can see cups was selected. The you have selected `cancel` instead of `ok` and therefor aborted the configuration. 

Please view [this configuration demo](https://www.linux-tips-and-tricks.de/images/raspiBackupInstall_de.gif) and see how the sequence configuration is finished with `OK`.

remirr, 2025-04-04:
So the issue is, it crashes before I get a chance to finish the config.

Also, do I need to select all of them? How do I know which order to select them in...

https://github.com/user-attachments/assets/3d269915-32b8-4db3-894f-e1dab6d1e63f

framps, 2025-04-04:
Thank you very much for the screen recording. It proves there is some bug - unfortunately 😟 

Unfortunately the debug log doesn't help.

> Also, do I need to select all of them?

No. According the log you have following services active:
```
containerd cron cups-browsed cups docker hd-idle lightdm nmbd packagekit pihole-FTL polkit rsync rtkit-daemon smbd teamviewerd vncserver-x11d.service
```
>  How do I know which order to select them in...

I would stop following service in following sequence (Note: I'm not experienced with docker)

1) cron
2) docker
3) containerd

Maybe pihole. It depends on the activity going on on your system. I would skip pihole first and check whether you get backup errors because of pihole.

I suggest to configure raspiBackup to stop these 3 services and test whether the configuration finishes without any issue. If it doesn't finish successful let me know and I have to enhance the installer to catch unexpected errors.

Note: If it's important for you you to create a backup right now: The installer just updates the raspiBackup configuration and you can manually add the required definitions to stop and start your 3 services and don't need the installer. The installer is just a convenience tool to setup an initial raspiBackup configuration.

But I appreciate if you help me to locate and fix the root cause of your issue. But this will take your time.

remirr, 2025-04-06:
Ok, I have gone through and tested multiple configurations and every time **'cups-browsed'** is in the list, it crashes. 

If I select every single service except 'cups-browsed' it works, as soon as that is in the list, it crashes because it gets to the confirmation showing in which order they will be stopped. 

**Works**
1. cron
2. docker
3. containerd
4. cups

**Crashes** 
1. cron
2. docker
3. containerd
4. cups
5. cups-browsed 

I hope this helps

framps, 2025-04-06:
> I hope this helps

👍 👍 👍 

Thank you very much you nailed down the root cause of this issue. Now I can reproduce the issue on my site and debug and fix the issue.

framps, 2025-04-06:
I just created a fix. 

@remirr Would you please verify the fix? Please use `curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_865 installation/raspiBackupInstallUI.sh` to download the fixed version.

If you successfully finish the verification please close the issue.

Again - thank you very much for your patience and help to locate the root cause for this issue 👍

remirr, 2025-04-06:
Yep, that worked! Thank you very much for the fast turnaround

closed 2025-04-06 by remirr

remirr, 2025-07-16:
Hi there, 
Sorry to come back to this - how do I check if the backup was successful and actually works?
Thank you!

framps, 2025-07-16:
1. Create a backup
2. Restore the backup on a different device
3. Start the restored Backup on a different device on a Raspberry
4. Test whether everything works fine