framps/raspiBackup #866: Raspi no longer boots after the restore. PARTUUID not correct

AndyZoIoT, 2025-04-06:
Raspi no longer boots after the restore, similar to issue: #842 
Here the error messages displayed (see screenshot).

[image]

Version
raspiBackup.sh V0.7.0.2

My system:
RaspberryPi4
Boot on SSD, backup on USBSTICK

Restore process:
New SSD (recognised as sda)
RaspiBackup V0.7.0.2 installed on a new SD card (mmcblk0), only for restore process.
Backup saved on USBSTICK (recognised as sdb), monted in /media/ssd/
commands used:
sudo raspiBackup.sh --updateUUIDs -g -d /dev/sda /media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250405-185423
or:
sudo raspiBackup.sh -g --resizeRootFS- -d /dev/sda /media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250405-224911

The backup has done successfully, also restore message seems to be a success, but removing the SD, no boot.
Probably UUID assigned is the one of the SD card, not the SSD one... but I'm not so skilled!

Enclosed the log raspiBackup.logr
[raspiBackup.logr.txt](https://github.com/user-attachments/files/19620752/raspiBackup.logr.txt)

Many thanks for the support you will provide, and thanks for the great tool you made.

Andyzo

framps, 2025-04-06:
Following two debug lines prove the two partitions were restored to /dev/sda1 and /dev/sda2

```
20250405-230650 DBG 8880:                  --> executeFilesystemCheck /dev/sda1
20250405-230650 MSG 8881:                      --- RBK0342I: Executing filesystem check on /dev/sda1 
fsck from util-linux 2.38.1
fsck.fat 4.2 (2021-01-31)
Checking we can access the last sector of the filesystem
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  67:f0/55, 68:11/6f, 69:64/f5, 70:90/4e
  Not automatically fixing this.
Boot sector contents:
System ID "mkfs.fat"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
      2048 bytes per cluster
        32 reserved sectors
First FAT starts at byte 16384 (sector 32)
         2 FATs, 32 bit entries
   1044480 bytes per FAT (= 2040 sectors)
Root directory start at cluster 2 (arbitrary size)
Data area starts at byte 2105344 (sector 4112)
    261115 data clusters (534763520 bytes)
63 sectors/track, 128 heads
      8192 hidden sectors
   1048572 sectors total
Reclaiming unconnected clusters.
Checking free cluster summary.
/dev/sda1: 392 files, 28080/261115 clusters
20250405-230650 DBG 8884:                      --- fsck rc: 0
20250405-230650 DBG 8889:                  <-- executeFilesystemCheck 
```

```
20250405-231602 DBG 8880:                  --> executeFilesystemCheck /dev/sda2
20250405-231602 MSG 8881:                      --- RBK0342I: Executing filesystem check on /dev/sda2 
fsck from util-linux 2.38.1

      187362 inodes used (3.86%, out of 4857856)
          59 non-contiguous files (0.0%)
          52 non-contiguous directories (0.0%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 179490/14
     1931072 blocks used (9.95%, out of 19405590)
           0 bad blocks
           1 large file

      161802 regular files
       17577 directories
           0 character device files
           0 block device files
           1 fifo
         358 links
        7973 symbolic links (7849 fast symbolic links)
           0 sockets
------------
      187711 files
```

The UUIDs were also updated in cmdline and fstab. So I don't see any error in the log.

The message /dev/mmcblk0p2 does not exist does not match with

```
20250405-231642 DBG 9129:                          *** blkid
20250405-231642 DBG 9129:                              /dev/mmcblk0p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="4EF5-6F55" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="d7e1cc98-01"
20250405-231642 DBG 9129:                              /dev/mmcblk0p2: LABEL="rootfs" UUID="ce208fd3-38a8-424a-87a2-cd44114eb820" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="d7e1cc98-02"
20250405-231642 DBG 9129:                              /dev/sdb1: LABEL="BKP" BLOCK_SIZE="512" UUID="88980AB0980A9CB6" TYPE="ntfs" PARTUUID="00698f70-01"
20250405-231642 DBG 9129:                              /dev/sda2: UUID="d17a889e-642d-4533-ac93-4d3e07e09939" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="74eb9306-02"
20250405-231642 DBG 9129:                              /dev/sda1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="51A9-DE82" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="74eb9306-01"
```
where you see /dev/sda2 was created and exists.

A common root cause of strange boot failures is a buggy SD card. I suggest to use a brand new SD card and test the restore again.

I also saw your comments on my website. I don't publish them. This git issue is the right way to handle your issue.

AndyZoIoT, 2025-04-07:
Many thanks Framp for your quick reply!

I’m so sorry if I was not clear in my explanation..

Correct, the two partitions were restored to /dev/sda1 and /dev/sda2...
but /dev/sda is not a SD Card, it is a SSD connected to USB.

And /dev/sdb is where the backup is saved, on an USB STICK.

The SD is used only to run the SO during the restore process, so it should not be relevant, isn’t it?

So I don’t explain myself why I receive the ALERT  “The message /dev/mmcblk0p2 does not exist” 
It seems that it search for the boot on SD, even if restored in SSD.
(of course, after restore finisched, I remove SD / USB Stick, only SSD still connected).

Looking forward for your help to restore the image, also suggesting different approach if any,
Thanks,
Andrea.

framps, 2025-04-07:
Ok. I didn't get your config right.

It sound's to me the SSD connected via USB cannot be discovered. Keep in mind a SSD requires a lot of current. Make sure you connect the SSD via an active USB hub to the Raspberry.

AndyZoIoT, 2025-04-07:
Sure, I did it, it works if I flash it with a blank rasbian whit Raspberry Pi Imager.

framps, 2025-04-07:
Just to make sure we're on the same page:

0) Whenever you use a SSD the SSD is connected via an active USB hub to your RPi4
1) You installed RaspianOS on SSD1 and can start RaspbianOS from SSD1.
2) You created a backup on your USB flash drive
3) You installed RaspbianOS on a SD card and restored the backup to SSD2
4) You removed all devices but kept SSD2 on the RPi4 and then you get the error

Please boot with SSD1 and plug in SSD2. then provide following info:

1) Execute command `sudo lsblk`
2) Execute command `sudo blkid`

AndyZoIoT, 2025-04-07:
Hi Framp,
1. SSD directly connected to USB3 of Raspberry Pi 4 (No issues on working system)
2. ,Corrects
3. Correct
4. Correct.

booted with SSD1 and plug in SSD2, then:

pi@RASPY4:~ $ sudo lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda      8:0    0  74,5G  0 disk
├─sda1   8:1    0   512M  0 part /boot/firmware
└─sda2   8:2    0    74G  0 part /
sdb      8:16   1   7,6G  0 disk
└─sdb1   8:17   1   7,6G  0 part /media/usbflash
sdc      8:32   0 238,5G  0 disk
├─sdc1   8:33   0   512M  0 part
└─sdc2   8:34   0    74G  0 part

pi@RASPY4:~ $ sudo blkid
/dev/sda2: UUID="ee038723-f333-4af1-86ee-eaeb6af33cb4" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="d29c81c1-02"
/dev/sda1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="9064-11F0" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="d29c81c1-01"
/dev/sdb1: LABEL="BKP" BLOCK_SIZE="512" UUID="88980AB0980A9CB6" TYPE="ntfs" PARTUUID="00698f70-01"
/dev/sdc2: UUID="d17a889e-642d-4533-ac93-4d3e07e09939" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="74eb9306-02"
/dev/sdc1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="51A9-DE82" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="74eb9306-01"

Thanks,
Andrea.

framps, 2025-04-07:
> 1. SSD directly connected to USB3 of Raspberry Pi 4 (No issues on working system)

What is about my statement (0)? Do you use all the time an active USB switch to connect your SSD to the RPi? If not that's the root cause of your issue.

Now execute `sudo mount /dev/sdc1 /mnt` and `sudo mount /dev/sdc2 /media`. And then `df -h`.

Next `cat /mnt/cmdline.txt` and `cat /media/etc/fstab`

AndyZoIoT, 2025-04-07:
Hi Framp,
even if it is more that 1 year that I run with this configuration without an active USB switch, I added one now for this test, to exclude problems related to supply.
Now, booted up with this configuration, here the results:

pi@RASPY4:~ $ sudo mount /dev/sdc1 /mnt
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
pi@RASPY4:~ $ systemctl daemon-reload
==== AUTHENTICATING FOR org.freedesktop.systemd1.reload-daemon ====
Autenticazione richiesta per ricaricare lo stato di systemd.
Authenticating as: ,,, (pi)
Password:
==== AUTHENTICATION COMPLETE ====

pi@RASPY4:~ $ sudo mount /dev/sdc2 /media
pi@RASPY4:~ $ cat /mnt/cmdline.txt
pi@RASPY4:~ $ cat /media/etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=d29c81c1-01  /boot/firmware  vfat    defaults          0       2
PARTUUID=d29c81c1-02  /               ext4    defaults,noatime  0       1

UUID=88980AB0980A9CB6 /media/usbflash ntfs auto,nofail,noatime,users,rw,uid=pi,gid=pi 0 0

As I can see, no answer for "cat /mnt/cmdline.txt"...

Thanks,
Andrea.

framps, 2025-04-07:
> As I can see, no answer for "cat /mnt/cmdline.txt"...

That's nasty because in the debug log I see cmdline.txt and /etc/fstab were updated correctly.

```
20250405-231641 MSG 9489:                          --- RBK0295I: Synchronizing /boot/firmware/cmdline.txt and /etc/fstab 
20250405-231641 DBG 9493:                          --- Org /mnt/raspiBackup/boot/cmdline.txt
20250405-231641 DBG 9494:                              *** cat /mnt/raspiBackup/boot/cmdline.txt
20250405-231641 DBG 9494:                                  console=serial0,115200 console=tty1 root=PARTUUID=d29c81c1-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=IT
20250405-231641 DBG 9500:                          --- CMDLINE - newPartUUID: 74eb9306-02, oldPartUUID: d29c81c1-02
20250405-231642 MSG 9504:                          --- RBK0102I: Updating PARTUUID from d29c81c1-02 to 74eb9306-02 in /boot/firmware/cmdline.txt 
20250405-231642 DBG 9543:                          --- Org /mnt/raspiBackup/root/etc/fstab
20250405-231642 DBG 9546:                          --- proc            /proc           proc    defaults          0       0
20250405-231642 DBG 9546:                          --- PARTUUID=d29c81c1-01  /boot/firmware  vfat    defaults          0       2
20250405-231642 DBG 9546:                          --- PARTUUID=d29c81c1-02  /               ext4    defaults,noatime  0       1
20250405-231642 DBG 9546:                          --- # a swapfile is not a swap partition, no line here
20250405-231642 DBG 9546:                          --- #   use  dphys-swapfile swap[on|off]  for that
20250405-231642 DBG 9546:                          --- UUID=88980AB0980A9CB6 /media/usbflash ntfs auto,nofail,noatime,users,rw,uid=pi,gid=pi 0 0
20250405-231642 DBG 9552:                          --- FSTAB root - newRootPartUUID: 74eb9306-02, oldRootPartUUID: d29c81c1-02
20250405-231642 MSG 9560:                          --- RBK0102I: Updating PARTUUID from d29c81c1-02 to 74eb9306-02 in /etc/fstab 
20250405-231642 DBG 9604:                          --- Org /mnt/raspiBackup/root/etc/fstab
20250405-231642 DBG 9605:                          --- proc            /proc           proc    defaults          0       0
20250405-231642 DBG 9605:                          --- PARTUUID=74eb9306-01  /boot/firmware  vfat    defaults          0       2
20250405-231642 DBG 9605:                          --- PARTUUID=74eb9306-02  /               ext4    defaults,noatime  0       1
20250405-231642 DBG 9605:                          --- # a swapfile is not a swap partition, no line here
20250405-231642 DBG 9605:                          --- #   use  dphys-swapfile swap[on|off]  for that
20250405-231642 DBG 9605:                          --- UUID=88980AB0980A9CB6 /media/usbflash ntfs auto,nofail,noatime,users,rw,uid=pi,gid=pi 0 0
20250405-231642 DBG 9611:                          --- FSTAB boot - newPartUUID: 74eb9306-01, oldPartUUID: 74eb9306-01
20250405-231642 DBG 9649:                          --- Upd /mnt/raspiBackup/boot/cmdline.txt
20250405-231642 DBG 9650:                              *** cat /mnt/raspiBackup/boot/cmdline.txt
20250405-231642 DBG 9650:                                  console=serial0,115200 console=tty1 root=PARTUUID=74eb9306-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=IT
20250405-231642 DBG 9654:                          --- Upd /mnt/raspiBackup/root/etc/fstab
20250405-231642 DBG 9655:                              *** cat /mnt/raspiBackup/root/etc/fstab
20250405-231642 DBG 9655:                                  proc            /proc           proc    defaults          0       0
20250405-231642 DBG 9655:                                  PARTUUID=74eb9306-01  /boot/firmware  vfat    defaults          0       2
20250405-231642 DBG 9655:                                  PARTUUID=74eb9306-02  /               ext4    defaults,noatime  0       1
20250405-231642 DBG 9655:                                  # a swapfile is not a swap partition, no line here
20250405-231642 DBG 9655:                                  #   use  dphys-swapfile swap[on|off]  for that
20250405-231642 DBG 9655:                                  UUID=88980AB0980A9CB6 /media/usbflash ntfs auto,nofail,noatime,users,rw,uid=pi,gid=pi 0 0
20250405-231642 DBG 9661:                      <-- synchronizeCmdlineAndfstab 
```

> even if it is more that 1 year that I run with this configuration without an active USB switch, I added one now for this test, to exclude problems related to supply.

Thank you very much. USB power is often the root cause of boot issues and now we know for sure it's not the power 😉 

I detected you asked in a comment on my website whether it's possible to restore a previous raspiBackup release. So you use raspiBackup for a longer time already. Please restore the previous release with option -V and test whether it's a regession.

AndyZoIoT, 2025-04-07:
Ok, I'll try with 0.6.9.1
To install that also to the SD (where it is not available, because fresh installed) how can I do?

framps, 2025-04-07:
Just copy the restored /usr/local/bin/raspiBackup.sh to your SD card into /usr/local/bin.

AndyZoIoT, 2025-04-07:
HI Framp, with 0.6.9.1 no errors restoring the image, all works fine!
No time today for other test, but I'll try also with 0.7.0-beta and 0.7.0.1
Andrea.

AndyZoIoT, 2025-04-07:
This is restore log about 0.6.9.1:

[raspiBackup0.6.9.1.logr.txt](https://github.com/user-attachments/files/19636846/raspiBackup0.6.9.1.logr.txt)

backup log has been probably overwritten... I'll provide later, but probably end of week when come back to home!

framps, 2025-04-07:
> probably end of week when come back to home!

Take your time. In the meantime I tried to reproduce your issue but failed 😒 . Looks like you have some special setup/environment which causes the issue.

AndyZoIoT, 2025-04-11:
Hi Framp,
here the backup logs:

[raspiBackupV0.6.9.1.log](https://github.com/user-attachments/files/19701205/raspiBackupV0.6.9.1.log)

[raspiBackup_v0.7.0.2.log](https://github.com/user-attachments/files/19701527/raspiBackup_v0.7.0.2.log)
hope this can helps.

Thanks for your time,
Andrea.

framps, 2025-04-11:
Thank you very much for the logs. Crossing fingers I'll find the root cause of your issue.

framps, 2025-04-12:
I checked the logs and noticed one difference: The IP masquerading work better with 0.7.0.2 than with 0.6.9.1. But that's not your issue :-( Other than that I didn't find anything at first glance which may be the root cause of you issue.

Then I thought it's maybe because you use a dedicated SD card for the restore and created a tar backup and restored the backup with the dedicated SD card. But the restored system booted successfully.

I unfortunately don't have two SSDs to be able to recreate exactly your environment and used just SD cards. But this shouldn't make any difference.

I frankly don't understand how this message: ALERT: /dev/mmcblk0p2 not found`

Can you show the output of `ls -lah /boot` of the restored device which creates this error message - most likely because of a missing cmdline.txt?

AndyZoIoT, 2025-04-13:
Hi Framp,
tried to restore again to USB SSD, booting from dedicated SD Card, dump saved on USB Flash.
- Removed the SD card, removed the USB Flash, no reply if I try  "ls -lah /boot" (of course, no SSD mounted, I mean).

Reboot from the dedicated SD Card, connected the SSD, here the result:

pi@RASPY4:~ $ ls -lah /boot
total 41M
drwxr-xr-x  3 root root 4.0K Apr  5 18:37 .
drwxr-xr-x 19 root root 4.0K Apr  5 22:37 ..
-rw-r--r--  1 root root   92 Nov 19 14:32 cmdline.txt
-rw-r--r--  1 root root 233K Oct  8  2024 config-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root 233K Oct  8  2024 config-6.6.51+rpt-rpi-v8
-rw-r--r--  1 root root   91 Nov 19 14:32 config.txt
drwxr-xr-x  4 root root 4.0K Jan  1  1970 firmware
-rw-r--r--  1 root root  12M Nov 19 14:39 initrd.img-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root  12M Nov 19 14:39 initrd.img-6.6.51+rpt-rpi-v8
lrwxrwxrwx  1 root root   18 Nov 19 14:39 issue.txt -> firmware/issue.txt
lrwxrwxrwx  1 root root   17 Nov 19 14:32 overlays -> firmware/overlays
-rw-r--r--  1 root root   83 Oct  8  2024 System.map-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root   83 Oct  8  2024 System.map-6.6.51+rpt-rpi-v8
-rw-r--r--  1 root root 8.9M Oct  8  2024 vmlinuz-6.6.51+rpt-rpi-2712
-rw-r--r--  1 root root 8.9M Oct  8  2024 vmlinuz-6.6.51+rpt-rpi-v8
pi@RASPY4:~ $

But, of course, cmdline.txt is the one of the SD Card, not the one of the SSD!

framps, 2025-04-13:
> But, of course, cmdline.txt is the one of the SD Card, not the one of the SSD!

I'm confused now 😕  According https://github.com/framps/raspiBackup/issues/866#issuecomment-2783487131 there shouldn't be a cmdline.txt on /boot partition of the restored SSD so I expected a missing cmdline.txt.

Did you copy the cmdline.txt from the SD card to the restored SSD? The restored SSD will not boot with the cmdline from the SD card.

Please execute following steps:
1) Boot from the SD card with no other device connected
2) Connect the restored SSD
3) Check with `lsblk` which device name is used by the restored SSD
4) Let's assume `/boot` is `/dev/sda1` and `/` is `/dev/sda2`. The execute `sudo mount /dev/sda1 /mnt`
5) Execute `la -la /mnt` (That's the info I need)

AndyZoIoT, 2025-04-13:
Here the results:

pi@RASPY4:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0 238.5G  0 disk
├─sda1        8:1    0   512M  0 part
└─sda2        8:2    0    74G  0 part
mmcblk0     179:0    0  14.8G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  14.3G  0 part /

and then:

pi@RASPY4:~ $ l**s** -la /mnt
total 54794
drwxr-xr-x  3 root root    4096 Jan  1  1970 .
drwxr-xr-x 19 root root    4096 Apr  5 22:37 ..
-rwxr-xr-x  1 root root   32519 Jan 27 18:19 bcm2710-rpi-2-b.dtb
-rwxr-xr-x  1 root root   34711 Jan 27 18:19 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root   35346 Jan 27 18:19 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root   32298 Jan 27 18:19 bcm2710-rpi-cm3.dtb
-rwxr-xr-x  1 root root   33688 Jan 27 18:19 bcm2710-rpi-zero-2.dtb
-rwxr-xr-x  1 root root   33688 Jan 27 18:19 bcm2710-rpi-zero-2-w.dtb
-rwxr-xr-x  1 root root   56112 Jan 27 18:19 bcm2711-rpi-400.dtb
-rwxr-xr-x  1 root root   56108 Jan 27 18:19 bcm2711-rpi-4-b.dtb
-rwxr-xr-x  1 root root   56654 Jan 27 18:19 bcm2711-rpi-cm4.dtb
-rwxr-xr-x  1 root root   38357 Jan 27 18:19 bcm2711-rpi-cm4-io.dtb
-rwxr-xr-x  1 root root   53395 Jan 27 18:19 bcm2711-rpi-cm4s.dtb
-rwxr-xr-x  1 root root   81167 Jan 27 18:19 bcm2712d0-rpi-5-b.dtb
-rwxr-xr-x  1 root root   81119 Jan 27 18:19 bcm2712-rpi-500.dtb
-rwxr-xr-x  1 root root   81163 Jan 27 18:19 bcm2712-rpi-5-b.dtb
-rwxr-xr-x  1 root root   81620 Jan 27 18:19 bcm2712-rpi-cm5-cm4io.dtb
-rwxr-xr-x  1 root root   81682 Jan 27 18:19 bcm2712-rpi-cm5-cm5io.dtb
-rwxr-xr-x  1 root root   81661 Jan 27 18:19 bcm2712-rpi-cm5l-cm4io.dtb
-rwxr-xr-x  1 root root   81723 Jan 27 18:19 bcm2712-rpi-cm5l-cm5io.dtb
-rwxr-xr-x  1 root root   52476 Mar  7 16:50 bootcode.bin
-rwxr-xr-x  1 root root     131 Apr 13 16:27 **cmdline.txt**
-rwxr-xr-x  1 root root    1312 Jan 11 09:37 config.txt
-rwxr-xr-x  1 root root    1312 Jan 11 09:37 config.txty
-rwxr-xr-x  1 root root    3231 Mar  7 16:50 fixup4cd.dat
-rwxr-xr-x  1 root root    5459 Mar  7 16:50 fixup4.dat
-rwxr-xr-x  1 root root    8455 Mar  7 16:50 fixup4db.dat
-rwxr-xr-x  1 root root    8453 Mar  7 16:50 fixup4x.dat
-rwxr-xr-x  1 root root    3231 Mar  7 16:50 fixup_cd.dat
-rwxr-xr-x  1 root root    7328 Mar  7 16:50 fixup.dat
-rwxr-xr-x  1 root root   10298 Mar  7 16:50 fixup_db.dat
-rwxr-xr-x  1 root root   10296 Mar  7 16:50 fixup_x.dat
-rwxr-xr-x  1 root root 6968461 Mar 17 09:59 initramfs_2712
-rwxr-xr-x  1 root root 6968401 Mar 17 09:59 initramfs8
-rwxr-xr-x  1 root root     145 Nov 19 14:39 issue.txt
-rwxr-xr-x  1 root root 9360385 Mar  7 16:50 kernel_2712.img
-rwxr-xr-x  1 root root 9362120 Mar  7 16:50 kernel8.img
-rwxr-xr-x  1 root root    1594 Mar  7 16:50 LICENCE.broadcom
drwxr-xr-x  2 root root   30720 Mar  3 14:42 overlays
-rwxr-xr-x  1 root root  813596 Mar  7 16:50 start4cd.elf
-rwxr-xr-x  1 root root 3761768 Mar  7 16:50 start4db.elf
-rwxr-xr-x  1 root root 2263328 Mar  7 16:50 start4.elf
-rwxr-xr-x  1 root root 3010984 Mar  7 16:50 start4x.elf
-rwxr-xr-x  1 root root  813596 Mar  7 16:50 start_cd.elf
-rwxr-xr-x  1 root root 4833736 Mar  7 16:50 start_db.elf
-rwxr-xr-x  1 root root 2987424 Mar  7 16:50 start.elf
-rwxr-xr-x  1 root root 3734632 Mar  7 16:50 start_x.elf
pi@RASPY4:~ $
and the file cmdline.txt seems empty:

[cmdline.txt](https://github.com/user-attachments/files/19725833/cmdline.txt)

framps, 2025-04-13:
> and the file cmdline.txt seems empty:

No. There are 131 * 0x00 😉 

```
xxd cmdline.txt 
00000000: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000010: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000020: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000030: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000040: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000050: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000070: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000080: 0000 00                                  ...
```
I have no clue how this can happen.

I'll check the logs again given this update.

framps, 2025-04-14:
First of all let's double check the correct cmdline.txt is included in the backup. According the log it should.

Execute following commands on the SD system when the backup USB is connected:

```
sudo losetup /dev/loop0 <backupdirector>/raspberrypi-bookworm64-lite-regression-backup.img
sudo mount /dev/loop0 /mnt
cat /mnt/cmdline.txt
sudo umount /mnt
sudo losetup -D
```

Next I created an updated raspiBackup version. Please use this to restore the backup and provide the debug log. Please download the updated version with 
```curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_866 raspiBackup.sh
curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_866 raspiBackup.sh
--- Downloading raspiBackup.sh from git branch m_866 into current directory ...
--- Download finished successfully
--- Retrieving commit meta data of raspiBackup.sh from m_866 ...
--- Inserting commit meta data into downloaded file raspiBackup.sh ...
--- Version: 0.7.0.2-m_866 CommitSHA: 6da3692 CommitDate: 2025-04-14 CommitTime: 16:56:10Z
--- Start raspiBackup.sh with `sudo ./raspiBackup.sh` now
```
Note the leading period to call the updated raspiBackup version !

Next please execute `curl -s https://raw.githubusercontent.com/framps/raspberryTools/master/invokeTool.sh | sudo bash -s -- syncUUIDs.sh -u /dev/<yourBackupSSD>` and check whether the UUIDs used in cmdline.txt and fstab match the backup SSD device. According the logs this should match.

AndyZoIoT, 2025-04-14:
pls give more guidance about:
sudo losetup /dev/loop0 <backupdirectory>/raspberrypi-bookworm64-lite-regression-backup.img
......
The backup is stored on USB Flash, mounted on: /media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932/

I'll be back at home in 3 days.

framps, 2025-04-14:
> I'll be back at home in 3 days.

Take your time. I'll be in China for 4 weeks starting next Tuesday :-)

> sudo losetup /dev/loop0 /raspberrypi-bookworm64-lite-regression-backup.img

> /media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932/

You have to cd into `/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932/` and use the filename which ends with .img into this directory in the losetup command.

AndyZoIoT, 2025-04-15:
Sorry if I'm so inexpert....

pi@RASPY4:~ $ sudo mount /dev/sdb1 /media/ssd -o uid=pi,gid=pi 
pi@RASPY4:~ $ cd /media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932/      
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ sudo losetup /dev/loop0 /raspberrypi-bookworm64-lite-regression-backup.img
losetup: /raspberrypi-bookworm64-lite-regression-backup.img: failed to set up loop device: No such file or directory
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ sudo losetup /dev/loop0 RASPY4-backup.img
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ sudo mount /dev/loop0 /mnt
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ cat /mnt/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=d29c81c1-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=ITpi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ sudo umount /mnt
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ sudo losetup -D
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ curl -s https://raw.githubusercontent.com/framps/raspiBackup/master/scripts/raspiBackupDownloadFromGit.sh | bash -s -- m_866 raspiBackup.sh
--- Downloading raspiBackup.sh from git branch m_866 into current directory ...
--- Download finished successfully
--- Retrieving commit meta data of raspiBackup.sh from m_866 ...
--- Inserting commit meta data into downloaded file raspiBackup.sh ...
--- Version: 0.7.0.2-m_866 CommitSHA: 6da3692 CommitDate: 2025-04-14 CommitTime: 16:56:10Z
--- Start raspiBackup.sh with `sudo ./raspiBackup.sh` now
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $ sudo ./raspiBackup.sh

--- RBK0009I: RASPY4: raspiBackup.sh V0.7.0.2-m_866 - 2025-04-14 (6da3692) started at Tue 15 Apr 17:56:43 CEST 2025
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available
--- RBK0151I: Using backuppath /backup with partition type
??? RBK0027E: No external device mounted on /backup. root partition would be used for the backup
--- RBK0033I: Please wait until cleanup has finished
--- RBK0320I: Removing incomplete backup. This may take some time. Please be patient
??? RBK0005E: Backup failed. Check previous error messages for details
--- RBK0010I: RASPY4: raspiBackup.sh V0.7.0.2-m_866 - 2025-04-14 (6da3692) stopped at Tue 15 Apr 17:56:46 CEST 2025 with rc 102
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log
pi@RASPY4:/media/ssd/RASPY4/RASPY4@debian12-tar-backup-20250413-154932 $

where I'm wrong?

AndyZoIoT, 2025-04-16:
To recap, because I assume to was not so clear.  I have:
A) a 'Running system', boot from SSD1, and weekly save backup to USBFLASH.
B) a 'Test system', used for test the restore:  boot from SD card, Restore the image from USBFLAH (moved from AtoB) to new SSD2.

Have I to do the above test on system A or system B?
Simply install updated RaspiBackup on system A, and run a new backup as usual, and try to restore on system B?

framps, 2025-04-16:
1) Create a backup on your running system with updated or original raspiBackup
2) Restore the backup with your SD card with the updated raspiBackup and attach the restore log
3) On the SD card run syncUUIDs on SSD2 to check the UUIDs

AndyZoIoT, 2025-04-17:
Here what I did:
pi@RASPY4:~ $ sudo fdisk -l | egrep "^Disk /|^/dev"  :
	Disk /dev/mmcblk0: 14.84 GiB, 15931539456 bytes, 31116288 sectors
	/dev/mmcblk0p1         8192  1056767  1048576  512M  c W95 FAT32 (LBA)
	/dev/mmcblk0p2      1056768 31116287 30059520 14.3G 83 Linux
	Disk /dev/sda: 238.47 GiB, 256060513792 bytes, 500118191 sectors
	/dev/sda1          8192   1056767   1048576  512M  c W95 FAT32 (LBA)
	/dev/sda2       1056768 156301487 155244720   74G 83 Linux
	Disk /dev/sdb: 7.62 GiB, 8179941376 bytes, 15976448 sectors
	/dev/sdb1  *       64 15976447 15976384  7.6G  7 HPFS/NTFS/exFAT

pi@RASPY4:~ $ sudo mount /dev/sdb1 /media/ssd -o uid=pi,gid=pi 
...
pi@RASPY4:~ $ ls /media/ssd/RASPY4
pi@RASPY4:~ $ RASPY4@debian12-tar-backup-20250413-154932

pi@RASPY4:~ $ sudo raspiBackup.sh -g --resizeRootFS- -d /dev/sda /media/ssd/RASPY4/RASPY4-tar-backup-20250411-091810
.......
--- RBK0076I: Restore finished successfully
--- RBK0010I: RASPY4: raspiBackup.sh V0.7.0.2 - 2025-03-29 (3a9ab26) stopped at Thu 17 Apr 12:33:00 CEST 2025 with rc 0
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.logr

here the log:  [raspiBackup.logr.txt](https://github.com/user-attachments/files/19793514/raspiBackup.logr.txt)

then... "syncUUIDs on SSD2 to check the UUIDs" ???? 

However, with this updated raspiBackup the result is succesfully! Thanks!
Andrea.

framps, 2025-04-18:
> However, with this updated raspiBackup the result is succesfully! Thanks!

Hm ... you used the original raspiBackup - not the updated on I created for you and it works now 🤔

AndyZoIoT, 2025-04-18:
Sure, it is strange that it works...

I wrongly started the restore with:
pi@RASPY4:~ $ sudo raspiBackup.sh -g --resizeRootFS- -d /dev/sda /media/ssd/RASPY4/RASPY4-tar-backup-20250418-071503/
--- RBK0009I: RASPY4: raspiBackup.sh V0.7.0.2 - 2025-03-29 (3a9ab26) started at Fri 18 Apr 09:00:23 CEST 2025

instead of:
pi@RASPY4:~ $ **sudo ./raspi**Backup.sh -g --resizeRootFS- -d /dev/sda /media/ssd/RASPY4/RASPY4-tar-backup-20250418-071503/

...so I used the wrong version, and again, it's strange that this time it worked successfully...

AndyZoIoT, 2025-04-18:
Thanks, I'm not able to reproduce the problem, in any case I can successfully restore using previous version in case, thanks for your time.

closed 2025-04-18 by AndyZoIoT