framps/raspiBackup #867: Use new QEMU environment for regression tests

framps, 2025-04-07:
Current regression setup should be upgraded to use [this](https://crycode.de/raspberry-pi-4-emulieren-mit-qemu/) setup.

UnconnectedBedna, 2025-06-09:
I would suggest `systemd-nspawn` instead. Only one apt package needed (and systemd ofc): `sudo apt install systemd-container`

Then this gives you "chroot access" (assuming you run as root):

```
if systemctl is-active -q systemctl-resolved.service; then
  systemd-nspawn --resolv-conf=replace-stub -D /mnt bin/bash
else
  systemd-nspawn --resolv-conf=copy-host -D /mnt bin/bash
fi
```

This is if you need internet access, otherwise `sudo systemd-nspawn -D /mnt bin/bash` is enough.

https://wiki.debian.org/RaspberryPi/qemu-user-static (Option 2.2)

UnconnectedBedna, 2025-06-10:
You are correct, this is not "booting" the system, you are just changing "root" to another filesystem. You still use the same kernel you booted from. I don't think it will detect those missconfigurations. Empathize on _think_, it actually might, I don't know...
This is to make changes you can't do with a simple editing of a config file, like installing with apt, or rebuilding initramsf for example.
But maybe there is some kind of application that runs a test for configurations, I don't know...

I would just do it in bash on a mounted loop device, no need to emulate/chroot/nspawn anything.
For example:
```
# loop img file and mount somewhere, I will assume /mnt here and that it is looped at loop0 (loop0p1 for boot and loop0p2 for /)

BACKUP_BOOT_PARTUUID=$(lsblk -no partuuid /dev/loop0p1)
BACKUP_ROOT_PARTUUID=$(lsblk -no partuuid /dev/loop0p2)

if echo "$BACKUP_BOOT_PARTUUID" | grep -q -f /mnt/etc/cmdline.txt; then
  echo 'config ok'
else
  <do whatever needed to be done to fix the config>
fi
```
Or: `grep -q $BACKUP_BOOT_PARTUUID /mnt/etc/cmdline.txt`, I just find the above line easier to read

And then something similar for fstab, like:
```
if echo "$BACKUP_ROOT_PARTUUID / " | grep -q -f /mnt/etc/fstab; then
  yadayadayada etc
```

UnconnectedBedna, 2025-06-10:
How would that even be possible?
I'm all for redundancy, but that seems extremely unlikely.
A rsync would confirm it in a second or two.

framps, 2025-06-10:
> My point is if there are tools present, why create a bunch of dependencies for stuff not needed.

I see your point. But I frankly trust a regression test which executes the Raspberry boot sequence more than to check any known possible boot problems. 

I agree it's much more difficult to setup a QEMU environment which executes the boot sequence than to check for known boot problems.

UnconnectedBedna, 2025-06-10:
Again, my point is not the difficulty, it's the application requiring to install a bunch of dependencies when there are already tools that can do the same thing.

A `diff` for example checks if there is even one single data bit of difference in the files. It can even tell you what differences there are.
You don't trust `git`? the tool the linux kernel uses to make sure things are correct? xD

UnconnectedBedna, 2025-06-10:
`diff` will tell you ALL differences, ie missing files too.

UnconnectedBedna, 2025-06-12:
> I know you also maintain a backup tool. Do you test it the way you proposed?

I do not need to, since rsync does it for me when the backup is created.

I try to keep dependencies to a minimum, that is why I suggested you to not force the user to install 14 packages:
```
gcc-aarch64-linux-gnu, g++-aarch64-linux-gnu, qemubuilder, qemu-system-gui, qemu-system-arm, qemu-utils, qemu-system-data, qemu-system, bison, flex, guestfs-tools, libssl-dev, telnet & xz-utils
```
but rather use the tool most likely already available on the system, ie `rsync`.

UnconnectedBedna, 2025-06-14:
> The regression test is executed by me when I finished my manual tests for the new features, enhancements and bugs on my local environment. A raspiBackup user doesn't have to install the regression environment I use.

Oh, I thought this whole thread was a "feature request" that will somehow be implemented into the application because you added the "feature" and "development" flags and:

> Current regression setup should be upgraded to use [this](https://crycode.de/raspberry-pi-4-emulieren-mit-qemu/) setup.

> I see your shrink-backup cannot use my QEMU approach and you have to use a different way to test your tool.

It could, but why should it? Why use 14 dependencies when only one is needed?
And no, I don't have to use anything to error correct the data, I just make sure the copied data is correct, and rsync does that very easily.

> Thank you very much for your hints but I will continue my QEMU approach given raspiBackup is targeted for Raspberries only. The approach you suggested is the way you have to go when you're not bounded to a Raspberry 👍

Has absolutely nothing to do with weather it's rpi or not, it has to do with the tools used.
Does not even have to do with linux, it has to do with data checks.

framps, 2025-06-14:
> The way I test shrink-backup is to make actual real life tests and then boot those images on real devices, that is how I make sure the application works correctly. An emulated boot would not suffice my quality assurance of the application actually working on bare metal.

> A lot of work and since shrink-backup "support" more than only a few rpi models, it takes a lot of time because I need to do it on quite a few devices. But I think to ensure the quality of the application it's worth it.

That's indeed the best way to test - just use real HW. Unfortunately this takes a lot of test time and a lot of real HW. I initially also spent a lot time to test with real Raspberry HW and finally decided to create a Regressionpackage which simulates a Raspberry HW with QEMU but runs without any user intervention over night. I also initially supported other HW but given the fact I don't own all the other HW raspiBackup may be run successfully I restricted raspiBackup to Raspberry HW and RaspbianOS only.

The QEMU regression simulation caught already a lot of issues and I'm confident it will catch any issue in the future. 

Great to know you spent your time to execute real HW tests.  

> you added the "feature" and "development" flags

It's a development feature, not a user feature 😉

framps, 2025-10-04:
Transition code available [here](https://github.com/framps/raspiBackup/tree/m_new_emu)