framps/raspiBackup #874: Extension not found even it is there

bcutter, 2025-04-21:
In every raspiBackup.log on one Pi I get

```
20250418-053733 DBG 2394:              --> callExtensions notify
20250418-053733 DBG 2461:                  --- raspiBackup_PushingBox_NotifyBackup.sh not found - skipping
20250418-053733 DBG 2466:              <-- callExtensions
```

On another machine with the same script with the same name at the same place, I positively get

```
04-18-2025 02:09:05 --- RBK0267I: Erweiterung raspiBackup_PushingBox_NotifyBackup_post.sh wird aufgerufen
```

The only difference is:
- Pi with missing log: `DEFAULT_EXTENSIONS="PushingBox_NotifyBackup"`
- Pi with existing log: `DEFAULT_EXTENSIONS="PushingBox_NotifyBackup ClearLocalBackupStorage"`

How to further debug this? Is the syntax correct in the config?

framps, 2025-04-21:
> Somehow quite confusing...

I see your point and will update the extension doc shortly.

In the `DEFAULT_EXTENSIONS` definition you have to use just `notify` for the notification extension and the script has to be called raspiBackup_notify.sh. No pre or post. Make sure the script is located in some of the default pathes.

If you want to use pushover, slack or telegram just define the corresponding definitions. There is no notify extension required.

bcutter, 2025-04-21:
Well the confusing part is: everything seems to work and one log file confirms that assumption - while the other mentions it can't find the script.

The notify script is a custom one using a specific service, it is in place for years "as it is" (name with post etc.).

So I'm not sure if I need to change anything at all.

framps, 2025-04-21:
It doesn't harm if you have a post suffix on the notify extension script. It's mandatory for mem, temp and disk extension. I just updated the extension doc ;-) 

> everything seems to work and one log file confirms that assumption - while the other mentions it can't find the script.

Do you run the same raspiBackup version on both systems? Maybe there is a regression in 0.7.

bcutter, 2025-04-21:
Yes same version (latest stable).

Just to make it clear: those two log files excerpts

```
04-18-2025 05:36:22 --- RBK0267I: Erweiterung raspiBackup_PushingBox_NotifyBackup_post.sh wird aufgerufen
```

```
20250418-053733 DBG 2394:              --> callExtensions notify
20250418-053733 DBG 2461:                  --- raspiBackup_PushingBox_NotifyBackup.sh not found - skipping
20250418-053733 DBG 2466:              <-- callExtensions
```

are from the very _same_ system.

framps, 2025-04-21:
Please attach both debug logs

bcutter, 2025-04-21:
Logs are too chatty, I won't provide them publicly/fully. What are you looking for?

That's from the `raspiBackup.log`:
```
20250418-053622 DBG 2394:              --> callExtensions post
20250418-053622 DBG 2430:                  --- Calling raspiBackup_PushingBox_NotifyBackup_post.sh 0
20250418-053622 MSG 2434:                  --- RBK0267I: Erweiterung raspiBackup_PushingBox_NotifyBackup_post.sh wird aufgerufen 
20250418-053622 DBG 2779:                  --> executeShellCommand . raspiBackup_PushingBox_NotifyBackup_post.sh 0
20250418-053722 DBG 2786:                  <-- executeShellCommand 0
20250418-053722 DBG 2456:                  --- Extension RC: 0
20250418-053722 DBG 2466:              <-- callExtensions 
```

And that's from the same file ~ 50 lines later:
```
20250418-053733 DBG 2369:          <-- logFinish 
20250418-053733 DBG 5194:          --> callNotificationExtension 0
20250418-053733 DBG 2394:              --> callExtensions notify
20250418-053733 DBG 2461:                  --- raspiBackup_PushingBox_NotifyBackup.sh not found - skipping
20250418-053733 DBG 2466:              <-- callExtensions 
20250418-053733 DBG 5203:              --- NotificationExtension rc: 0
20250418-053733 DBG 5208:          <-- callNotificationExtension 0
```

Does that maybe mean the script is once run as "post" script and once tried to run as "notification extension"?

`raspiBackup.conf` references that script only once:
```
# Aufzurufende Erweiterungen beim Backuplauf
DEFAULT_EXTENSIONS="PushingBox_NotifyBackup"
```

framps, 2025-04-23:
For all extensions defined in `DEFAULT_EXTENSIONS` raspiBackup_<extension>.sh is not called. Instead either raspiBackup_<extension>_pre.sh and/o raspiBackup_<extension>_post.sh.  If you want to customize your own notification just provide a script raspiBackup_notify.sh.