framps/raspiBackup #882: Installation issue

truter1234, 2025-05-10:
Dear all,
I have 2 raspberry pis. I have installed raspiBackup on one successfully and created a backup (raspi5). I wanted to install on the second one (Raspi zero 2 w) following the same steps (executing the automatic installation script) and I hit error 'RBI0021E: Unable to connect to www.linux-tips-and-tricks.de. wget RC: 4'. I've tested connectivity on the PI and it works (i've pinged various pages incl. the Linux-tips...).
What could I be doing wrong?

truter1234, 2025-05-10:
After a number of reboots, retries it started working. Very weird as I have not changed the config.

closed 2025-05-10 by truter1234

framps, 2025-05-10:
Maybe my websiteprovider did some kind of maintenance ...

Glad it works now

0x62, 2025-05-30:
I'm getting this error too (v0.4.8.3)

framps, 2025-05-31:
I have a managed webspace and my provider maintains this webspace from time to time and the website is then offline. 

Try again later 😉