framps/raspiBackup #885: Recycle is not working; maybe because of hostname with dash

mhertelxavo, 2025-05-17:
Hi,

old backups are not recycled.
Please find attached the logfile.
Maybe the issue is because of my raspis hostname: pi-eg?

Interesting point in logfile comes here:

   <-- SR_listBackupsToDelete 0
20250514-050432 DBG 9662:          --> countLines 
20250514-050432 DBG 9670:          <-- countLines 0
20250514-050432 DBG 6769:          --- TobeDeletedBackups 0: 
20250514-050432 MSG 6783:          --- RBK0219I: Keine Backups werden smart recycled 

[raspiBackup.log](https://github.com/user-attachments/files/20263563/raspiBackup.log)

Kind regards
Michael

framps, 2025-05-17:
That's nasty. Please test if the default option "7 4 12 1" works.

mhertelxavo, 2025-05-19:
Hi framps,

unfortunatly no difference with that parameters.

Output of DEFAULT_SMART_RECYCLE_DRYRUN=1 looks good with both configs:

root@pi-eg:/media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250518-050001# /usr/local/bin/raspiBackup.sh -v
--- RBK0009I: pi-eg: raspiBackup.sh V0.7.0.2 - 2025-03-27 (0a41b79) Mo 19. Mai 10:08:59 CEST 2025 gestartet
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt
--- RBK0151I: Backuppfad /media/usbStick mit Partitionstyp ext4 wird benutzt
!!! RBK0216W: Wende nur Backupstrategie in /media/usbStick/pi-eg an
--- RBK0218I: Wende smarte Backupstrategie an. Täglich:10 Wöchentlich:4 Monatlich:14 Jährlich:10
--- RBK0219I: Keine Backups werden smart recycled
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250329-180120 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250401-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250428-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250501-050002 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250505-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250510-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250511-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250512-050005 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250513-050002 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250514-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250515-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250518-050001 Backup behalten
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0017I: Backup erfolgreich beendet

--- RBK0010I: pi-eg: raspiBackup.sh V0.7.0.2 - 2025-03-27 (0a41b79) Mo 19. Mai 10:09:07 CEST 2025 beendet mit Returncode 0
--- RBK0167I: Eine eMail wird versendet
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert
root@pi-eg:/media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250518-050001# vi /usr/local/etc/raspiBackup.conf
root@pi-eg:/media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250518-050001# /usr/local/bin/raspiBackup.sh -v
--- RBK0009I: pi-eg: raspiBackup.sh V0.7.0.2 - 2025-03-27 (0a41b79) Mo 19. Mai 10:09:42 CEST 2025 gestartet
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt
--- RBK0151I: Backuppfad /media/usbStick mit Partitionstyp ext4 wird benutzt
!!! RBK0216W: Wende nur Backupstrategie in /media/usbStick/pi-eg an
--- RBK0218I: Wende smarte Backupstrategie an. Täglich:7 Wöchentlich:4 Monatlich:12 Jährlich:1
--- RBK0219I: Keine Backups werden smart recycled
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250329-180120 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250401-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250428-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250501-050002 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250505-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250512-050005 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250513-050002 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250514-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250515-050001 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250518-050001 Backup behalten
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0017I: Backup erfolgreich beendet
--- RBK0010I: pi-eg: raspiBackup.sh V0.7.0.2 - 2025-03-27 (0a41b79) Mo 19. Mai 10:09:50 CEST 2025 beendet mit Returncode 0
--- RBK0167I: Eine eMail wird versendet

Kind regards
Michael

framps, 2025-05-19:
OK. I will try to recreate your issue. The debug log doesn't help - unfortunately. Maybe you are right with the dash in the hostname.

mhertelxavo, 2025-05-21:
Hi framps,

thank you very much for your effort!

Hostname is **pi-eg** (see also pathnames in third comment above).

[image]

Masquarading is fully automated/self contained in the script, right?

Kind regards
Michael

framps, 2025-05-21:
Ok. Looks like you use `pi-eg` as your sender eMail because that's masqueraded before the hostname is masqueraded.

```
20250514-050002 DBG 4136:      --- DEFAULT_SENDER_EMAIL="@@@@5"
```

I now create a similar bunch of backups and try to recreate your issue. As of now I have no clue why it doesn't work for you 😒

mhertelxavo, 2025-05-21:
Looks good for me:

root@pi-eg:/usr/local/bin# ./raspiBackup.sh --smartRecycleDryrun+
--- RBK0009I: pi-eg: raspiBackup.sh V0.7.0.2-m_885 - 2025-05-21 (5afbd4e) Mi 21. Mai 16:38:38 CEST 2025 gestartet
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt
--- RBK0151I: Backuppfad /media/usbStick mit Partitionstyp ext4 wird benutzt
!!! RBK0216W: Wende nur Backupstrategie in /media/usbStick/pi-eg an
--- RBK0218I: Wende smarte Backupstrategie an. Täglich:7 Wöchentlich:4 Monatlich:12 Jährlich:1
--- RBK0217I: 36 Backups werden smart recycled. 11 Backups werden aufgehoben. Bitte Geduld
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250331-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250402-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250403-050004 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250404-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250405-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250406-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250407-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250408-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250409-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250410-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250412-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250413-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250414-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250415-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250416-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250417-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250419-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250420-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250421-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250422-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250423-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250424-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250426-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250427-050008 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250429-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250430-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250502-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250503-050005 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250504-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250506-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250507-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250509-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250511-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250513-050002 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250514-050001 Backup löschen
!!! RBK0220W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250521-050001 Backup löschen
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250329-180120 Backup behalten
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250401-050001 Backup behalten
[... cut ...]
!!! RBK0222W: Smart Backup Strategie würde /media/usbStick/pi-eg/pi-eg@raspbian10-rsync-backup-20250521-153833 Backup behalten
--- RBK0033I: Bitte warten bis aufgeräumt wurde
--- RBK0017I: Backup erfolgreich beendet
--- RBK0010I: pi-eg: raspiBackup.sh V0.7.0.2-m_885 - 2025-05-21 (5afbd4e) Mi 21. Mai 16:38:50 CEST 2025 beendet mit Returncode 0
--- RBK0167I: Eine eMail wird versendet
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert

My old config (DEFAULT_SMART_RECYCLE_OPTIONS="10 4 14 10") keeps 2 more backups.

framps, 2025-05-21:
Thank you very much you verified the fix 👍 

I will spend some additional time to understand why this happens for you but not for the other users. When I'm done and published the fix I'll update this issue and let you know.

framps, 2025-05-23:
@mhertelxavo I spent a lot of time to check why the recycle doesn't work for you. I have to understand why it doesn't work to make sure it's a specific issue or there is a general issue with smart recycle and raspiBackup. If it's a general issue it's a serious bug. But actually smart recycle is out there for a long time and nobody detected this issue.

My 7412 tests all succeed with current code. I checked the test code - and it took a while until I was able to understand it again. I wrote this  code a lot of time ago and it's complicated code. 

The I ran the 7412 test on a test system which uses pi-eg as hostname and the test fails. But if I use `raspberrypi-bullseye64-lite` everything works fine 🤔  So the issue is somehow related to the hostname `pi-eg`.

Initially you mentioned it may be because of the hostname you use. Why thought you it's related to the hostname? I have to find out why it's related to your hostname ❗ . Maybe you can give me some hint which helps me to locate the exact root cause why it's related to pi-eg and save me investigation time 😉

framps, 2025-05-23:
Modified 7412 test to check whether files not even still exist after recycle but also if all other backups are deleted in https://github.com/framps/raspiBackup/commit/207c5894980b97d9a53f84fe7420d52744c0c9ad

Root cause why hostname `pi-eg` causes the issue still not known 😒

mhertelxavo, 2025-05-23:
Hi framps,

I just thought maybe it is a problem of the separator dash.

Maybe the problem is because of it's shortness?
2 letter dash 2 letters.

framps, 2025-05-23:
Never mind. I know it's not a general smart recycle issue and somehow related to your hostname and I don't have to alert everybody who uses raspiBackup and smart recycle.

I now have to check the code very carefully to find out why raspiBackup complaints  about hostname `pi-eg` 😏

framps, 2025-05-24:
Tests without fix:
 
OK `raspberry-pi-bookworm`
FAIL `pi-additional-eg`

framps, 2025-05-24:
Root cause is [this code line](https://github.com/framps/raspiBackup/blob/207c5894980b97d9a53f84fe7420d52744c0c9ad/raspiBackup.sh#L9805) and the option `-e "_"` followed by an incorrect closing parenthesis.

This causes all smart recycles to fail if a hostname includes the string `-e`

framps, 2025-05-24:
Fix published in v0.7.0.3. Please use `sudo raspiBackup -U` to upgrade your current version.

closed 2025-05-24 by framps