framps/raspiBackup #898: SMB Share mounted to /backup

mark-cassidy, 2025-07-07:
Hi,

I have an SMB share mounted by fstab at boot to /backup.

I've checked the source code for the backup script and it seems the command "lsblk -l -o name,mountpoint" is used to determine if the local device is being used.

Since this essentially shows I a false positive - lsblk will say that my /backup folder is on the local device; there's no check if its a mount point for other storage.

I can run the script with the -c switch to ignore the error but that's not giving me the warm and fuzzies.

My questions is; Is there a better way I should be mounting the SMB share at boot to avoid this issue entirely?

framps, 2025-07-07:
Please attach the debug log.

mark-cassidy, 2025-07-07:
Hi,

Thanks for getting back to me.

Never mind I fixed it.  There was a permission error on the credentials file used to connect to the SMB share.  Backup is currently running.

Regards

Mark

framps, 2025-07-07:
Hi Mark,

great you found the root cause of your issue 👍 . This proves it was a great idea to add this check. 😉 If the mount fails because of an access permission issue the backup will indeed be written on the local device. Not sure whether you completed a backup run with option `-c` but if your root partition is used already heavily this may cause your system to die because it will run out of space.

When raspiBackup was quite young (approx 2013) and folks used small SD cards with about 8GB I had multiple folks who had the same issue like you - and their system died when they used raspiBackup 😒 That's why I added this guard.